zephyrproject-rtos/zephyr #29740: OTA using Thread

nehalrohit, 2020-11-03:
**Is your feature request related to a problem? Please describe.**
A clear and concise description of what the problem is.

**Describe the solution you'd like**
A clear and concise description of what you want to happen.

**Describe alternatives you've considered**
A clear and concise description of any alternative solutions or features you've considered.

**Additional context**
Add any other context or graphics (drag-and-drop an image) about the feature request here.

Hello Community,

I want to implement OTA using thread on NRF52840dk (PCA10056) and nrf52840 dongle (PCA10059).
How am I suppose to start with OTA and multi broadcasting OTA using thread  ? (Documentation and Link)

rlubos, 2020-11-03:
Hi, unfortunately, we don't have an out-of-the-box example for OTA with OpenThread. For sure you'd need to enable `mcuboot` in your OT project and use it's API ([mcuboot](https://github.com/zephyrproject-rtos/zephyr/blob/master/include/dfu/mcuboot.h), [flash_img](https://github.com/zephyrproject-rtos/zephyr/blob/master/include/dfu/flash_img.h)) to write consecutive FW chunks recevied from the server (I'm not sure what FW delivery method are you aiming at).

Two examples of existing FOTA libraries that we currently have in Zephyr are [updatehub](https://github.com/zephyrproject-rtos/zephyr/blob/master/subsys/mgmt/updatehub/updatehub.c) and [hawkbit](https://github.com/zephyrproject-rtos/zephyr/blob/master/subsys/mgmt/hawkbit/hawkbit.c). The former one uses CoAP to deliver FW, the latter uses HTTP.

markus-becker-tridonic-com, 2020-11-03:
Another option is [SMP](https://docs.zephyrproject.org/latest/samples/subsys/mgmt/mcumgr/smp_svr/README.html) with UDP  as transport layer.

retfie, 2020-11-04:
I have a working FOTA over OT that I'm using successfully for more than a year and half in a test bed of more than 10 FTD and SED devices. It's based on @mike-scott  work on https://github.com/foundriesio/dm-lwm2m , but not sure what its license is, and could we use it on Zephyr. If he confirm that it's OK to use it as a base I could push a PR for review here.

nehalrohit, 2020-11-06:
Hello Community,
Thanks for your reply.
I am trying to updatehub the sample demo for nrf52840dk and follow the below link.
https://docs.zephyrproject.org/latest/samples/subsys/mgmt/updatehub/README.html 

the step follows.
1- build  mcuboot **west build -b nrf52840dk_nrf52840** and flash using **west flash**.
2 -updatehub demo  **west build -b  nrf52840dk_nrf52840 -- -DOVERLAY_CONFIG="prj.conf overlay-prj.conf "** and flash using **west flash --hex build/zephyr/app_signed.hex** 
3- run on terminal **docker run -it -p 8080:8080 -p 5683:5683/udp --rm   updatehub/updatehub-ce:latest** 

I have a problem with **updatehub: Cannot connect to UDP remote** see the below image.

[image]

nehalrohit, 2020-11-10:
> I have a working FOTA over OT that I'm using successfully for more than a year and half in a test bed of more than 10 FTD and SED devices. It's based on @mike-scott work on [...]

Hello retfie,
can you please suggest to me how you do OTA with Openthread or you can provide some steps to follow( link or code)?

                                                                                                                                         Thanks 
                                                                                                                                          Nehal

nandojve, 2020-11-11:
CC: @otavio

nandojve, 2020-11-11:
Hi @nvlsianpu , @utzig ,

I ran UpdateHub using WIFI, just to make sure everything is Ok.
Board: nrf52840dk_nrf52840
WIFI: ESP-8266 Shield
Zephyr: 89f6e24590 + my configs

1) Flash MCUboot
```
*** Booting Zephyr OS build zephyr-v2.4.0-1442-gbc4823a21183  ***
[00:00:00.465,484] <inf> mcuboot: Starting bootloader
[00:00:00.471,862] <inf> mcuboot: Primary image: magic=unset, swap_type=0x1, copy_done=0x3, image_ok=0x3
[00:00:00.482,147] <inf> mcuboot: Secondary image: magic=unset, swap_type=0x1, copy_done=0x3, image_ok=0x3
[00:00:00.492,553] <inf> mcuboot: Boot source: none
[00:00:00.498,016] <wrn> mcuboot: Failed reading image headers; Image=0
[00:00:00.505,523] <err> mcuboot: Unable to find bootable image
```

2) Flash initial image, version 1.0.0
```
*** Booting Zephyr OS build zephyr-v2.4.0-1442-gbc4823a21183  ***
[00:00:00.426,116] <inf> mcuboot: Starting bootloader
[00:00:00.432,525] <inf> mcuboot: Primary image: magic=good, swap_type=0x1, copy_done=0x3, image_ok=0x3
[00:00:00.442,687] <inf> mcuboot: Secondary image: magic=unset, swap_type=0x1, copy_done=0x3, image_ok=0x3
[00:00:00.453,094] <inf> mcuboot: Boot source: primary slot
[00:00:00.465,362] <inf> mcuboot: Swap type: none
[00:00:00.671,875] <inf> mcuboot: Bootloader chainload address offset: 0xc000
[00:00:00.679,687] <inf> mcuboot: Jumping to the first image slot

uart:~$ *** Booting Zephyr OS build zephyr-v2.4.0-1442-gbc4823a21183  ***
[00:00:00.251,190] <inf> wifi_esp: Waiting for interface to come up
[00:00:00.570,800] <inf> wifi_esp: ESP Wi-Fi ready
[00:00:00.577,209] <inf> main: UpdateHub sample app started
[00:00:00.577,209] <inf> main: Confirming the boot image
[00:00:00.581,604] <inf> main: Network disconnected
[00:00:06.222,717] <inf> main: Network connected
[00:00:06.222,747] <inf> main: Starting UpdateHub polling mode
[00:00:06.222,747] <inf> updatehub: SHA-256 verification on download and from flash
[00:00:08.439,239] <err> updatehub: Could not receive data
[00:01:08.982,330] <inf> updatehub: No update available
[00:02:09.403,137] <inf> updatehub: No update available
uart:~$ updatehub info 
Unique device id: 3205bbf32cf69807
Firmware Version: 1.0.0
Product uid: e4d37cfe6ec48a2d069cc0bbb8b078677e9a0d8df3a027c4d8ea131130c4265f
UpdateHub Server: uhu.geo4cities.com
[00:03:09.756,195] <inf> updatehub: Probe metadata received
[00:04:57.230,255] <inf> updatehub: Firmware download complete
[00:04:58.535,461] <inf> updatehub: Image flashed successfully, you can reboot now
```

3) Reboot after receive new image, version 2.0.0
```
uart:~$ *** Booting Zephyr OS build zephyr-v2.4.0-1442-gbc4823a21183  ***
[00:00:00.260,772] <inf> mcuboot: Starting bootloader
[00:00:00.267,150] <inf> mcuboot: Primary image: magic=good, swap_type=0x1, copy_done=0x3, image_ok=0x1
[00:00:00.277,343] <inf> mcuboot: Secondary image: magic=good, swap_type=0x2, copy_done=0x3, image_ok=0x3
[00:00:00.287,658] <inf> mcuboot: Boot source: none
[00:00:00.293,273] <inf> mcuboot: Swap type: test
[00:00:00.583,190] <inf> mcuboot: Bootloader chainload address offset: 0xc000
[00:00:00.590,972] <inf> mcuboot: Jumping to the first image slot

uart:~$ *** Booting Zephyr OS build zephyr-v2.4.0-1442-gbc4823a21183  ***
[00:00:00.250,579] <inf> wifi_esp: Waiting for interface to come up
[00:00:00.570,251] <inf> wifi_esp: ESP Wi-Fi ready
[00:00:00.576,660] <inf> main: UpdateHub sample app started
[00:00:00.576,660] <inf> main: Confirming the boot image
[00:00:00.580,993] <inf> main: Network disconnected
uart:~$ updatehub info 
Unique device id: 3205bbf32cf69807
Firmware Version: 1.0.0
Product uid: e4d37cfe6ec48a2d069cc0bbb8b078677e9a0d8df3a027c4d8ea131130c4265f
```

Details:
1) [00:00:06.222,747] <inf> updatehub: SHA-256 verification on download and from flash
This means that we will run SHA-256 on downloaded data and on flash after stored all received frames

2) [00:04:58.535,461] <inf> updatehub: Image flashed successfully, you can reboot now
This means that UpdateHub storaged a valid image at slot-2

3) [00:00:00.277,343] <inf> mcuboot: Secondary image: magic=good, swap_type=0x2, copy_done=0x3, image_ok=0x3
This confirms that, there is a valid image

Q: Why the system is not swamping image-2 on slot-1?

utzig, 2020-11-11:
MCUboot does not start an upgrade unless it is requested by writing the proper metadata in the secondary slot. To do it you can build a padded image, with `imgtool` would be `--pad` and optionally use `--confirm`. Unfortunately this makes the image as large as the slot. A better alternative is to add support in your running firmware (updatehub library or something like that), that after downloading the full image, and optionally checking that it is valid, must call `boot_request_upgrade` from the dfu/mcuboot code. This should trigger a test upgrade, or pass the confirm flag to make it permanent.

utzig, 2020-11-11:
> MCUboot does not start an upgrade unless it is requested by writing the proper metadata in the secondary slot. To do it you can build a padded image, with `imgtool` would be `--pad` and optionally use `--confirm`. Unfortunately this makes the image as large as the slot. A better alternative is to add support in your running firmware (updatehub library or something like that), that after downloading the full image, and optionally checking that it is valid, must call `boot_request_upgrade` from the dfu/mcuboot code. This should trigger a test upgrade, or pass the `permanent` flag to make it, duh, permanent!

nandojve, 2020-11-11:
I think we are doing it, see below:

https://github.com/zephyrproject-rtos/zephyr/blob/89f6e24590cf9e9b66f123ee0ce2ae5818ce1aeb/subsys/mgmt/updatehub/updatehub.c#L928-L946

We print the message "Image flashed successfully, you can reboot now". It is the last line before step 3.

utzig, 2020-11-11:
Yeah, the code looks correct; try building MCUboot will debug logging, there may be more helpful messages.

nandojve, 2020-11-12:
```
[00:00:00.260,986] <inf> mcuboot: Starting bootloader
[00:00:00.267,333] <inf> mcuboot: Primary image: magic=unset, swap_type=0x1, copy_done=0x3, image_ok=0x1
[00:00:00.277,648] <inf> mcuboot: Secondary image: magic=good, swap_type=0x2, copy_done=0x3, image_ok=0x3
[00:00:00.287,994] <inf> mcuboot: Boot source: none
[00:00:00.293,609] <inf> mcuboot: Swap type: test
[00:00:00.298,919] <dbg> mcuboot.swap_erase_trailer_sectors: erasing trailer; fa_id=2
[00:00:00.594,024] <inf> mcuboot: Bootloader chainload address offset: 0xc000
[00:00:00.601,867] <inf> mcuboot: Jumping to the first image slot

```

nehalrohit, 2020-11-12:
Hi @nandojve  

I want to use it as WIFI: nordic border -router.

nandojve, 2020-11-12:
> ```
> [00:00:00.260,986] <inf> mcuboot: Starting bootloader
> [00:00:00.267,333] <inf> mcuboot: Primary image: magic=unset, swap_type=0x1, copy_done=0x3, image_ok=0x1
> [00:00:00.277,648] <inf> mcuboot: [...]

Hi @utzig , it is working now!
I've been doing something wrong when creating the package for UpdateHub.

```
uart:~$ *** Booting Zephyr OS build zephyr-v2.4.0-1442-gbc4823a21183  ***
[00:00:00.260,986] <inf> mcuboot: Starting bootloader
[00:00:00.267,303] <inf> mcuboot: Primary image: magic=unset, swap_type=0x1, copy_done=0x3, image_ok=0x1
[00:00:00.277,587] <inf> mcuboot: Secondary image: magic=good, swap_type=0x2, copy_done=0x3, image_ok=0x3
[00:00:00.287,902] <inf> mcuboot: Boot source: none
[00:00:00.293,518] <inf> mcuboot: Swap type: test
[00:00:00.501,190] <dbg> mcuboot.swap_erase_trailer_sectors: erasing trailer; fa_id=1
[00:00:00.592,926] <dbg> mcuboot.swap_status_init: initializing status; fa_id=1
[00:00:00.600,952] <dbg> mcuboot.boot_write_swap_info: writing swap_info; fa_id=1 off=0x66fd8 (0x72fd8), swap_type=0x2 image_num=0x0
[00:00:00.613,769] <dbg> mcuboot.boot_write_swap_size: writing swap_size; fa_id=1 off=0x66fd0 (0x72fd0)
[00:00:00.623,992] <dbg> mcuboot.boot_write_magic: writing magic; fa_id=1 off=0x66ff0 (0x72ff0)
[00:00:00.633,636] <dbg> mcuboot.swap_erase_trailer_sectors: erasing trailer; fa_id=2
[00:00:12.921,905] <dbg> mcuboot.boot_write_copy_done: writing copy_done; fa_id=1 off=0x66fe0 (0x72fe0)
[00:00:13.149,230] <inf> mcuboot: Bootloader chainload address offset: 0xc000
[00:00:13.171,813] <inf> mcuboot: Jumping to the first image slot

uart:~$ *** Booting Zephyr OS build zephyr-v2.4.0-1442-gbc4823a21183  ***
[00:00:00.251,129] <inf> wifi_esp: Waiting for interface to come up
[00:00:00.570,983] <inf> wifi_esp: ESP Wi-Fi ready
[00:00:00.577,362] <inf> main: UpdateHub sample app started
[00:00:00.577,362] <inf> main: Confirming the boot image
[00:00:00.581,756] <inf> main: Network disconnected
uart:~$ updatehub info 
Unique device id: 3205bbf32cf69807
Firmware Version: 2.0.0
```

@nehalrohit ,

I did the test using WIFI to make sure all infrastructure is OK using nrf52840dk_nrf52840 board, and it is.
We are working to enable network manager on UpdateHub and there is a branch that you can test WIFI:
https://github.com/UpdateHub/zephyr/tree/topic/uhu_thread_network_ut
In fact, that branch allows you to select any ETH/WIFI/MODEM.

I've been testing UpdateHub using https://docs.zephyrproject.org/latest/boards/shields/esp_8266/doc/index.html shield.
Currently, the https://docs.zephyrproject.org/latest/boards/shields/inventek_eswifi/doc/index.html have an UDP issue that we are working on it. The Inventek's have a far better performance.

We will add UpdateHub overlays for IEEE 802.15.4 and BLE for Zephyr 2.5. In fact, it is a matter of having the transport well configured and have internet access to enable UpdateHub on it.

If you already have a Thread Networking with Internet access we can tune UpdateHub to work.

nandojve, 2020-11-13:
Hi @nehalrohit 

I got UpdateHub over 6lowPAN working. The below config was necessary to fix DNS error:
```
CONFIG_DNS_SERVER_IP_ADDRESSES=y
CONFIG_DNS_SERVER1="2001:db8::1"

CONFIG_UPDATEHUB_SERVER="2001:db8::1"
```
I tested on my local network using RADVD to serve IPs and CoAP block size equal to 64 `CONFIG_UPDATEHUB_COAP_BLOCK_SIZE_EXP=2`

As I mention, we will update the sample application with the overlays soon.
In the meantime, I hope this could help you to move forward.

```
[00:00:00.256,622] <inf> mcuboot: Starting bootloader
[00:00:00.263,061] <inf> mcuboot: Primary image: magic=good, swap_type=0x1, copy_done=0x3, image_ok=0x1
[00:00:00.273,315] <inf> mcuboot: Secondary image: magic=unset, swap_type=0x1, copy_done=0x3, image_ok=0x3
[00:00:00.283,782] <inf> mcuboot: Boot source: primary slot
[00:00:00.296,142] <inf> mcuboot: Swap type: none
[00:00:00.548,004] <inf> mcuboot: Bootloader chainload address offset: 0xc000
[00:00:00.555,847] <inf> mcuboot: Jumping to the first image slot

*** Booting Zephyr OS build zephyr-v2.4.0-1459-g423ceeb89683  ***
[00:00:00.001,861] <inf> ieee802154_nrf5: nRF5 802154 radio initialized
[00:00:00.011,840] <inf> main: UpdateHub sample app started
[00:00:00.011,871] <inf> main: Confirming the boot image
[00:00:00.011,932] <inf> main: Network disconnected
uart:~$ net iface 

Interface 0x20000b1c (IEEE 802.15.4) [1]
========================================
Link addr : F4:CE:36:08:9E:58:3C:B7
MTU       : 125
IPv6 unicast addresses (max 3):
        fe80::f6ce:3608:9e58:3cb7 autoconf preferred infinite
        2001:db8::f6ce:3608:9e58:3cb7 autoconf preferred
IPv6 multicast addresses (max 4):
        ff02::1
        ff02::1:ff58:3cb7
IPv6 prefixes (max 2):
        2001:db8::/64
IPv6 default router :
        fe80::10a2:27ee:a8d1:c641
IPv6 hop limit           : 3
IPv6 base reachable time : 30000
IPv6 reachable time      : 37394
IPv6 retransmit timer    : 0
uart:~$ net ping 2001:db8::1
PING 2001:db8::1
8 bytes from 2001:db8::1 to 2001:db8::f6ce:3608:9e58:3cb7: icmp_seq=0 ttl=3 rssi=202 time=22 ms
8 bytes from 2001:db8::1 to 2001:db8::f6ce:3608:9e58:3cb7: icmp_seq=1 ttl=3 rssi=201 time=9 ms
8 bytes from 2001:db8::1 to 2001:db8::f6ce:3608:9e58:3cb7: icmp_seq=2 ttl=3 rssi=203 time=8 ms
uart:~$ updatehub run 
Starting UpdateHub run...
No update found
[00:00:37.747,619] <inf> updatehub: No update available
uart:~$ updatehub run 
Starting UpdateHub run...
[00:06:17.577,697] <err> updatehub: Could not receive data
[00:06:25.283,416] <inf> updatehub: Firmware download complete
[00:06:25.642,639] <inf> updatehub: Image flashed successfully, you can reboot now
*** Booting Zephyr OS build zephyr-v2.4.0-1442-gbc4823a21183  ***
[00:00:00.256,683] <inf> mcuboot: Starting bootloader
[00:00:00.263,122] <inf> mcuboot: Primary image: magic=good, swap_type=0x1, copy_done=0x3, image_ok=0x1
[00:00:00.273,376] <inf> mcuboot: Secondary image: magic=good, swap_type=0x2, copy_done=0x3, image_ok=0x3
[00:00:00.283,752] <inf> mcuboot: Boot source: none
[00:00:00.289,398] <inf> mcuboot: Swap type: test
[00:00:16.691,162] <inf> mcuboot: Bootloader chainload address offset: 0xc000
[00:00:16.699,005] <inf> mcuboot: Jumping to the first image slot

uart:~$ *** Booting Zephyr OS build zephyr-v2.4.0-1459-g423ceeb89683  ***
[00:00:00.001,831] <inf> ieee802154_nrf5: nRF5 802154 radio initialized
[00:00:00.010,772] <inf> main: UpdateHub sample app started
[00:00:00.010,772] <inf> main: Confirming the boot image
[00:00:00.010,894] <inf> main: Network disconnected
uart:~$ updatehub info 
Unique device id: 3c589e08fd6088b7
Firmware Version: 2.0.0
Product uid: e4d37cfe6ec48a2d069cc0bbb8b078677e9a0d8df3a027c4d8ea131130c4265f
UpdateHub Server: 2001:db8::1
uart:~$ net ping 2001:db8::1
PING 2001:db8::1
8 bytes from 2001:db8::1 to 2001:db8::f6ce:3608:9e58:3cb7: icmp_seq=0 ttl=3 rssi=210 time=21 ms
8 bytes from 2001:db8::1 to 2001:db8::f6ce:3608:9e58:3cb7: icmp_seq=1 ttl=3 rssi=209 time=10 ms
8 bytes from 2001:db8::1 to 2001:db8::f6ce:3608:9e58:3cb7: icmp_seq=2 ttl=3 rssi=210 time=9 ms
uart:~$ 
```

nehalrohit, 2020-11-19:
Hi @nandojve,
did you use the updatehub cloud server?
can you giving me the step of how you do with 6lowpan?

nandojve, 2020-11-19:
> Hi @nandojve,
> did you use the updatehub cloud server?

No, I used a local server. You can use the cloud server if you have IPv6 on your environment.
The easy way is creating a bridge between 6lowpan0 and eth0 interfaces. If you don't have a IPv6, you need a 6to4 configuration.
I talked with @otavio and we will work to create a yocto layer to enable a 6to4 NAT.

> can you giving me the step of how you do with 6lowpan?

We are documenting how to do it.

nehalrohit, 2020-11-20:
Hi @nandojve 
Thanks for your reply  :) 
I have a nordic border router with Internet access.
here  nrf52840dk ------> nordic border router (connected using thread)
where should I run the updatehub local server on the local machine or border router?
Here I merge updatehub+ openthread demo. and flash into nrf52840dk
[image]

what parameter i have to update in **CONFIG_UPDATEHUB_SERVER="10.42.0.8"**  ? is it local IP or DNS IP ?

nandojve, 2020-11-20:
This is my Linux config:

```
2: eth0: <BROADCAST,MULTICAST,UP,LOWER_UP> mtu 1500 qdisc mq state UP group default qlen 1000
    link/ether <MAC> brd ff:ff:ff:ff:ff:ff
    inet <NET V4>/24 brd xxxxxxxx scope global secondary noprefixroute eth0
       valid_lft forever preferred_lft forever
    inet6 <NET V6>/64 scope global dynamic noprefixroute 
       valid_lft 2591916sec preferred_lft 604716sec
    inet6 <LINK LOCAL V6>/64 scope link noprefixroute 
       valid_lft forever preferred_lft forever
5: wpan0: <BROADCAST,NOARP,UP,LOWER_UP> mtu 123 qdisc pfifo_fast state UNKNOWN group default qlen 300
    link/ieee802.15.4 1a:3e:98:f4:d3:17:66:e1 brd ff:ff:ff:ff:ff:ff:ff:ff
6: lowpan0@wpan0: <BROADCAST,MULTICAST,UP,LOWER_UP> mtu 1280 qdisc noqueue state UNKNOWN group default qlen 1000
    link/6lowpan 1a:3e:98:f4:d3:17:66:e1 brd ff:ff:ff:ff:ff:ff:ff:ff
    inet6 2001:db8::1/64 scope global 
       valid_lft forever preferred_lft forever
    inet6 fe80::183e:98f4:d317:66e1/64 scope link 
       valid_lft forever preferred_lft forever
```
I added manually **2001:db8::1/64** (may not be the best addr) to the lowpan0 iface. This is the IP address that boards will try reach.

`ip addr add 2001:db8::1/64 dev lowpan0`

You can add this to your RADVD config:
```
interface lowpan0 {
	AdvSendAdvert on;
	MinRtrAdvInterval 3;
	MaxRtrAdvInterval 30;
	AdvCurHopLimit 3;

	route ::/0 {
	};

	prefix ::/64 {
		Base6Interface eth0;
		AdvOnLink on;
		AdvAutonomous on;
		AdvRouterAddr off;
	};
};
```
This means, share eth0 prefix and share any other prefix at interface to lowpan0 network.

At board using native IEEE 802.15.4 6lowPAN I got:
```
Interface 0x20000b54 (IEEE 802.15.4) [1]
========================================
Link addr : F4:CE:36:08:9E:58:3C:B7
MTU       : 125
IPv6 unicast addresses (max 3):
        fe80::f6ce:3608:9e58:3cb7 autoconf preferred infinite
        <INET V6 PREFIX>:f6ce:3608:9e58:3cb7 autoconf preferred
        2001:db8::f6ce:3608:9e58:3cb7 autoconf preferred
IPv6 multicast addresses (max 4):
        ff02::1
        ff02::1:ff58:3cb7
IPv6 prefixes (max 2):
        <INET V6 PREFIX>::/64
        2001:db8::/64
IPv6 default router :
        fe80::183e:98f4:d317:66e1
IPv6 hop limit           : 3
IPv6 base reachable time : 30000
IPv6 reachable time      : 25205
IPv6 retransmit timer    : 0
uart:~$ net ping 2001:db8::1
PING 2001:db8::1
8 bytes from 2001:db8::1 to 2001:db8::f6ce:3608:9e58:3cb7: icmp_seq=0 ttl=3 rssi=213 time=17 ms
8 bytes from 2001:db8::1 to 2001:db8::f6ce:3608:9e58:3cb7: icmp_seq=1 ttl=3 rssi=213 time=8 ms
8 bytes from 2001:db8::1 to 2001:db8::f6ce:3608:9e58:3cb7: icmp_seq=2 ttl=3 rssi=213 time=8 ms
uart:~$ 
```

Once you can ping Linux Machine, the below config will give you access to UpdateHub docker at same machine:

```
# Add DNS Server
CONFIG_DNS_SERVER_IP_ADDRESSES=y
CONFIG_DNS_SERVER1="2001:db8::1"

# UpdateHub
CONFIG_UPDATEHUB_SERVER="2001:db8::1"
CONFIG_UPDATEHUB_COAP_BLOCK_SIZE_EXP=2
```

**CONFIG_UPDATEHUB_COAP_BLOCK_SIZE_EXP=2** this will instruct CoAP to use a payload block with 64 bytes. You can try using value **1** (32-bytes) and **0** (16-bytes). I not recommend CoAP blocks >=128 on 6lowPAN networks.

nehalrohit, 2020-11-21:
**If you follow the OT notes about Tayga (NAT-64) you probably will ping internet. Than, you can disable UpdateHub CE and remove CONFIG_UPDATEHUB_SERVER="2001:db8::1" to get access to "coap.updatehub.io" that is UpdateHub cloud server.
You may need test with/without CONFIG_DNS_SERVER1.**  --  as you suggest I changed in the .conf file.

# Turn on the updatehub
CONFIG_UPDATEHUB=y

#Minimal Heap mem pool size for the updatehub working
CONFIG_HEAP_MEM_POOL_SIZE=8192  
CONFIG_SYSTEM_WORKQUEUE_STACK_SIZE=2048

CONFIG_NET_IPV4=n
CONFIG_NET_CONFIG_NEED_IPV4=n

CONFIG_NET_IPV6=n
CONFIG_NET_DHCPV4=y

#Optional if you would like test on the your server
CONFIG_SHELL=y
CONFIG_UPDATEHUB_SHELL=y
CONFIG_SHELL_STACK_SIZE=6144

# Debug helpers
CONFIG_LOG=y
#CONFIG_UPDATEHUB_LOG_LEVEL_INF=y
CONFIG_UPDATEHUB_PRODUCT_UID="f9007a8c24b86aa426c4214bc05a2e931b5b53c8830ec8838e7699b68e58c666"

I still get a DNS error 
I am successfully sending data to https://panel.thethings.io.

[image]

[image]

nandojve, 2020-11-21:
UpdateHub uses sockets so I'm assuming that works with OT.
I try compile OT from current master and got:
```
[1/500] cd /home/gfbudke/zephyros/mainline/zephyr/build/modules/openthread/build && echo -e O...=1"\n"NRF_802154_IFS_ENABLED=0"\n"NRF52840_XXAA"\n"MBEDTLS_CONFIG_FILE="config-tls-generic.h"
-e OPENTHREAD_CONFIG_ENABLE_BUILTIN_MBEDTLS=0
OPENTHREAD_CONFIG_ENABLE_BUILTIN_MBEDTLS_MANAGEMENT=0
PACKAGE_NAME=OPENTHREAD
PACKAGE_VERSION=gf9fbfad01
OPENTHREAD_CONFIG_THREAD_VERSION=OT_THREAD_VERSION_1_1
KERNEL
__ZEPHYR__=1
_FORTIFY_SOURCE=2
BUILD_VERSION=zephyr-v2.4.0-1763-g0452db4c7ebd
__LINUX_ERRNO_EXTENSIONS__
__PROGRAM_START
NRF_802154_USE_RAW_API=1
NRF_802154_PENDING_SHORT_ADDRESSES=16
NRF_802154_PENDING_EXTENDED_ADDRESSES=16
NRF_802154_RX_BUFFERS=16
NRF_802154_CCA_MODE_DEFAULT=NRF_RADIO_CCA_MODE_ED
NRF_802154_CCA_CORR_LIMIT_DEFAULT=2
NRF_802154_CCA_CORR_THRESHOLD_DEFAULT=45
NRF_802154_CCA_ED_THRESHOLD_DEFAULT=45
NRF_802154_CSMA_CA_ENABLED=1
NRF_802154_TX_STARTED_NOTIFY_ENABLED=1
NRF_802154_ACK_TIMEOUT_ENABLED=1
NRF_802154_INTERNAL_RADIO_IRQ_HANDLING=0
NRF_802154_FRAME_TIMESTAMP_ENABLED=0
NRF_802154_DELAYED_TRX_ENABLED=1
NRF_802154_IFS_ENABLED=0
NRF52840_XXAA
MBEDTLS_CONFIG_FILE=config-tls-generic.h
[2/500] Preparing syscall dependency handling

[495/500] Linking CXX executable zephyr/zephyr_prebuilt.elf
Memory region         Used Size  Region Size  %age Used
           FLASH:      287756 B         1 MB     27.44%
            SRAM:       95712 B       256 KB     36.51%
        IDT_LIST:         120 B         2 KB      5.86%
[500/500] Linking CXX executable zephyr/zephyr.elf
/home/gfbudke/zephyr-sdk-0.12.0.b/arm-zephyr-eabi/bin/arm-zephyr-eabi-objdump: DWARF error: could not find variable specification at offset 2b55
/home/gfbudke/zephyr-sdk-0.12.0.b/arm-zephyr-eabi/bin/arm-zephyr-eabi-objdump: DWARF error: could not find variable specification at offset 19f8
/home/gfbudke/zephyr-sdk-0.12.0.b/arm-zephyr-eabi/bin/arm-zephyr-eabi-objdump: DWARF error: could not find variable specification at offset 1a04
/home/gfbudke/zephyr-sdk-0.12.0.b/arm-zephyr-eabi/bin/arm-zephyr-eabi-objdump: DWARF error: could not find variable specification at offset 15c1
/home/gfbudke/zephyr-sdk-0.12.0.b/arm-zephyr-eabi/bin/arm-zephyr-eabi-objdump: DWARF error: could not find variable specification at offset 4d0d
/home/gfbudke/zephyr-sdk-0.12.0.b/arm-zephyr-eabi/bin/arm-zephyr-eabi-objdump: DWARF error: could not find variable specification at offset 73e
/home/gfbudke/zephyr-sdk-0.12.0.b/arm-zephyr-eabi/bin/arm-zephyr-eabi-objdump: DWARF error: could not find variable specification at offset 74a
/home/gfbudke/zephyr-sdk-0.12.0.b/arm-zephyr-eabi/bin/arm-zephyr-eabi-objdump: DWARF error: could not find variable specification at offset 756
/home/gfbudke/zephyr-sdk-0.12.0.b/arm-zephyr-eabi/bin/arm-zephyr-eabi-objdump: DWARF error: could not find variable specification at offset 762
/home/gfbudke/zephyr-sdk-0.12.0.b/arm-zephyr-eabi/bin/arm-zephyr-eabi-objdump: DWARF error: could not find variable specification at offset 1a7b
/home/gfbudke/zephyr-sdk-0.12.0.b/arm-zephyr-eabi/bin/arm-zephyr-eabi-objdump: DWARF error: could not find variable specification at offset 4f86
/home/gfbudke/zephyr-sdk-0.12.0.b/arm-zephyr-eabi/bin/arm-zephyr-eabi-objdump: DWARF error: could not find variable specification at offset 4f93
/home/gfbudke/zephyr-sdk-0.12.0.b/arm-zephyr-eabi/bin/arm-zephyr-eabi-objdump: DWARF error: could not find variable specification at offset 4fa0
/home/gfbudke/zephyr-sdk-0.12.0.b/arm-zephyr-eabi/bin/arm-zephyr-eabi-objdump: DWARF error: could not find variable specification at offset 4fad
/home/gfbudke/zephyr-sdk-0.12.0.b/arm-zephyr-eabi/bin/arm-zephyr-eabi-objdump: DWARF error: could not find variable specification at offset 68e6
/home/gfbudke/zephyr-sdk-0.12.0.b/arm-zephyr-eabi/bin/arm-zephyr-eabi-objdump: DWARF error: could not find variable specification at offset 6900
/home/gfbudke/zephyr-sdk-0.12.0.b/arm-zephyr-eabi/bin/arm-zephyr-eabi-objdump: DWARF error: could not find variable specification at offset 68f3
/home/gfbudke/zephyr-sdk-0.12.0.b/arm-zephyr-eabi/bin/arm-zephyr-eabi-objdump: DWARF error: could not find variable specification at offset 691a
/home/gfbudke/zephyr-sdk-0.12.0.b/arm-zephyr-eabi/bin/arm-zephyr-eabi-objdump: DWARF error: could not find variable specification at offset 690d
/home/gfbudke/zephyr-sdk-0.12.0.b/arm-zephyr-eabi/bin/arm-zephyr-eabi-objdump: DWARF error: could not find variable specification at offset d29b
/home/gfbudke/zephyr-sdk-0.12.0.b/arm-zephyr-eabi/bin/arm-zephyr-eabi-objdump: DWARF error: could not find variable specification at offset d2a8
/home/gfbudke/zephyr-sdk-0.12.0.b/arm-zephyr-eabi/bin/arm-zephyr-eabi-objdump: DWARF error: could not find variable specification at offset 7193
/home/gfbudke/zephyr-sdk-0.12.0.b/arm-zephyr-eabi/bin/arm-zephyr-eabi-objdump: DWARF error: could not find variable specification at offset 2d2d
```

```
*** Booting Zephyr OS build zephyr-v2.4.0-1442-gbc4823a21183  ***
[00:00:00.377,502] <inf> mcuboot: Starting bootloader
[00:00:00.383,880] <inf> mcuboot: Primary image: magic=good, swap_type=0x1, copy_done=0x3, image_ok=0x3
[00:00:00.394,042] <inf> mcuboot: Secondary image: magic=unset, swap_type=0x1, copy_done=0x3, image_ok=0x3
[... 4 lines cut ...]

uuart:~$ 
[00:00:00.002,075] <inf> ieee802154_nrf5: nRF5 802154 radio initialized
[00:00:00.006,225] <err> os: ***** MPU FAULT *****
[00:00:00.006,225] <err> os:   Data Access Violation
[... cut ...]
[00:00:00.006,256] <err> os: Faulting instruction address (r15/pc): 0x0003edec
[00:00:00.006,286] <err> os: >>> ZEPHYR FATAL ERROR 0: CPU exception on CPU 0
[00:00:00.006,286] <err> os: Current thread: 0x20001ea8 (main)
[00:00:01.174,255] <err> os: Halting system
```

It is a fault inside memset.

Try build samples/net/sockets/echo_client and got same.

@rlubos , can you assist with this, please? What is the latest hash that you know is ok?

```
Zephyr bc6d16e422
OT f9fbfad0119d7b0d473eaac50c2f0a0a483c2823
```

nandojve, 2020-11-21:
@nehalrohit can you try ping `coap.updatehub.io`? Since you are using Tayga (NAT64) you will navigate, in fact, on IPv4, try this:

1- Get IPv4 from coap.updatehub.io
```
ping -4 coap.updatehub.io
PING coap.updatehub.io (159.89.225.65) 56(84) bytes of data.
```
2- Translate IPv4 to IPv6
```
https://findipv6.com/ipv4-to-ipv6/159.89.225.65
IPv4 address: 159.89.225.65
IPv4 Mapped IPv6 Address: ::ffff:9f59:e141
```
3- Create the OT IPv6 to be translated to IPv4
```
Prefix: 64:ff9b::
Sufix: 9f59:e141
results: 64:ff9b::9f59:e141
```
4- Try ping coap.updatehub.io from a Joiner
```
> ping 64:ff9b::9f59:e141
Done
> 16 bytes from 64:ff9b:0:0:0:0:9f59:e141: icmp_seq=8 hlim=52 time=173ms
```

If this works, you can configure DNS and UpdateHub as:
```
# Add Google 8.8.8.8 DNS Server
CONFIG_DNS_SERVER_IP_ADDRESSES=y
CONFIG_DNS_SERVER1="64:ff9b::808:808"

# Add UpdateHub Cloud
CONFIG_UPDATEHUB_SERVER="64:ff9b::9f59:e141"
CONFIG_UPDATEHUB_COAP_BLOCK_SIZE_EXP=2
```

Since UpdateHub uses sockets and you have internet access it should work.

nehalrohit, 2020-11-23:
Hi @nandojve 
I am able to ping coap.updatehub.io from a Joiner.
[image]

here this setting in prj.conf
# Add Google 8.8.8.8 DNS Server
CONFIG_DNS_SERVER_IP_ADDRESSES=y
CONFIG_DNS_SERVER1="64:ff9b::808:808"

# Add UpdateHub Cloud
CONFIG_UPDATEHUB_SERVER="64:ff9b::9f59:e141"
CONFIG_UPDATEHUB_COAP_BLOCK_SIZE_EXP=2

I am still getting a DNS resolver issue. anything else need to change?

nandojve, 2020-11-23:
> UpdateHub uses sockets so I'm assuming that works with OT.
> I try compile OT from current master and got:
>
> ```
> [1/500] cd /home/gfbudke/zephyros/mainline/zephyr/build/modules/openthread/build && echo [...]

This error happen with SDK 0.12.0-beta-2. When I switch back to 0.11.3 I didn't see anymore `objdump: DWARF error`
CC: @rlubos @galak

rlubos, 2020-11-23:
@nandojve 
> Try build samples/net/sockets/echo_client and got same.

I was not able to reproduce the crash on my side, are there any special conditions to trigger it? The radio driver was updated recently (which I'd suspect), but as said I fail to reproduce. Did you call `west update`? Perhaps the hal_nordic went out of sync with Zephyr on your side?

nandojve, 2020-11-23:
You can ignore the run crash. After switch to 0.11.3 I figure out. It was needed to increase RAM from main scope.

nandojve, 2020-11-23:
Hi @nehalrohit ,

By some reason the DNS name can not be resolved. It is not necessary a problem with UpdateHub.
I took a note to review that code.

I open my development branch `https://github.com/UpdateHub/zephyr/tree/topic/uhu_net-management`. It is working with OpenThread.

By default I'm using UpdateHub CE, so at **overlay-ot.conf**:
`CONFIG_UPDATEHUB_CE=n`
You need add your server IP:
`CONFIG_UPDATEHUB_SERVER="64:ff9b::xxxx:xxxx"`

Procedure o discover, example from google IPv4 DNS:
```
# https://www.ultratools.com/tools/ipv4toipv6Result?address=8.8.8.8
# Mapped Address: 0:0:0:0:0:ffff:808:808
# Extract value                 >808:808<
# Add prefix        64:ff9b::
CONFIG_DNS_SERVER1="64:ff9b::808:808"
```

For UpdateHub Cloud Server:
```
ping -4 coap.updatehub.io
PING coap.updatehub.io (159.89.225.65) 56(84) bytes of data.
Mapped Address: 0:0:0:0:0:ffff:9f59:e141
CONFIG_DNS_SERVER1="64:ff9b::9f59:e141"
```

retfie, 2020-11-23:
> Hello retfie,
> can you please suggest to me how you do OTA with Openthread or you can provide some steps to follow( link or code)?
>
> Thanks
> Nehal

HI Hehal,
sorry for late response, but here is the essential change that you need on lwm2m-client re-based on current zephyr master branch:

 https://github.com/retfie/zephyr/commit/3b9dd7281845c281106b2e85eca604f2665919a6

You need mcuboot bootloader on device and a working leshan server to register on. Then use 5/0/0 or 5/0/1 resource to send zephyr.signed.bin to lwm2m_client. Then look for status (5/0/3) to became 2 (ready to apply), then start update with Exec on /5/0/2.

nehalrohit, 2020-11-24:
Hi @nandojve 

I already try this configuration 
**CONFIG_DNS_SERVER1="64:ff9b::9f59:e141"**   still not work for me.  my joiner device can ping updatehub.io cloud server but the device cant show on the server.

**I open my development branch https://github.com/UpdateHub/zephyr/tree/topic/uhu_net-management. It is working with OpenThread.** 
if you successfully did then give me a step on how you do with nrf52840dk or code reference?

nandojve, 2020-11-26:
Hi @nehalrohit 

I confirmed here yesterday. The UpdateHub Cloud only accept connections with DTLS. I noted that when I enabled DTLS on Zephyr, for some reason, "send" returns with error and packages have been not sent. It seems something goes wrong at DTLS challenge phase.

I'll recommend, in the mean time that we are investigating the DTLS issue, you to use UpdateHub CE. You can pull the docker container and run it in your environment or at any service provider (digital ocean, vultr, etc) with docker support. The smallest machine is enough to run the service.

nehalrohit, 2020-11-28:
hi @nandojve  
I can see the device on UpdateHub CE. 
[image]

but updatehub run cmd not working

nehalrohit, 2020-11-28:
Hi @nandojve  
by mistake, the thread is closed my issue is still going on its not yet close

closed 2020-11-28 by nehalrohit

nehalrohit, 2020-11-28:
Hi @nandojve 
I have successfully  seen the device on the update server 
[image]

[image]

create rollout not work . 
[image]