zephyrproject-rtos/zephyr #29758: edtlib not reporting proper matching_compat for led nodes (and other children nodes)

galak, 2020-11-03:
For a DTS like (from frdm_k64f.dts):

```
        leds {
                compatible = "gpio-leds";
                red_led: led_0 {
                        gpios = < &gpiob 0x16 0x1 >;
                        label = "User LD1";
                };
                green_led: led_1 {
                        gpios = < &gpioe 0x1a 0x1 >;
                        label = "User LD2";
                };
                blue_led: led_2 {
                        gpios = < &gpiob 0x15 0x1 >;
                        label = "User LD3";
                };
        };
```

the `matching_compat` for `/leds/led_0` should be `gpio-leds` but its being reported as `None`

galak, 2020-11-03:
This is a regression introduced by:

commit 7165b77a81f35bd76fbbdc391f8c76f3d64d58a4 (refs/bisect/bad)
Author: Martí Bolívar <marti.bolivar@nordicsemi.no>
Date:   Mon Oct 12 12:32:25 2020 -0700

    scripts: edtlib: refactor for first class bindings

Causes issues for things like sanitycheck `dt_compat_enabled_with_alias("gpio-leds", "led0")` will always report False

pabigot, 2020-11-03:
> the `matching_compat` for `/leds/led_0` should be `gpio-leds` but its being reported as `None`

Is there documentation for this expectation?  The structure of a child node is certainly not equivalent to the structure of its parent, so it should not be given the same compatible.  I'm not seeing anything in the devicetree specification about this, nor in DT-Schema.

For the proposal in https://github.com/zephyrproject-rtos/zephyr/pull/28320#pullrequestreview-502138742 I definitely need a node with one compatible to have children that are another compatible.  I'm not sure I need them to be associated with a "child binding", but I do need to be able to iterate over the children.

In this specific case, if I were to iterate over nodes with compatible `gpio-leds`, what would I get?  Just the children?  Just the parent?  Both?

PR #29759 (2020-11-03): dt: child binding compatibles should match their parents
Fixes: #29758

See commit log in the second patch for details.
commit 0d4dca10 (2020-11-03): scripts: edtlib: child binding compatibles match parents

Fixes: #29758

Commit 7165b77a81f35bd76fbbdc391f8c76f3d64d58a4 ("scripts: edtlib:
refactor for first class bindings") introduced a Binding class.

Its child_binding instance attribute has a compatible which can be
None. Adjust this behavior so the child Binding object has the same
compatible as the parent binding which ultimately has a compatible.

Without this, sanitycheck's expr_parser is doing some matching on
compatibles in child nodes that is producing unexpected results.
closed 2020-11-03 by galak