zephyrproject-rtos/zephyr #29764: Add Docker based testing support to network samples

jukkar, 2020-11-03:
We have infrastructure to run network samples against a Docker container running Linux, see information at https://github.com/zephyrproject-rtos/zephyr/blob/master/scripts/net/README.txt. Currently echo-server/client and mqtt_publisher samples work with the `scripts/net/run-sample-tests.sh` script.

Collect here a list of network samples that work properly with the runner script so we can track the progress.

- [ ] dhcpv4_client
- [ ] dns_resolve
- [ ] gptp
- [ ] ipv4_autoconf
- [ ] lldp
- [ ] lwm2m_client
- [ ] mdns_responder
- [x] mqtt_publisher
- [ ] promiscuous_mode
- [ ] syslog_net
- [ ] telnet
- [ ] vlan
- [ ] big_http_download
- [ ] can
- [ ] civetweb
- [ ] coap_client
- [ ] coap_server
- [ ] dumb_http_server
- [ ] dumb_http_server_mt
- [x] echo_client
- [x] echo_server
- [ ] http_client
- [ ] http_get
- [ ] packet
- [ ] sntp_client
- [ ] websocket_client

Original issue at #19540

jukkar, 2020-11-09:
Short presentation about Docker based testing https://drive.google.com/file/d/1LYPcVOyHrJGUaD4wuAwZYlBq310jgVV_/view

Nukersson, 2020-11-21:
Please note #30175 for CivetWeb