zephyrproject-rtos/zephyr #29769: mimxrt1050_evk: build error at tests/subsys/usb/device/

Zhaoningx, 2020-11-04:
Describe the bug
running tests/subsys/usb/device/, and build erroe

To Reproduce
Steps to reproduce the behavior:
sanitycheck -p mimxrt1050_evk --device-testing --device-serial /dev/ttyACM1  -T tests/subsys/usb/device/ -vv
see error

*** Booting Zephyr OS build v2.4.0-rc1-11-g0d4dca10b2 ***
[ 29%] Building C object zephyr/CMakeFiles/zephyr.dir/lib/os/dec.c.obj
[ 29%] Building C object zephyr/CMakeFiles/zephyr.dir/lib/os/fdtable.c.obj
[ 30%] Building C object zephyr/CMakeFiles/zephyr.dir/lib/os/hex.c.obj
[ 31%] Building C object zephyr/CMakeFiles/zephyr.dir/lib/os/mempool.c.obj
[ 31%] Building C object zephyr/CMakeFiles/zephyr.dir/lib/os/notify.c.obj
[ 32%] Building C object zephyr/CMakeFiles/zephyr.dir/lib/os/printk.c.obj
[ 33%] Building C object zephyr/CMakeFiles/zephyr.dir/lib/os/onoff.c.obj
[ 33%] Building C object zephyr/CMakeFiles/zephyr.dir/lib/os/rb.c.obj
[ 34%] Building C object zephyr/CMakeFiles/zephyr.dir/lib/os/sem.c.obj
[ 35%] Building C object zephyr/CMakeFiles/zephyr.dir/lib/os/thread_entry.c.obj
[ 35%] Building C object zephyr/CMakeFiles/zephyr.dir/lib/os/timeutil.c.obj
[ 36%] Building C object zephyr/CMakeFiles/zephyr.dir/lib/os/work_q.c.obj
[ 37%] Building C object zephyr/CMakeFiles/zephyr.dir/lib/os/heap.c.obj
[ 37%] Building C object zephyr/CMakeFiles/zephyr.dir/lib/os/heap-validate.c.obj
[ 38%] Building C object zephyr/CMakeFiles/zephyr.dir/lib/os/prf.c.obj
[ 39%] Building C object zephyr/CMakeFiles/zephyr.dir/lib/os/assert.c.obj
[ 39%] Building C object zephyr/CMakeFiles/zephyr.dir/misc/generated/configs.c.obj
[ 40%] Building C object zephyr/CMakeFiles/zephyr.dir/soc/arm/nxp_imx/rt/soc.c.obj
In file included from /home/xuhao/work/zephyrproject/zephyr/soc/arm/nxp_imx/rt/soc.c:19:
/home/xuhao/work/zephyrproject/modules/hal/nxp/mcux/middleware/usb/device/./port/zephyr/usb_dc_mcux.h:336:2: error: unknown type name 'usb_device_callback_t'
  336 |  usb_device_callback_t deviceCallback;
      |  ^~~~~~~~~~~~~~~~~~~~~
/home/xuhao/work/zephyrproject/modules/hal/nxp/mcux/middleware/usb/device/./port/zephyr/usb_dc_mcux.h:338:2: error: unknown type name 'usb_device_endpoint_callback_struct_t'
  338 |  usb_device_endpoint_callback_struct_t
      |  ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
/home/xuhao/work/zephyrproject/modules/hal/nxp/mcux/middleware/usb/device/./port/zephyr/usb_dc_mcux.h:339:14: error: 'USB_DEVICE_CONFIG_ENDPOINTS' undeclared here (not in a function); did you mean 'USB_DEVICE_CONFIG_EHCI'?
  339 |   epCallback[USB_DEVICE_CONFIG_ENDPOINTS << 1U];
      |              ^~~~~~~~~~~~~~~~~~~~~~~~~~~
      |              USB_DEVICE_CONFIG_EHCI
gmake[2]: *** [zephyr/CMakeFiles/zephyr.dir/build.make:336: zephyr/CMakeFiles/zephyr.dir/soc/arm/nxp_imx/rt/soc.c.obj] Error 1
gmake[1]: *** [CMakeFiles/Makefile2:1010: zephyr/CMakeFiles/zephyr.dir/all] Error 2
gmake: *** [Makefile:84: all] Error 2

Environment (please complete the following information):
OS: Fedora28
Toolchain: Zephyr-sdk-0.11.3
Commit ID: g0d4dca10b2

Zhaoningx, 2020-11-04:
The tests/subsys/usb/bos has same issue with this issue.

jfischer-phytec-iot, 2020-11-04:
duplicating, closing in favor or #29710

closed 2020-11-04 by jfischer-phytec-iot