zephyrproject-rtos/zephyr #29775: TCP socket stream

liang8057, 2020-11-04:
Describe the bug
I set up a client socket to communicate with the remote server, and the server send five 1 byte packets. 
I want to recv 5 bytes at a time, but there is no way. I need to call recv 5 times to receive the packets.

To Reproduce
struct sockaddr_in info;
int sock, ret;
char buf[100];

sock = socket(PF_INET, **SOCK_STREAM**, IPPROTO_TCP);
memset(&info, 0, sizeof(info));
info.sin_family = PF_INET;

info.sin_addr.s_addr = inet_addr("192.168.127.5");
info.sin_port = htons(8700);

connect(sock, (struct sockaddr *)&info, sizeof(info));

k_sleep(K_SECONDS(10));

ret = recv(sock, buf, 5, 0);
printf("ret %d\n", ret);

The server transmits the five packets in 10 seconds, and then I start to receive.
But I found that I had to call recv 5 times to receive all the packets.

**Environment (please complete the following information):**
 - OS: Zephyr2.4
 - Toolchain: zephyr-sdk-0.11.2

jukkar, 2020-11-04:
This is very typical in TCP, if the peer sends us things one byte a time then we receive each byte in separate TCP segments. We do not collect the packets anywhere in receive path so you will also receive things one byte a time. This is not a bug.

liang8057, 2020-11-05:
But I tested on Linux under the same situation, and the result is that there is unnecessary to recv five times.

jukkar, 2020-11-05:
> But I tested on Linux under the same situation, and the result is that there is unnecessary to recv five times.

Sure, but in Linux there is queueing in receive side which collects smaller packets into larger ones before giving them to the application. Zephyr does not do it currently because it requires memory and we try to limit the memory usage.

closed 2021-01-19 by github-actions[bot]