zephyrproject-rtos/zephyr #29779: [Doc] How to link dts bindings with Kconfig flags

erwango, 2020-11-04:
**Is your enhancement proposal related to a problem? Please describe.**
DTS bindings are now available in zephyr documentation (https://docs.zephyrproject.org/latest/reference/devicetree/bindings)
and maintainers have a way to document these important pieces of driver configuration.
The last important missing piece of information is to indicate how a driver should be enabled, operation typically achieved using Kconfig flags.

This piece of info would would be missing to automatize generation of supported features in boards documentation (cf https://github.com/zephyrproject-rtos/zephyr/pull/29652/).

One easy way to get this link would be to document drivers matching Kconfig flag in dts bindings flag directly.
But, as said @mbolivar-nordic: "kind of defeats the 'OS independent' goal for bindings IMO".

So I'm raising this issue in order to reach an agreement on how to document dts bindings Kconfig relation w/o breaking OS independent promises.

erwango, 2020-11-04:
@mbolivar-nordic, @MaureenHelm, @galak

mbolivar-nordic, 2020-11-04:
I did some prototyping on this last night. So far I have a way for drivers to declare (via Doxygen) what devicetree compatibles they support, and build an index mapping compatible <-> driver from that.

Next steps are to generate documentation pages for these drivers and include the detailed description (again from Doxygen) into the Sphinx pages. I verified that the usual ways of embedding restructured text work. That allows us to say something like this in a driver file comment:

```
/**
 * @file
 * @dtcompatible{vnd,foo}
 * This the Foo driver for Vnd devices. To enable this driver, set :option:`CONFIG_FOO` ...
 */
```

And the `:option:` part provides a link to the existing Kconfig index.

I will post a proof of concept sometime this week.

mbolivar-nordic, 2020-11-04:
cc @pabigot

erwango, 2020-11-05:
> I did some prototyping on this last night. So far I have a way for drivers to declare (via Doxygen) what devicetree compatibles they support, and build an index mapping compatible(s) <-> driver(s) from that.

I'm fine with the approach! Txs @mbolivar-nordic.
One tiny reservation is that it will rely purely on developer and reviewers. Maybe and automated check could be added, or we can get the doc generation could fail if not present ?

mbolivar-nordic, 2020-11-05:
> One tiny reservation is that it will rely purely on developer and reviewers. Maybe and automated check could be added, or we can get the doc generation could fail if not present ?

Sure, I'm happy to discuss and add CI checks like that depending on what subsystem maintainers decide should be required for driver documentation.

mbolivar-nordic, 2020-11-18:
Sorry for the delay, but here is a draft: https://github.com/zephyrproject-rtos/zephyr/pull/30104

There is a new "device drivers" index that pulls in Doxygen content from driver sources. It's at top level and is available on the sidebar.

mbolivar-nordic, 2020-12-08:
> Sorry for the delay, but here is a draft: #30104

any comments on this?

erwango, 2020-12-09:
It's in my TODO list, sorry for the delay.

erwango, 2024-02-09:
@kartben Feel free to close if this looks as a duplicate or if direction is not the good one

closed 2025-05-26 by erwango