zephyrproject-rtos/zephyr #29780: nRF SDK hci_usb sample disconnects after 40 seconds with extended connection via coded PHY

kasapreyizz, 2020-11-04:
**Describe the bug**
hci_usb sample with following configuration always disconnects after 40 seconds.

**To Reproduce**
Steps to reproduce the behavior:
1. Build the hci_usb sample in v1.3.1 in nrf-sdk with following prj.conf

```
CONFIG_STDOUT_CONSOLE=y
CONFIG_GPIO=y
CONFIG_SERIAL=y
CONFIG_UART_INTERRUPT_DRIVEN=y

CONFIG_BT=y
CONFIG_BT_HCI=y

CONFIG_USB=y
CONFIG_USB_DEVICE_STACK=y
CONFIG_USB_DEVICE_BLUETOOTH=y
CONFIG_USB_DEVICE_BLUETOOTH_VS_H4=y

CONFIG_MAIN_STACK_SIZE=1024
CONFIG_SYSTEM_WORKQUEUE_STACK_SIZE=1024

CONFIG_BT_CTLR_RX_BUFFERS=18
CONFIG_BT_CTLR_TX_BUFFERS=19
CONFIG_BT_CTLR_TX_BUFFER_SIZE=251
CONFIG_BT_CTLR_DATA_LENGTH_MAX=251
CONFIG_BT_HCI_CMD_COUNT=20
CONFIG_BT_RX_BUF_COUNT=20
CONFIG_BT_RX_BUF_LEN=258
CONFIG_BT_DISCARDABLE_BUF_SIZE=257
CONFIG_BT_MAX_CONN=10
CONFIG_USB_REQUEST_BUFFER_SIZE=384

CONFIG_BT_EXT_ADV=y
CONFIG_BT_CTLR_PHY_CODED=y
CONFIG_BT_LLL_VENDOR_NORDIC=y
CONFIG_BT_LL_NRFXLIB=y
```

2. Connect the nrf52840-dongle flashed with following code to a linux host (my host is Rpi3 with kernel version 5.4.51-v7+

3. Command to make an extended connection via hcitool which is similar to: hcitool cmd 0x08 0x0043 0x00 0x01 0x00 0x00 0xSS 0xRE 0xDD 0xCA 0xMA 0x04 0x0000 0x0001 0x0000 0x0001 0x0006 0x0000 0x18 0x00 0x00 0x00 0x80 0x0C 0x00 0x00 0x00 0x0047

**Expected behavior**
After commanding, following btmon output is taken and client device is indicating that an connection has been established.

```
< HCI Command: LE Extended C.. (0x08|0x0043) plen 26  #2798 [hci0] 12808.706224
        Filter policy: White list is not used (0x00)
        Own address type: Random (0x01)
        Peer address type: Public (0x00)
        Peer address: 00:85:C0:06:F8:9B (OUI 00-85-C0)
        Initiating PHYs: 0x04
        Entry 0: LE Coded
          Scan interval: 160.000 msec (0x0100)
          Scan window: 160.000 msec (0x0100)
          Min connection interval: 7.50 msec (0x0006)
          Max connection interval: 30.00 msec (0x0018)
          Connection latency: 0 (0x0000)
          Supervision timeout: 32000 msec (0x0c80)
          Min connection length: 0.000 msec (0x0000)
          Max connection length: 11360.000 msec (0x4700)
> HCI Event: Command Status (0x0f) plen 4             #2799 [hci0] 12808.707731
      LE Extended Create Connection (0x08|0x0043) ncmd 1
        Status: Success (0x00)
< HCI Command: Disconnect (0x01|0x0006) plen 3        #2800 [hci0] 12808.707963
        Handle: 0
        Reason: Remote User Terminated Connection (0x13)
[... 1139 lines cut ...]
          Static Address
@ MGMT Event: New Settings (0x0006) plen 4             {0x0002} [hci0] 5.410056
        Current settings: 0x00008a11
          Powered
          Bondable
          Low Energy
          Secure Connections
          Static Address
@ RAW Close: hciconfig                                        {0x0003} 5.410926
> HCI Event: Command Complete (0x0e) plen 4                 #48 [hci0] 5.411952
      LE Set Extended Advertising Data (0x08|0x0037) ncmd 1
        Status: Success (0x00)
```

**Impact**
Showstopper for my project.

**Environment (please complete the following information):**
 - HCI Device: nrf52840 Dongle and nrf52840-DK (PCA10059 and PCA10056)
 - Host Device: Raspberry Pi 3 B+
 - Host OS: Linux Raspbian, Kernel version: 5.4.51-v7+
 - Toolchain: nRF Connect SDK (Zephyr OS with nrfxlib)
 - HCI device and clients' connection parameters are exactly consistent. 

Any help will appreciated, thanks in advance.

joerchan, 2020-11-04:
@kasapreyizz Can you expand on the setup, which device is master, which device is slave. Which side is the one disconnecting.
In your logs you have both sides of the disconnect reason. It appears that the master dropped the connection because an LL procedure did not complete within the timeout (40 seconds, and disconnect reason: LMP Response Timeout / LL Response Timeout (0x22))

If you have the full logs from both sides can you attach them as files in full (zipped) to the issue?

kasapreyizz, 2020-11-04:
Hello there,

I got an nrf52dongle programmed with hci_usb sample. The master is nrf52dongle and it establish the connections via HCI commands at host side. Here we got somethink weird, with legacy scanning and connection settings, that disconnects doesnt happen. But with extended scanning and enhanced connection it disconnects. And it disconnects every single extended device and with all initiating PHYs. So I think that the problem at master.

 When no data is exchanged between initiator and acceptor the disconnect reason looks 0x13 and 0x16 respectively. But when I connect them via hcitool and than a 6lowpan tcp connection is established disconnect happens and that time reason  LMP / LL Response Timeout. 

I have logged full of HCI_USB side of my setup in original entry and my IPSP device does not have any debug option. If I attach them again, they mostly look like same. But I know that IPSP device have not any problem since an mobile device can connect it without connection lost issue (via Coded PHY) and I tried much samples to connect with my hci_usb device and all of them I got disconnected issue. 

Thanks in advance.

joerchan, 2020-11-04:
The 0x13/0x16 I'm not so sure about.
```
> HCI Event: Disconnect Complete (0x05) plen 4        #2809 [hci0] 12837.631685
        Status: Success (0x00)
        Handle: 0
        Reason: Connection Timeout (0x08)
@ MGMT Event: Device Disconnected (0x000c) plen 8  {0x0002} [hci0] 12837.631741
        LE Address: 00:85:C0:06:F8:9B (OUI 00-85-C0)
        Reason: Connection timeout (0x01)
@ MGMT Event: Device Disconnected (0x000c) plen 8  {0x0001} [hci0] 12837.631741
        LE Address: 00:85:C0:06:F8:9B (OUI 00-85-C0)
        Reason: Connection timeout (0x01)
< HCI Command: Disconnect (0x01|0x0006) plen 3        #2810 [hci0] 12839.696163
        Handle: 1
        Reason: Remote User Terminated Connection (0x13)
> HCI Event: Command Status (0x0f) plen 4             #2811 [hci0] 12839.698692
      Disconnect (0x01|0x0006) ncmd 1
        Status: Success (0x00)
> HCI Event: Disconnect Complete (0x05) plen 4        #2812 [hci0] 12839.739683
        Status: Success (0x00)
        Handle: 1
        Reason: Connection Terminated By Local Host (0x16)
```

As you can see here, it is first disconnected with reason 0x08 (timeout). And then for some reason there is a Disconnect command, and a disconnect complete event with a different reason.

Can you give me an example on connecting two samples that would result in this disconnect problem? Since we don't have access to your IPSP device.

kasapreyizz, 2020-11-04:
I just tried now with peripheral_long_range sample which provided by nordic.
For reproduce that error: I got ncs 1.3.1 hci_usb sample on nrf52dongle with the prj.conf in original post
And I just flashed another nrf52dk with peripheral_long_range sample. There is no data exchange between them but it's still disconnects. 

Here is the btmon full log: 

Scanning device:
```
< HCI Command: LE Set Extende.. (0x08|0x0041) plen 9  #3157 [hci0] 88438.012359
        Own address type: Random (0x01)
        Filter policy: Accept all advertisement (0x00)
        PHYs: 0x04
        Entry 0: LE Coded
          Type: Active (0x01)
          Interval: 160.000 msec (0x0100)
          Window: 160.625 msec (0x0101)
> HCI Event: Command Complete (0x0e) plen 4           #3158 [hci0] 88438.013932
      LE Set Extended Scan Parameters (0x08|0x0041) ncmd 1
        Status: Success (0x00)
@ RAW Close: hcitool                               {0x0003} [hci0] 88438.014104
@ RAW Open: hcitool (privileged) version 2.22             {0x0003} 88443.214873
@ RAW Close: hcitool                                      {0x0003} 88443.214933
@ RAW Open: hcitool (privileged) version 2.22             {0x0003} 88443.215011
@ RAW Close: hcitool                                      {0x0003} 88443.215044
@ RAW Open: hcitool (privileged) version 2.22      {0x0003} [hci0] 88443.215150
< HCI Command: LE Set Extende.. (0x08|0x0042) plen 6  #3159 [hci0] 88443.215444
        Extended scan: Enabled (0x01)
        Filter duplicates: Disabled (0x00)
        Duration: 0 msec (0x0000)
        Period: 0.00 sec (0x0000)
> HCI Event: Command Complete (0x0e) plen 4           #3160 [hci0] 88443.216913
      LE Set Extended Scan Enable (0x08|0x0042) ncmd 1
        Status: Success (0x00)
@ RAW Close: hcitool                               {0x0003} [hci0] 88443.217083
> HCI Event: LE Meta Event (0x3e) plen 46             #3161 [hci0] 88443.344915
      LE Extended Advertising Report (0x0d)
        Num reports: 1
        Entry 0
          Event type: 0x0001
            Props: 0x0001
              Connectable
            Data status: Complete
          Address type: Random (0x01)
          Address: EF:85:C0:06:F8:9B (Static)
          Primary PHY: LE Coded
          Secondary PHY: LE Coded
          SID: 0x00
          TX power: 127 dBm
          RSSI: -27 dBm (0xe5)
          Periodic advertising invteral: 0.00 msec (0x0000)
          Direct address type: Reserved (0xff)
          Direct address: 00-00-00-00-00-00
          Data length: 0x14
        02 01 06 10 09 4c 6f 6e 67 20 72 61 6e 67 65 20  .....Long range 
        64 65 6d 6f                                      demo            

```

Connecting device and disconnect: 
```

< HCI Command: LE Set Extende.. (0x08|0x0042) plen 6  #3177 [hci0] 88447.478143
        Extended scan: Disabled (0x00)
        Filter duplicates: Disabled (0x00)
        Duration: 0 msec (0x0000)
        Period: 0.00 sec (0x0000)
> HCI Event: Command Complete (0x0e) plen 4           #3178 [hci0] 88447.479904
      LE Set Extended Scan Enable (0x08|0x0042) ncmd 1
        Status: Success (0x00)
@ RAW Close: hcitool                               {0x0003} [hci0] 88447.480085
[... 764 lines cut ...]
        Num handles: 1
        Handle: 0
        Count: 1
> ACL Data RX: Handle 0 flags 0x02 dlen 12            #3587 [hci0] 89245.576365
      LE L2CAP: LE Flow Control Credit (0x16) ident 65 len 4
        Source CID: 64
        Credits: 1
> ACL Data RX: Handle 0 flags 0x02 dlen 12            #3588 [hci0] 89245.605240
      LE L2CAP: LE Flow Control Credit (0x16) ident 67 len 4
        Source CID: 64
        Credits: 1
> HCI Event: Disconnect Complete (0x05) plen 4        #3589 [hci0] 89264.382819
        Status: Success (0x00)
        Handle: 0
        Reason: LMP Response Timeout / LL Response Timeout (0x22)
@ MGMT Event: Device Disconnected (0x000c) plen 8  {0x0002} [hci0] 89264.382889
        LE Address: 00:85:C0:06:F8:9B (OUI 00-85-C0)
        Reason: Unspecified (0x00)
@ MGMT Event: Device Disconnected (0x000c) plen 8  {0x0001} [hci0] 89264.382889
        LE Address: 00:85:C0:06:F8:9B (OUI 00-85-C0)
        Reason: Unspecified (0x00)
```

With all of those samples, I can reproduce the problem. 
Regards.

joerchan, 2020-11-04:
@kasapreyizz Can you check the nRF5 SDK sample. I suspect these two events are not replied to:
BLE_GAP_EVT_DATA_LENGTH_UPDATE_REQUEST
BLE_GAP_EVT_PHY_UPDATE_REQUEST

The should be replied to with sd_ble_gap_data_length_update and sd_ble_gap_phy_update respectively:

```
        case BLE_GAP_EVT_DATA_LENGTH_UPDATE_REQUEST:
        {
            ble_gap_data_length_params_t dl_params;

            // Clearing the struct will effectively set members to @ref BLE_GAP_DATA_LENGTH_AUTO.
            memset(&dl_params, 0, sizeof(ble_gap_data_length_params_t));
            UNUSED_RETURN_VALUE(sd_ble_gap_data_length_update(p_ble_evt->evt.gap_evt.conn_handle, &dl_params, NULL));
            break;
        }

        case BLE_GAP_EVT_PHY_UPDATE_REQUEST:
        {
            ble_gap_phys_t const phys =
            {
                .rx_phys = BLE_GAP_PHY_AUTO,
                .tx_phys = BLE_GAP_PHY_AUTO,
            };

            UNUSED_RETURN_VALUE(sd_ble_gap_phy_update(p_ble_evt->evt.gap_evt.conn_handle, &phys);
         }
```

kasapreyizz, 2020-11-06:
@joerchan My SDK sample replies those events but my problem is stay still. I double checked my connection parameters and they are completely same. Actually all of the nrf sdk examples produces that error. But I suspect PDU sizes. Does any minimal requirements in PDU sizes in extended scanning and enhanced connections? Or could my problem may rooted from somethink else ? I will try making enhanced connection with an Android phone and see if it's disconnects too, If I succeed in doing this. 

Thanks in advance.

joerchan, 2020-11-06:
@kasapreyizz Ok, since this is related the nRF5 SDK samples then you should open up an issue on nordic devzone instead, and close this one.
They will help you out there.

kasapreyizz, 2020-11-06:
@joerchan If I disconnect from an Android device too via my hci_usb sample, that proves that problem is at the Zephyr / nrfsdk side actually. If it wouldn't be a problem, could this issue still open while I am testing that case ?

carlescufi, 2020-11-10:
@kasapreyizz as @joerchan said, you are using the SoftDevice Controller, which means this is not applicable for the Zephyr Project. Please open an issue in https://devzone.nordicsemi.com/

closed 2020-11-10 by carlescufi