zephyrproject-rtos/zephyr #29782: smp_svr: Flashing zephyr.signed.bin does not seem to work on nrf52840dk

de-nordic, 2020-11-04:
**Describe the bug**
The bug has been identified with nrf52840dk_nrf52840 platform, but may be reproducible with other platforms.

The documentation for the sample (https://docs.zephyrproject.org/latest/samples/subsys/mgmt/mcumgr/smp_svr/README.html) provides instruction on how to flash build image to device (https://docs.zephyrproject.org/latest/samples/subsys/mgmt/mcumgr/smp_svr/README.html#flashing-the-sample-image); unfortunately after flashing the application is not bootable.

**To Reproduce**
Follow the steps of instruction. After "Flashing the sample image" step connect to device via UART, attached to console within Zephyr,  and reset device to obtain the console dump similar to the provided below.

**Expected behavior**
Following sample documentation should result in fully bootable and working application.
Either the `west flash` command needs to be fixed or the documentation should be changed to use hex-file instead of bin-file.

**Impact**
Users can not rely on documentation sample to successfully build and test unmodified sample.

**Logs and console output**
UART Console dump after reset of the device, where sample instruction has been followed step by step; the proper hardware platform has been selected for all steps:
```
** Booting Zephyr OS build zephyr-v2.4.0-1231-gc31148b04cc9  ***
[00:00:00.256,317] <inf> mcuboot: Starting bootloader
[00:00:00.262,847] <inf> mcuboot: Primary image: magic=unset, swap_type=0x1, copy_done=0x3, image_ok=0x3
[00:00:00.273,254] <inf> mcuboot: Secondary image: magic=unset, swap_type=0x1, copy_done=0x3, image_ok=0x3
[00:00:00.283,813] <inf> mcuboot: Boot source: none
[00:00:00.289,367] <wrn> mcuboot: Failed reading image headers; Image=0
[00:00:00.297,088] <err> mcuboot: Image in the primary slot is not valid!
[00:00:00.304,901] <err> mcuboot: Unable to find bootable image
```
**Environment (please complete the following information):**
zephyrproject-rtos/Zephyr c31148b04cc99d3a590fc72adffdd168cb31f81c, vanilla

**Additional context**
The `zephyr.signed.bin` is built correctly and works, when uploaded via mcumgr, so the problem is that `west flash` cannot correctly flash file in such format, at least via `nrfjprog` back-end.

utzig, 2020-11-04:
Does `nrfjprog` even support binaries? I've never been able to use it that way, only with hex files. I flashed the frdm_k64f with binaries last week so at least with pyocd it works correctly.

de-nordic, 2020-11-04:
I do not think that the `nrfjprog` works with bin files.
The sample documentation should allow user to build and run application; the flash command should be either universal or better described; the user should not end up figuring out why the sample does not work.
West flash, on bin file, with `nrfjprog` does not return any error, and console output, from `west flash`, suggest that everything has been flashed correctly, even progress of flashing is visible.
If all platforms can be correctly flashed with hex file, then the documentation should be changed to use hex. If not, then documentation requires update to point that out.

closed 2021-01-24 by github-actions[bot]