zephyrproject-rtos/zephyr #29790: The zephyr-app-commands macro does not honor :generator: option

jukkar, 2020-11-04:
I created documentation where I need to use `make` to compile zephyr.
I used this
```
.. zephyr-app-commands::
   :zephyr-app: samples/net/sockets/echo_server
   :host-os: unix
   :generator: make
   :board: qemu_x86
   :build-dir: server
   :gen-args: -DOVERLAY_CONFIG=overlay-qemu_802154.conf
   :goals: server
   :compact:
```
This generated following output
```
west build -b qemu_x86 -d build/server samples/net/sockets/echo_server -- -DOVERLAY_CONFIG=overlay-qemu_802154.conf
```
and `ninja` is used instead of `make`.
I managed to get `make` to be used by having these options:
```
.. zephyr-app-commands::
   :zephyr-app: samples/net/sockets/echo_server
   :host-os: unix
   :board: qemu_x86
   :build-dir: server
   :gen-args: -G'Unix Makefiles' -DOVERLAY_CONFIG=overlay-qemu_802154.conf
   :goals: server
   :compact:
```
This generated following output which uses `make` as expected
```
west build -b qemu_x86 -d build/server samples/net/sockets/echo_server -- -G'Unix Makefiles' -DOVERLAY_CONFIG=overlay-qemu_802154.conf
```

The documentation of `zephyr-app-commands` in `doc/extensions/zephyr/application.py` says that the `:generator:` option should work.

carlescufi, 2020-11-05:
I will take a look at it.

nashif, 2021-01-30:
@carlescufi  can you please take a look?

carlescufi, 2021-02-02:
> @carlescufi can you please take a look?

Looking at this now @nashif

PR #31927 (2021-02-03): doc: extensions: Fix generator being ignored in west
The generator option was being ignored in the west build instructions.
Include it now for the case of Makefiles, since west defaults to ninja
on all platforms.

Fixes #29790.
commit cf946d33 (2021-02-03): doc: extensions: Fix generator being ignored in west

The generator option was being ignored in the west build instructions.
Include it now for the case of Makefiles, since west defaults to ninja
on all platforms.

Fixes #29790.
closed 2021-02-03 by nashif