zephyrproject-rtos/zephyr #29796: Zephyr API for writing to flash for STM32G474 doesn't work as expected

azeemshatp, 2020-11-04:
**Describe the bug**
I have been trying to use the zephyr API for writing to flash (STM32G474) as below, which never worked.

    const struct device *flash_dev = device_get_binding(DT_CHOSEN_ZEPHYR_FLASH_CONTROLLER_LABEL);
    if (!flash_dev) {
        printk("Flash: Device driver not found.\n");
    }
    flash_write_protection_set(flash_dev, false);
    flash_erase(flash_dev, ota_status_address - CONFIG_FLASH_BASE_ADDRESS, FLASH_PAGE_SIZE);
    flash_write_protection_set(flash_dev, false);
    uint32_t *data = (uint32_t *)&status;
    uint32_t address = ota_status_address - CONFIG_FLASH_BASE_ADDRESS;
    flash_write(flash_dev, address, data, FLASH_PAGE_SIZE);
    flash_write_protection_set(flash_dev, true); // enable is recommended
	
It finds the device, but later enters the below point:
https://github.com/zephyrproject-rtos/zephyr/blob/v2.4-branch/drivers/flash/flash_stm32.c#L125

**Logs and console output**
The code, LOG_ERR("Status: 0x%08x", FLASH_STM32_REGS(dev)->SR & error) prints 
"Status: 0x000000a0"

**Environment (please complete the following information):**
 - OS: Windows
 - Toolchain: Zephyr
 - Commit SHA or Version used: v2.4-branch

ABOSTM, 2020-11-05:
Hi @azeemshatp,
I tested on my nucleo_g747re  `samples/drivers/flash_shell/` and it is working fine with shell command `flash erase 0x1F800 2048`
I also tested the following code, and it is working fine (data are properly written to flash 0x0801F800)
```
/*
 * Copyright (c) 2017 Linaro Limited
 *
 * SPDX-License-Identifier: Apache-2.0
 */
#include <stdlib.h>
#include <string.h>
#include <zephyr.h>
#include <sys/printk.h>
#include <logging/log.h>
#include <shell/shell.h>
#include <drivers/flash.h>
#include <device.h>
#include <soc.h>
#include <stdlib.h>

static const struct device *flash_device;

uint8_t data[2048];

void main(void)
{
	int ret;
	off_t offset = 0x1F800;
	size_t size = 2048;

	flash_device =
		device_get_binding(DT_CHOSEN_ZEPHYR_FLASH_CONTROLLER_LABEL);
	if (flash_device) {
		printk("Found flash controller %s.\n",
			DT_CHOSEN_ZEPHYR_FLASH_CONTROLLER_LABEL);
	} else {
		printk("**No flash controller found!**\n");
	}

	for(int i = 0; i < 2048; i++) {data[i]= i;}

	ret = flash_write_protection_set(flash_device, false);
	if (ret) {
		printk("Failed to disable flash protection (err: %d)."
				"\n", ret);
		return;
	}
	ret = flash_erase(flash_device, offset, size);
	if (ret) {
		printk("flash_erase failed (err:%d).\n", ret);
		return;
	}

	flash_write(flash_device, offset, &data, FLASH_PAGE_SIZE);
	if (ret) {
		printk("flash_write failed (err:%d).\n", ret);
		return;
	}

	ret = flash_write_protection_set(flash_device, true);
	if (ret) {
		printk("Failed to enable flash protection (err: %d)."
				"\n", ret);
		return;
	}

	printk("Erase OK");

}
```

ABOSTM, 2020-11-05:
Could you share an application, not just a snippet of code, but a complete code, to reproduce the issue.
Please make it as simple as possible.
So that I can reproduce and debug.
Thanks

azeemshatp, 2020-11-06:
Thanks for the reply. I just used your code in my main(). I can still see the below error during flash_erase():
	E: Status: 0x000000a0 -> seems to be programming sequence and alignment error?
	E: Status: 0x00000080 -> Prog. sequence error?
	E: erase 0000003f failed
	flash_erase failed (err:-5).

I will try to make a shareable code to test this. I do not know if there is anything else that I need to be careful about. Below is my flash partition. 

&flash0 {
	partitions {
		compatible = "fixed-partitions";
		#address-cells = <1>;
		#size-cells = <1>;

		slot0_partition: partition@8000000 {
			label = "image-0";
			reg = <0x08000000 0x1F800>;
		};

		slot1_partition: partition@8020000 {
			label = "image-1";
			reg = <0x08020000 0x1F800>;
		};

		image_header_partition: partition@801F000 {
			label = "image-header";
			reg = <0x801F000 0x800>;
		};

		ota_status_partition: partition@801F800 {
			label = "ota-status";
			reg = <0x0801F800 0x800>;
		};
	};
};

ABOSTM, 2020-11-06:
It looks like an overlap:
slot0_partition: From 0x08000000 to 0x0801F800
and
image_header_partition: From 0x0801F000 0x0x0801F800>;

Can you try with my code and with default nucelo_g474re flash partition ?

azeemshatp, 2020-11-06:
Actually, this overlap didn't really matter, and it gives me the same response. Yes, I tried this with the default nucelo_g474re flash partition

ABOSTM, 2020-11-06:
So I don't catch the difference between my setup which is working and your setup which is not working ?

azeemshatp, 2020-11-06:
I can now confirm that your code actually works on my nucleo G431 board. Just copied the code to the main() of samples/drivers/flash_shell/.  It prints, 

Found flash controller FLASH_CTRL.
Flash I/O commands can be run.
Erase OK

So I think my issue need to be now looked at my custom board configurations.

azeemshatp, 2020-11-09:
I finally found the issue with my board configuration that prevented the flash write. I was actually missing the below in the Kconfig.

if ARM_MPU
config MPU_ALLOW_FLASH_WRITE
	def_bool y
endif

I can now see the flash erase happening successfully. Thanks a lot for helping this out.

erwango, 2020-11-09:
Thanks @azeemshatp for updating us. Don't hesitate to close the PR if we're done.
@alexanderwachter I guess this is not the same issue on your side?

azeemshatp, 2020-11-09:
Closing the issue. Thanks again

closed 2020-11-09 by azeemshatp

alexanderwachter, 2020-11-09:
@erwango Nope. Mine is sporadic and only happen during excessive testing.