zephyrproject-rtos/zephyr #29798: test spi loopback fails on nucleo_f746zg

avigreen1978, 2020-11-04:
**Describe the bug**
When running the test in test/drivers/spi/spi_loopback on nucleo_f746zg, the test fails.
I tried running it on nucleo_f411re and nucleo_f413zh also, both pass.
I tried those specifically since they're both using dma v1, same as f746zg.

**To Reproduce**
Steps to reproduce the behavior:
1. go to tests/drivers/spi/spi_loopback
2. in the boards subdirectory, create a nucleo_f746zg.overlay file with the same content as nucleo_f411re.overlay
3. copy the contents of nucleo_f411re.conf to nucleo_f476zg.conf
4. run west build -b nucleo_f746zg
5. west flash
6. see the test fail (buffer contents are different - only zeros instead of same content).

**Expected behavior**
Test passes

**Environment (please complete the following information):**
 - OS: Linux
 - Toolchain: Zephyr SDK

**Additional context**
From my debugging so far, it seems to be a timing issue. If I run this without optimizations at all (not the default), sometimes it works. If I use default optimization and add LOG_INF anywhere, it also works.
I'll update if there are any more updates from my debugging.

erwango, 2020-11-04:
@avigreen1978 Txs for reporting this bug. Can you try to play with SPI pin slew-rate to see if this has an impact ?
nucleo_f411re and nucleo_f413zh are somewhat faster than nucleo_f746zg, this might explain different behavior.

avigreen1978, 2020-11-04:
I tried to raise the slew rate for all pins to very-high-speed for all except sck which default is very-high-speed, didn't help. (also tried high-speed for all except sck, didn't help).

avigreen1978, 2020-11-05:
Update: the bug is related to the enabling of the D-Cache in f7 soc.c init function.
For some reason, when D-Cache is disabled everything works perfectly.

ABOSTM, 2020-11-05:
Hi @avigreen1978,
The problem is probably cache coherency: because DMA 'bypass' Cache mechanism.
See some explanation here: https://community.st.com/s/question/0D50X00009XkhUW/hint-dma-and-cache-coherency
See also AN4839 https://www.st.com/content/ccc/resource/technical/document/application_note/group0/08/dd/25/9c/4d/83/43/12/DM00272913/files/DM00272913.pdf/jcr:content/translations/en.DM00272913.pdf

erwango, 2020-11-05:
@ABOSTM Txs.
So, as mentioned in these links, we have 2 options then:
- Deal with cache coherency on rx/tx (as done in i2s_ll_stm32.c)
  - +: Quick and easy to implement
  - -: Depends on handling of __DCACHE_PRESENT which doesn't look quite well done today
- Set dma buffers on DTCM (as done in eth_stm32_hal.c)
  - +: Automatically handled
  - -: Harder to implement and requires all impacted boards to declare the DTCM zone

ABOSTM, 2020-11-30:
This issue is fixed by #27911

FRASTM, 2021-03-01:
The same test also fails on nucleo stm32f767zi.
The irq of the Transfer complete for the Rx channel never comes.

FRASTM, 2021-03-03:
More than  #27911  the test config requires the CONFIG_SPI_STM32_UNALIGNED_DMA=y
for example in a `tests/drivers/spi/spi_loopback/boards/nucleo_f746zg.conf`

closed 2021-05-17 by github-actions[bot]