zephyrproject-rtos/zephyr #29813: BLE Range delay

a1ien, 2020-11-05:
**Is your enhancement proposal related to a problem? Please describe.**
On Core 5.2 Vol 6, Part B 4.2.3 describe Range delay. But on zephyr there is no way to set this delay.

**Describe the solution you'd like**
Can we add configure for setting this delay

**Describe alternatives you've considered**
Maybe allow change this value in runtime?

**Additional context**
We try to test long range with coded phy 2km distance and got disconnect around 600m on 80db.

P.S. 
I correctly understood that to set this delay, we must increase EVENT_IFS_MAX_US?

cvinayak, 2020-11-09:
Take a look at this: https://github.com/zephyrproject-rtos/zephyr/blob/master/subsys/bluetooth/controller/ll_sw/nordic/lll/lll_tim_internal.h#L13

Zephyr open source controller does not support connecting over long range using extended advertising, hence, please let me know how you are testing?, are you connecting over short range 1M PHY then switching to coded S8 PHY  before moving beyond 600m?

Please provide the Zephyr commit hash you are on, and the procedure to reproduce your observations.

a1ien, 2020-11-09:
> are you connecting over short range 1M PHY then switching to coded S8 PHY before moving beyond 600m?

Yes.

We use 45c8c51a627567f864c8803d2b414521cbe1e3bb but can update to latest master.
And it's around 600m. Maybe 1km. We try change RANGE_DELAY_US and check again

a1ien, 2020-11-10:
RANGE_DELAY_US only use in lll_conn.c 
What about in other lll code?
https://github.com/zephyrproject-rtos/zephyr/blob/master/subsys/bluetooth/controller/ll_sw/nordic/lll/lll_adv_aux.c#L304
https://github.com/zephyrproject-rtos/zephyr/blob/master/subsys/bluetooth/controller/ll_sw/nordic/lll/lll_scan.c#L490
As now we not support connect to ext adv. future proof?

cvinayak, 2020-11-10:
> As now we not support connect to ext adv. But for future proof?

Where you able to achieve the range required? Please send pull requests based on your testing, and if any changes required.

@andrzej-kaczmarek FYI

a1ien, 2020-11-10:
Not fully tested yet with increased RANGE_DELAY_US. We will try to collect more data and post some questions and info.
We use bt840x module and in theory we should get the same 2 km at 1M phy. So I correctly understood that to achieve connection with large distance on 1M phy we must also add RANGE_DELAY_US to this line?
https://github.com/zephyrproject-rtos/zephyr/blob/master/subsys/bluetooth/controller/ll_sw/nordic/lll/lll_adv.c#L413

cvinayak, 2020-11-13:
You mean here: https://github.com/zephyrproject-rtos/zephyr/blob/master/subsys/bluetooth/controller/ll_sw/nordic/lll/lll_adv.c#L622

a1ien, 2020-11-13:
Yes. missclick in my comment

carlescufi, 2021-12-22:
This is now exposed in this file:
https://github.com/zephyrproject-rtos/zephyr/blob/main/subsys/bluetooth/controller/ll_sw/openisa/lll/lll_tim_internal.h

closed 2021-12-22 by carlescufi