zephyrproject-rtos/zephyr #29829: On-PR CI needs to build a subset of tests for a subset of platforms in all cases

carlescufi, 2020-11-05:
There are some tests that are required to be built for a certain set of platforms in order to verify that a set of changes of changes in a PR has not broken the code.
As an example, we can use `tests/bluetooth/init`. In those tests, there is a subset with a `platform_allow` set to a few Nordic platforms. For every PR (or at the very least whenever code in `subsys/bluetooth` is modified), those tests should be built for those nRF platforms. This is not the case today:

```
$ scripts/sanitycheck --integration -T tests/bluetooth/init/
INFO    - Zephyr version: zephyr-v2.4.0-1254-gfbc487cce3
INFO    - JOBS: 8
INFO    - Selecting default platforms per test case
INFO    - Building initial testcase list...
INFO    - 40 test configurations selected, 11920 configurations discarded due to filters.
INFO    - Adding tasks to the queue...

INFO    - 0 of 0 tests passed (0.00%), 0 failed, 40 skipped with 0 warnings in 0.61 seconds
INFO    - In total 0 test cases were executed on 1 out of total 299 platforms (0.33%)
INFO    - 0 tests executed on platforms, 0 tests were only built.
```
Another example is building the MCUBoot tests for nRF platforms to avoid regressions, this is documented in #29397

carlescufi, 2020-11-05:
CC @trond-snekvik @cvinayak @nvlsianpu

nashif, 2020-11-05:
changing to a bug, at some point this used to work as expected here.

PerMac, 2020-11-05:
I have an idea for a solution for workflows where we want to be sure that building is happening on selected platforms:
-add the required platform to integration_platforms list in tests' yaml. However, this will need additional changes in the framework.
The idea is that integration_platforms list in tests' yamls is used to list the platforms to which testing will be limited when -G (--integration) arg is used for sanitycheck. However, currently, the first check is if the platform has testing: default: true in its board.yaml. If not, even if the platform is in the integration list for the test, it is skipped when -G is used. I wonder if it won't be better that when the platform is in the integration_platforms in a test's yaml the test will be called regardless of testing: default: true in the board.yaml. What do you think about such workflow?

PerMac, 2020-11-05:
Maybe this usage of `integration_platforms` would help in the confusion regarding `platform_allow`? It seems that often devs assume that having a board in `platform_allow` will make the CI run on this platform. But in fact, it only prevents from running on anything except this list. Adding the board to `integration_platforms` would then actually assure the needed behavior.

nashif, 2020-11-13:
>  I wonder if it won't be better that when the platform is in the integration_platforms in a test's yaml the test will be called regardless of `testing: default: true` in the board.yaml. What do you think about such workflow?

yes, this is one way to solve this.

> It seems that often devs assume that having a board in `platform_allow` will make the CI run on this platform.

this is the wrong assumption. platform_allow is a capability and compatibility filter, not a way to tell sanitycheck to run on those regardless of options.

PR #30009 (2020-11-17): sanitycheck: test on integration platforms on top of default platforms
When running sanitycheck without any platform filters (-p option), we
take the default platforms and run on those only. When the -G options is
used for integration testing, the platforms listed as integration
platforms are not being added on top of default platforms.

This change adds integration platforms to the list an treats
integration_platforms as a positive filter, so that
integration_platforms are always being considered.

Fixes #29829
commit 87e3ec44 (2020-11-17): sanitycheck: test on integration platforms on top of default platforms

When running sanitycheck without any platform filters (-p option), we
take the default platforms and run on those only. When the -G options is
used for integration testing, the platforms listed as integration
platforms are not being added on top of default platforms.

This change adds integration platforms to the list an treats
integration_platforms as a positive filter, so that
integration_platforms are always being considered.

Fixes #29829
closed 2020-11-17 by carlescufi