zephyrproject-rtos/zephyr #29866: Drivers/PCIE: read/write 8/16/32 bit word to an endpoint's configuration space

maxbachmann, 2020-11-07:
**Is your enhancement proposal related to a problem? Please describe.**
Right now there are only the functions `pcie_conf_read`/`pcie_conf_write` to read/write a 32 bit word to an endpoints configuration space. It would be good to have similar functions for 8 and 16 bit words aswell. 
What should be the meaning of `unsigned int reg` for these functions. There are the following two options:
1) same index for all functions, so the same index for each version of the function uses the same base address and only reads different lengths of data (for 16 bit words the provided index has to be 16 bit aligned and for 32 bit words 32 bit aligned).
2) Base this index on the word size, like it is currently done for `pcie_conf_read`/`pcie_conf_write`. So the index 4 for 8 bit words is the same as the index 2 for 16 bit words, which is the same as the index 1 for 32 bit words.

I personally prefer the first approach, since users might define an index and want to use it to read/write different word sizes. 

**Describe the solution you'd like**
Add the functions:
- uint8_t pcie_conf_read_8(pcie_bdf_t bdf, unsigned int reg);
- uint16_t pcie_conf_read_16(pcie_bdf_t bdf, unsigned int reg);
- uint32_t pcie_conf_read_32(pcie_bdf_t bdf, unsigned int reg);
- void pcie_conf_write_8(pcie_bdf_t bdf, unsigned int reg, uint8_t *data);
- void pcie_conf_write_16(pcie_bdf_t bdf, unsigned int reg, uint16_t *data);
- void pcie_conf_write_32(pcie_bdf_t bdf, unsigned int reg, uint32_t *data);

**Describe alternatives you've considered**
Pass the size to `pcie_conf_read`/`pcie_conf_write` as additional argument

tbursztyka, 2020-11-10:
>  It would be good to have similar functions for 8 and 16 bit words aswell.

To change an existing API, you'll need to explain a bit better the reason for it.

In earlier versions of Zephyr we used to have such size r/w functions (it was PCI only though, not PCIe), if I remember well.
And so far, moving away of these in the new API did not prove to be problematic. It means users have to use mask/shift. It's a different approach.

maxbachmann, 2020-11-10:
Your right with the current API it is already possible to access smaller areas by reading 32 bit and then using mask/shift to access the correct area. I prefer this api for the following reasons:
- I personally find it more straightforward to read the correct register config directly instead of reading the whole register and then extracting the value. Even though that might just be a personal preference. An advantage is, that the user can choose his prefered way of accessing them, since he can still use pcie_conf_read_32 and extract the contents using mask/shift aswell.
- Writing areas of less than 32 bit currently requires to read the register, change it and then to write it back.

closed 2022-07-20 by maxbachmann