zephyrproject-rtos/zephyr #29869: Missing error check of function entropy_get_entropy()

Hxinrong, 2020-11-08:
Hi,
As described in the project API documentation, function entropy_get_entropy() fills a buffer with entropy, returns 0 on success, and returns -ERRNO on error. However, as shown in the following code, a missing error check take places.
https://github.com/zephyrproject-rtos/zephyr/blob/577d588b3a108c34430b8fbaec9bb4f2a690b7fb/subsys/random/rand32_ctr_drbg.c#L133-L135

Furthermore, the return values of function entropy_get_entropy() from other call sites are all checked. See the following code which is from the same file, for example.
https://github.com/zephyrproject-rtos/zephyr/blob/577d588b3a108c34430b8fbaec9bb4f2a690b7fb/subsys/random/rand32_ctr_drbg.c#L85-L90

carlescufi, 2020-11-09:
@Hxinrong would you mind sending a pull request for this?

Hxinrong, 2020-11-12:
OK, thank you for your replay. Also, in the same file, seeing the following code, the return value of function tc_ctr_prng_reseed() is not checked before it is assigned by another function.
https://github.com/zephyrproject-rtos/zephyr/blob/577d588b3a108c34430b8fbaec9bb4f2a690b7fb/subsys/random/rand32_ctr_drbg.c#L136-L148

commit 6ac3d0b5 (2020-12-01): random: ctr_drbg: Check entropy_get_entropy return

entropy_get_entropy return is not being checked what may result in a
vulnerability because tc_ctr_prng_reseed will not get proper entropy
data.

Fixes #29869
closed 2020-12-01 by MaureenHelm