zephyrproject-rtos/zephyr #29877: WS2812 SPI LED strip driver produces bad SPI data

Nicell, 2020-11-09:
**Describe the bug**
The [WS2812 SPI LED strip driver](https://github.com/zephyrproject-rtos/zephyr/blob/master/drivers/led_strip/ws2812_spi.c) 
 seems to produce data with large gaps between SPI words and produces SPI words that are too long.

**To Reproduce**
Run the [WS2812 sample code](https://github.com/zephyrproject-rtos/zephyr/tree/master/samples/drivers/led_ws2812) on an nRF52840 based board (and possibly others). You should see gaps between SPI words and also see SPI words that are too long.

**Expected behavior**
An 8 bit long SPI word with no gaps between words.

**Impact**
We're using this driver on the [ZMK keyboard firmware](https://github.com/zmkfirmware/zmk). This issue results in flickering of the LEDs, which is a big annoyance.

**Logs and console output**
We have a few oscilloscope readings which show both some gaps between SPI words as well as SPI words that look to be 11bits in length rather than 8bits. https://imgur.com/a/iAFWzPA

The first image shows the length of an SPI word is about 11 bits (4MHz is about 250ns per bit, length of word from oscilloscope shows 2,760ns). The 4th image shows a big gap between the 2nd and 3rd bit. It also shows a few smaller but still possibly problematic gaps between SPI words. These are from ZMK. The last 4 signify the sample from Zephyr. Here it seems that the length of an SPI word is about 12 bits.

**Additional context**
This is occurring on a [nice!nano](https://docs.nicekeyboards.com/#/nice!nano/), which is an nRF52840 based board as well as a couple other nRF52840 based boards. Interestingly, not all users have found this results in the LEDs flickering. We've speculated that some WS2812 variants are able to deal with these SPI inconsistences better than others. We've also noted that another 
 different keyboard firmware using GPIO bitbanging does not have this same issue using the same hardware.

mbolivar-nordic, 2020-11-09:
Thanks for filing the report and for all the screenshots!

Have you tried hooking up the scope to the SPI clock pin as well? That may help diagnose. It's also a bit hard to understand what some of the other pictures in the imgur link are trying to show without context in the issue. Are any of the T0L or T1L pulses out of spec or otherwise getting close to Trset?

nixward, 2020-11-10:
Hi Nicell,
For the nRF52840 you have multiple SPI driver choices, which one are you using?  You ideally should be using the SPIM driver so each word transfer is off loaded to DMA.

Nicell, 2020-11-10:
@nixward, thanks for the suggestion! Switching to SPIM has indeed fixed our issues for those who were affected. I'm assuming the CPU was busy for a few µs, which caused the strip update to get reset in the middle of transfer.

I'd like to note the fact that the LED strip sample uses SPI rather than SPIM for the nRF52dk. Should this be changed or noted as a possible fix? This issue doesn't seem to be present in the sample program, which again I'm assuming is because there's nothing else going on to make the CPU busy.

mbolivar-nordic, 2020-11-13:
> I'd like to note the fact that the LED strip sample uses SPI rather than SPIM for the nRF52dk. Should this be changed or noted as a possible fix?

Ah, yep, I was unaware this was defaulting to SPI instead of SPIM on this SoC. Thanks, I've sent #29984 to address this.

Please make sure you understand PAN 58 before using this in ZMK. It should be fine as long as the SPI peripheral is only being used for LEDs and not some other chip on the same bus.

PR #29984 (2020-11-17): samples: led_ws2812: force SPIM on nRF52-DK
The WS2812 SPI driver needs the pixel data to be clocked out on time,
with no gaps between SPI frames. This makes the nordic,nrf-spi
compatible unsuitable for use, since the corresponding driver doesn't
use EasyDMA to clock out the tx buffers.

Force the arduino_spi node in the DT to SPIM mode by setting its
compatible to nordic,nrf-spim. On nRF52832, this driver is disabled by
default due to Product Anomaly Notice (PAN) 58.

We happen to not be affected by PAN 58 in this instance, because we
never read data over the SPIM, only write it. So it's safe [... cut]
commit 8ea9941f (2020-11-17): samples: led_ws2812: force SPIM on nRF52-DK

The WS2812 SPI driver needs the pixel data to be clocked out on time,
with no gaps between SPI frames. This makes the nordic,nrf-spi
compatible unsuitable for use, since the corresponding driver doesn't
use EasyDMA to clock out the tx buffers.

Force the arduino_spi node in the DT to SPIM mode by setting its
compatible to nordic,nrf-spim. On nRF52832, this driver is disabled by
default due to Product Anomaly Notice (PAN) 58.

We happen to not be affected by PAN 58 in this instance, because we
never read data over the SPIM, only write it. So it's saf [... cut]
closed 2020-11-17 by MaureenHelm