zephyrproject-rtos/zephyr #29879: samples/net/gptp compile failed on frdm_k64f board in origin/master (work well in origin/v2.4-branch)

KelvinTsoi, 2020-11-09:
Here is the compile information.

```
➜  zephyr git:(master) west build -b frdm_k64f  samples/net/gptp
-- west build: generating a build system
Including boilerplate (Zephyr base): /home/yujiancai/zephyrproject/zephyr/cmake/app/boilerplate.cmake
-- Application: /home/yujiancai/zephyrproject/zephyr/samples/net/gptp
-- Zephyr version: 2.4.99 (/home/yujiancai/zephyrproject/zephyr)
-- Found Python3: /usr/bin/python3.7 (found suitable exact version "3.7.9") found components: Interpreter 
-- Found west (found suitable version "0.7.3", minimum required is "0.7.1")
-- Board: frdm_k64f
-- Cache files will be written to: /home/yujiancai/.cache/zephyr
ZEPHYR_TOOLCHAIN_VARIANT not set, trying to locate Zephyr SDK
-- Found toolchain: zephyr (/opt/zephyr-sdk-0.11.4)
-- Found dtc: /opt/zephyr-sdk-0.11.4/sysroots/x86_64-pokysdk-linux/usr/bin/dtc (found suitable version "1.5.0", minimum required is "1.4.6")
-- Found BOARD.dts: /home/yujiancai/zephyrproject/zephyr/boards/arm/frdm_k64f/frdm_k64f.dts
-- Generated zephyr.dts: /home/yujiancai/zephyrproject/zephyr/build/zephyr/zephyr.dts
-- Generated devicetree_unfixed.h: /home/yujiancai/zephyrproject/zephyr/build/zephyr/include/generated/devicetree_unfixed.h
Parsing /home/yujiancai/zephyrproject/zephyr/samples/net/gptp/Kconfig
Loaded configuration '/home/yujiancai/zephyrproject/zephyr/boards/arm/frdm_k64f/frdm_k64f_defconfig'
Merged configuration '/home/yujiancai/zephyrproject/zephyr/samples/net/gptp/prj.conf'
Merged configuration '/home/yujiancai/zephyrproject/zephyr/samples/net/gptp/boards/frdm_k64f.conf'
Configuration saved to '/home/yujiancai/zephyrproject/zephyr/build/zephyr/.config'
Kconfig header saved to '/home/yujiancai/zephyrproject/zephyr/build/zephyr/include/generated/autoconf.h'

warning: TEST_RANDOM_GENERATOR (defined at subsys/random/Kconfig:8) was assigned the value 'y' but
got the value 'n'. Check these unsatisfied dependencies: (!ENTROPY_HAS_DRIVER) (=n). See
http://docs.zephyrproject.org/latest/reference/kconfig/CONFIG_TEST_RANDOM_GENERATOR.html and/or look
up TEST_RANDOM_GENERATOR in the menuconfig/guiconfig interface. The Application Development Primer,
Setting Configuration Values, and Kconfig - Tips and Best Practices sections of the manual might be
helpful too.

-- The C compiler identification is GNU 9.2.0
-- The CXX compiler identification is GNU 9.2.0
-- The ASM compiler identification is GNU
-- Found assembler: /opt/zephyr-sdk-0.11.4/arm-zephyr-eabi/bin/arm-zephyr-eabi-gcc
-- Configuring done
-- Generating done
-- Build files have been written to: /home/yujiancai/zephyrproject/zephyr/build
-- west build: building application
[1/194] Preparing syscall dependency handling

[85/194] Building C object zephyr/CMakeFiles/zephyr.dir/drivers/ethernet/eth_mcux.c.obj
FAILED: ccache /opt/zephyr-sdk-0.11.4/arm-zephyr-eabi/bin/arm-zephyr-eabi-gcc -DBOARD_FLASH_SIZE="CONFIG_FLASH_SIZE*1024" -DBUILD_VERSION=zephyr-v2.4.0-1305-g577d588b3a10 -DCPU_MK64FN1M0VLL12 -DENET_ENHANCEDBUFFERDESCRIPTOR_MODE -DKERNEL -DNDEBUG -D_FORTIFY_SOURCE=2 -D__PROGRAM_START -D__ZEPHYR__=1 -I../include -Izephyr/include/generated -I../soc/arm/nxp_kinetis/k6x -I../lib/fnmatch/. -I../subsys/net/l2 -I/home/yujiancai/zephyrproject/modules/hal/cmsis/CMSIS/Core/Include -I/home/yujiancai/zephyrproject/modules/hal/nxp/mcux/devices/MK64F12 -I/home/yujiancai/zephyrproject/modules/hal/nxp/mcux/components/phyksz8081 -I/home/yujiancai/zephyrproject/modules/hal/nxp/mcux/drivers/kinetis/. -isystem ../lib/libc/minimal/include -isystem /opt/zephyr-sdk-0.11.4/arm-zephyr-eabi/bin/../lib/gcc/arm-zephyr-eabi/9.2.0/include -isystem /opt/zephyr-sdk-0.11.4/arm-zephyr-eabi/bin/../lib/gcc/arm-zephyr-eabi/9.2.0/include-fixed -Os -imacros /home/yujiancai/zephyrproject/zephyr/build/zephyr/include/generated/autoconf.h -ffreestanding -fno-common -g -mcpu=cortex-m4 -mthumb -mabi=aapcs -imacros /home/yujiancai/zephyrproject/zephyr/include/toolchain/zephyr_stdint.h -Wall -Wformat -Wformat-security -Wno-format-zero-length -Wno-main -Wno-pointer-sign -Wpointer-arith -Wno-address-of-packed-member -Wno-unused-but-set-variable -Werror=implicit-int -fno-asynchronous-unwind-tables -fno-pie -fno-pic -fno-strict-overflow -fno-reorder-functions -fno-defer-pop -fmacro-prefix-map=/home/yujiancai/zephyrproject/zephyr/samples/net/gptp=CMAKE_SOURCE_DIR -fmacro-prefix-map=/home/yujiancai/zephyrproject/zephyr=ZEPHYR_BASE -fmacro-prefix-map=/home/yujiancai/zephyrproject=WEST_TOPDIR -ffunction-sections -fdata-sections -std=c99 -nostdinc -MD -MT zephyr/CMakeFiles/zephyr.dir/drivers/ethernet/eth_mcux.c.obj -MF zephyr/CMakeFiles/zephyr.dir/drivers/ethernet/eth_mcux.c.obj.d -o zephyr/CMakeFiles/zephyr.dir/drivers/ethernet/eth_mcux.c.obj -c /home/yujiancai/zephyrproject/zephyr/drivers/ethernet/eth_mcux.c
/home/yujiancai/zephyrproject/zephyr/drivers/ethernet/eth_mcux.c: In function 'eth_tx':
/home/yujiancai/zephyrproject/zephyr/drivers/ethernet/eth_mcux.c:673:12: error: too many arguments to function 'ENET_SendFrame'
  673 |   status = ENET_SendFrame(context->base, &context->enet_handle,
      |            ^~~~~~~~~~~~~~
In file included from /home/yujiancai/zephyrproject/zephyr/drivers/ethernet/eth_mcux.c:39:
/home/yujiancai/zephyrproject/modules/hal/nxp/mcux/drivers/kinetis/./fsl_enet.h:1217:10: note: declared here
 1217 | status_t ENET_SendFrame(ENET_Type *base, enet_handle_t *handle, const uint8_t *data, uint32_t length);
      |          ^~~~~~~~~~~~~~
/home/yujiancai/zephyrproject/zephyr/drivers/ethernet/eth_mcux.c:689:12: error: too many arguments to function 'ENET_SendFrame'
  689 |   status = ENET_SendFrame(context->base, &context->enet_handle,
      |            ^~~~~~~~~~~~~~
In file included from /home/yujiancai/zephyrproject/zephyr/drivers/ethernet/eth_mcux.c:39:
/home/yujiancai/zephyrproject/modules/hal/nxp/mcux/drivers/kinetis/./fsl_enet.h:1217:10: note: declared here
 1217 | status_t ENET_SendFrame(ENET_Type *base, enet_handle_t *handle, const uint8_t *data, uint32_t length);
      |          ^~~~~~~~~~~~~~
/home/yujiancai/zephyrproject/zephyr/drivers/ethernet/eth_mcux.c: In function 'eth_rx':
/home/yujiancai/zephyrproject/zephyr/drivers/ethernet/eth_mcux.c:718:11: error: too many arguments to function 'ENET_GetRxFrameSize'
  718 |  status = ENET_GetRxFrameSize(&context->enet_handle,
      |           ^~~~~~~~~~~~~~~~~~~
In file included from /home/yujiancai/zephyrproject/zephyr/drivers/ethernet/eth_mcux.c:39:
/home/yujiancai/zephyrproject/modules/hal/nxp/mcux/drivers/kinetis/./fsl_enet.h:1160:10: note: declared here
 1160 | status_t ENET_GetRxFrameSize(enet_handle_t *handle, uint32_t *length);
      |          ^~~~~~~~~~~~~~~~~~~
/home/yujiancai/zephyrproject/zephyr/drivers/ethernet/eth_mcux.c:725:3: error: too many arguments to function 'ENET_GetRxErrBeforeReadFrame'
  725 |   ENET_GetRxErrBeforeReadFrame(&context->enet_handle,
      |   ^~~~~~~~~~~~~~~~~~~~~~~~~~~~
[... 40 lines cut ...]
/home/yujiancai/zephyrproject/zephyr/drivers/ethernet/eth_mcux.c:1372:1165: error: unknown type name 'enet_frame_info_t'
 1372 | DT_INST_FOREACH_STATUS_OKAY(ETH_MCUX_INIT)
      |                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                             ^                
/home/yujiancai/zephyrproject/zephyr/drivers/ethernet/eth_mcux.c:1372:1681: error: 'enet_buffer_config_t' {aka 'const struct _enet_buffer_config'} has no member named 'rxMaintainEnable'
 1372 | DT_INST_FOREACH_STATUS_OKAY(ETH_MCUX_INIT)
[... cut ...]
/home/yujiancai/zephyrproject/zephyr/drivers/ethernet/eth_mcux.c:705:13: warning: 'eth_rx' defined but not used [-Wunused-function]
  705 | static void eth_rx(struct eth_context *context)
      |             ^~~~~~
[85/194] Building C object zephyr/arch/arch/arm/core/aarch32/cortex_m/CMakeFiles/arch__arm__core__aarch32__cortex_m.dir/fault.c.obj
ninja: build stopped: subcommand failed.
FATAL ERROR: command exited with status 1: /usr/bin/cmake --build /home/yujiancai/zephyrproject/zephyr/build

```
Would anyone help to take a look? Thanks very much.

jukkar, 2020-11-09:
You need new version of the NXP hal, so please do `west update` and try again.

closed 2020-11-09 by jukkar

KelvinTsoi, 2020-12-21:
```
➜  zephyr git:(master) west build -b frdm_k64f samples/net/gptp
[1/1] Re-running CMake...
Including boilerplate (Zephyr base (cached)): /home/yujiancai/zephyrproject/zephyr/cmake/app/boilerplate.cmake
-- Application: /home/yujiancai/zephyrproject/zephyr/samples/net/gptp
-- Zephyr version: 2.4.99 (/home/yujiancai/zephyrproject/zephyr)
-- Found west (found suitable version "0.7.3", minimum required is "0.7.1")
-- Board: frdm_k64f
-- Cache files will be written to: /home/yujiancai/.cache/zephyr
-- Found toolchain: zephyr (/opt/zephyr-sdk-0.11.4)
-- Found dtc: /opt/zephyr-sdk-0.11.4/sysroots/x86_64-pokysdk-linux/usr/bin/dtc (found suitable version "1.5.0", minimum required is "1.4.6")
-- Found BOARD.dts: /home/yujiancai/zephyrproject/zephyr/boards/arm/frdm_k64f/frdm_k64f.dts
-- Generated zephyr.dts: /home/yujiancai/zephyrproject/zephyr/build/zephyr/zephyr.dts
-- Generated devicetree_unfixed.h: /home/yujiancai/zephyrproject/zephyr/build/zephyr/include/generated/devicetree_unfixed.h
Parsing /home/yujiancai/zephyrproject/zephyr/samples/net/gptp/Kconfig
Loaded configuration '/home/yujiancai/zephyrproject/zephyr/build/zephyr/.config'
No change to configuration in '/home/yujiancai/zephyrproject/zephyr/build/zephyr/.config'
No change to Kconfig header in '/home/yujiancai/zephyrproject/zephyr/build/zephyr/include/generated/autoconf.h'
-- Configuring done
-- Generating done
-- Build files have been written to: /home/yujiancai/zephyrproject/zephyr/build
[2/173] Building C object zephyr/CMakeFiles/offsets.dir/arch/arm/core/offsets/offsets.c.obj
In file included from ../kernel/include/kernel_offsets.h:6,
                 from /home/yujiancai/zephyrproject/zephyr/arch/arm/core/offsets/offsets_aarch32.c:30,
                 from /home/yujiancai/zephyrproject/zephyr/arch/arm/core/offsets/offsets.c:12:
../include/device.h:394:15: warning: 'device_is_ready' declared 'static' but never defined [-Wunused-function]
  394 | __syscall bool device_is_ready(const struct device *dev);
      |               ^~~~~~~~~~~~~~~
[18/173] Building C object CMakeFiles/app.dir/src/gptp.c.obj
In file included from /home/yujiancai/zephyrproject/zephyr/include/net/net_if.h:22,
                 from /home/yujiancai/zephyrproject/zephyr/include/net/net_pkt.h:27,
                 from /home/yujiancai/zephyrproject/zephyr/subsys/net/l2/ethernet/gptp/gptp_messages.h:17,
                 from /home/yujiancai/zephyrproject/zephyr/samples/net/gptp/src/gptp.c:17:
/home/yujiancai/zephyrproject/zephyr/include/device.h:394:15: warning: 'device_is_ready' declared 'static' but never defined [-Wunused-function]
  394 | __syscall bool device_is_ready(const struct device *dev);
      |               ^~~~~~~~~~~~~~~
[20/173] Building C object CMakeFiles/app.dir/src/main.c.obj
In file included from /home/yujiancai/zephyrproject/zephyr/include/net/net_if.h:22,
                 from /home/yujiancai/zephyrproject/zephyr/samples/net/gptp/src/main.c:15:
/home/yujiancai/zephyrproject/zephyr/include/device.h:394:15: warning: 'device_is_ready' declared 'static' but never defined [-Wunused-function]
  394 | __syscall bool device_is_ready(const struct device *dev);
      |               ^~~~~~~~~~~~~~~
[25/173] Building C object zephyr/CMakeFiles/zephyr.dir/soc/arm/nxp_kinetis/k6x/soc.c.obj
In file included from /home/yujiancai/zephyrproject/zephyr/soc/arm/nxp_kinetis/k6x/soc.c:17:
../include/device.h:394:15: warning: 'device_is_ready' declared 'static' but never defined [-Wunused-function]
  394 | __syscall bool device_is_ready(const struct device *dev);
      |               ^~~~~~~~~~~~~~~
[26/173] Building C object zephyr/CMakeFiles/zephyr.dir/subsys/shell/modules/kernel_service.c.obj
In file included from /home/yujiancai/zephyrproject/zephyr/subsys/shell/modules/kernel_service.c:15:
../include/device.h:394:15: warning: 'device_is_ready' declared 'static' but never defined [-Wunused-function]
  394 | __syscall bool device_is_ready(const struct device *dev);
      |               ^~~~~~~~~~~~~~~
[34/173] Building C object zephyr/CMakeFiles/zephyr.dir/subsys/shell/modules/device_service.c.obj
In file included from /home/yujiancai/zephyrproject/zephyr/subsys/shell/modules/device_service.c:11:
../include/device.h:394:15: warning: 'device_is_ready' used but never defined
  394 | __syscall bool device_is_ready(const struct device *dev);
      |               ^~~~~~~~~~~~~~~
[35/173] Building C object zephyr/CMakeFiles/zephyr.dir/subsys/shell/shell_utils.c.obj
In file included from /home/yujiancai/zephyrproject/zephyr/subsys/shell/shell_utils.c:7:
../include/device.h:394:15: warning: 'device_is_ready' used but never defined
  394 | __syscall bool device_is_ready(const struct device *dev);
      |               ^~~~~~~~~~~~~~~
[39/173] Building C object zephyr/CMakeFiles/zephyr.dir/subsys/shell/shell_uart.c.obj
In file included from ../include/drivers/uart.h:26,
                 from /home/yujiancai/zephyrproject/zephyr/subsys/shell/shell_uart.c:8:
../include/device.h:394:15: warning: 'device_is_ready' declared 'static' but never defined [-Wunused-function]
  394 | __syscall bool device_is_ready(const struct device *dev);
      |               ^~~~~~~~~~~~~~~
[45/173] Building C object zephyr/CMakeFiles/zephyr.dir/drivers/console/uart_console.c.obj
In file included from /home/yujiancai/zephyrproject/zephyr/drivers/console/uart_console.c:24:
[... 180 lines cut ...]
      |               ^~~~~~~~~~~~~~~
[... cut ...]
                 from ../include/net/net_pkt.h:27,
                 from /home/yujiancai/zephyrproject/zephyr/subsys/net/ip/net_if.c:18:
../include/device.h:394:15: warning: 'device_is_ready' declared 'static' but never defined [-Wunused-function]
  394 | __syscall bool device_is_ready(const struct device *dev);
      |               ^~~~~~~~~~~~~~~
ninja: build stopped: subcommand failed.
FATAL ERROR: command exited with status 1: /usr/bin/cmake --build /home/yujiancai/zephyrproject/zephyr/build

```

KelvinTsoi, 2020-12-21:
still has some problems after updating the lib.

jukkar, 2020-12-21:
I am not seeing errors like that, gptp sample compiles just fine. Have yo done `west update`?
```
$ git describe 
zephyr-v2.4.0-2666-gfe83118432
$ cd samples/net/gptp
$ cmake -Bbuild . -DBOARD=frdm_k64f
$ make -C build
...
[ 97%] Linking C executable zephyr_prebuilt.elf
Memory region         Used Size  Region Size  %age Used
           FLASH:      156840 B         1 MB     14.96%
            SRAM:       65296 B       192 KB     33.21%
        IDT_LIST:         216 B         2 KB     10.55%
...
```