zephyrproject-rtos/zephyr #29884: x_nucleo_iks01a2 device tree overlay issue with stm32mp157c_dk2 board

cpriouzeau, 2020-11-09:
**Describe the bug**
On master branch, the overlay of board for shield x_nucleo_iks01a2 are not applied on correct order.

The zephyr/boards/shields/x_nucleo_iks01a2/boards/stm32mp157c_dk2.overlay overlay are applied before zephyr/boards/shields/x_nucleo_iks01a2/boards/stm32mp157c_dk2.overlay which generate an issue during the generation of device tree.

**To Reproduce**
Steps to reproduce the behavior:
1. get zephyr on master branch (tested with ec0e737b0c7c0d19fcb9733275ab8bb2743b828e)
2. west build -b stm32mp157c_dk2 -d build-stm32mp1-master zephyr/samples/shields/x_nucleo_iks01a2/standard
3. See error
devicetree error: child specifier for <Node /soc/i2c@40015000/lsm303agr-magn@1e in 'stm32mp157c_dk2.dts.pre.tmp'> (b'\x00\x00\x00\x03\x00\x00\x00\x00') does not appear in <Property 'gpio-map' at '/connector' in 'stm32mp157c_dk2.dts.pre.tmp'>

**Expected behavior**
We  expected to have the overlay applied on the correct order, in this case: 
   1. zephyr/boards/shields/x_nucleo_iks01a2/boards/stm32mp157c_dk2.overlay 
   2. zephyr/boards/shields/x_nucleo_iks01a2/boards/stm32mp157c_dk2.overlay
and not the board before the shield 

**Impact**
showstopper: doesnt able to generate a correct device tree when we try to associate stm32mp157c-dk2 board with shield x_nucleo_iks01a2 (same issue see with x_nucleo_iks01a3)

**Logs and console output**
Device tree generated on zephyr branch 2.2.0 on which the device tree are correctly generated
Device tree : build-stm32mp1-2.2.0/zephyr/stm32mp157c_dk2.dts.pre.tmp
/* content of zephyr/boards/shields/x_nucleo_iks01a2/x_nucleo_iks01a2.overlay */
&arduino_i2c {
 ...
 lsm303agr-magn@1e {
  compatible = "st,lis2mdl","st,lsm303agr-magn";
  reg = <0x1e>;
  label = "LSM303AGR-MAGN";
  irq-gpios = <&arduino_header 3 (0 << 0)>;
 };
 lsm303agr-accel@19 {
  compatible = "st,lis2dh", "st,lsm303agr-accel";
  reg = <0x19>;
  label = "LSM303AGR-ACCEL";
  irq-gpios = <&arduino_header 3 (0 << 0)>;
 };
};
/* content of zephyr/boards/shields/x_nucleo_iks01a2/boards/stm32mp157c_dk2.overlay */
&arduino_i2c {
 lsm303agr-magn@1e {
  /delete-property/ irq-gpios;
 };
 lsm303agr-accel@19 {
  /delete-property/ irq-gpios;
 };

Device tree generated on zephyr branch master (ec0e737b0c7c0d19fcb9733275ab8bb2743b828e)
Device tree : build-stm32mp1-master/zephyr/stm32mp157c_dk2.dts.pre.tmp
/* content of zephyr/boards/shields/x_nucleo_iks01a2/boards/stm32mp157c_dk2.overlay */
&arduino_i2c {
 lsm303agr-magn@1e {
  /delete-property/ irq-gpios;
 };
 lsm303agr-accel@19 {
  /delete-property/ irq-gpios;
 };
};
/* content of zephyr/boards/shields/x_nucleo_iks01a2/x_nucleo_iks01a2.overlay*/
&arduino_i2c {
 ...
 lsm303agr-magn@1e {
  compatible = "st,lis2mdl","st,lsm303agr-magn";
  reg = <0x1e>;
  label = "LSM303AGR-MAGN";
  irq-gpios = <&arduino_header 3 (0 << 0)>;
 };
 lsm303agr-accel@19 {
  compatible = "st,lis2dh", "st,lsm303agr-accel";
  reg = <0x19>;
  label = "LSM303AGR-ACCEL";
  irq-gpios = <&arduino_header 3 (0 << 0)>;
 };
};

To test:
On zephyr 2.2.0
west build -b stm32mp157c_dk2 -d build-stm32mp1-2.2.0 zephyr/samples/shields/x_nucleo_iks01a2
On Zephyr master (ec0e737b0c7c0d19fcb9733275ab8bb2743b828e)
west build -b stm32mp157c_dk2 -d build-stm32mp1-master zephyr/samples/shields/x_nucleo_iks01a2/standard

**Environment (please complete the following information):**
 - OS: Linux( ubuntu 18.04)
 - Toolchain: Zephyr SDK (zephyr-sdk-0.10.1)

erwango, 2020-11-20:
Fixed by https://github.com/zephyrproject-rtos/zephyr/pull/29824.
@cpriouzeau can you have a check ?

cpriouzeau, 2020-11-20:
@erwango Checked with x_nucleo_iks01a2: OK
The issue corrected for me

closed 2020-11-26 by erwango