zephyrproject-rtos/zephyr #29891: mcumgr image upload (with smp_svr) does not work over serial/shell on the nrf52840dk

utzig, 2020-11-09:
I'm not able to get `image upload` to upload new images using the serial or shell overlays on the nrf52840DK. Passing in a longer timeout does not seem to have any effect on behavior. All other mcumgr commands seem to work correctly. I am able to upload images using bluetooth, so it's probably not flash driver related. `image upload` over shell on UART works correctly on the `frdm_k64f` BSP so this bug is probably not something on my computer or just a generic bug in the mcumgr port.

utzig, 2020-11-09:
/cc @de-nordic @nvlsianpu

de-nordic, 2020-11-09:
Which Zephyr commit do you have on top?

utzig, 2020-11-09:
> Which Zephyr commit do you have on top?

If you mean the tip, for the tests today it is: https://github.com/zephyrproject-rtos/zephyr/commit/f8956cfbc733bc76605e4fa1136a9542bd56d7ba

Also I should mention I have not tested the `frdm_k64f` today. I tested it a few weeks ago and it was working, and by then the nrf52840dk already was not. Also yesterday I tested some new MCUboot stuff with Mynewt and all mcumgr commands work correctly over UART on the nrf52840dk, so this bug must be something specific to Zephyr+mcumgr+UART.

de-nordic, 2020-11-09:
Unfortunately works on my nrf52840dk.

Do you have 'CONFIG_UART_CONSOLE=n' ?

de-nordic, 2020-11-09:
My config for comparison
[smp_svr.config.txt](https://github.com/zephyrproject-rtos/zephyr/files/5510959/smp_svr.config.txt)

utzig, 2020-11-09:
> Do you have `CONFIG_UART_CONSOLE=n` ?

Could be, I will try again later today to check. Do you have any particular option set for you mcumgr config, like mtu?

nvlsianpu, 2020-11-09:
I was building according to https://docs.zephyrproject.org/latest/samples/subsys/mgmt/mcumgr/smp_svr/README.html serial option - works fine.

utzig, 2020-11-09:
Ok, so I tried again now, and both `overlay-serial.conf` and `overlay-shell.conf` are working. Actually I was deceiving when I answered the SHA in the tip of my `master`, which is the one I am using today with BLE, but I was testing over serial/shell two weeks ago on 28/10, so either something changed in the interim, like a bug fix, or something else happened in my setup which will be quite hard to track down. Sorry about the noise; closing this!

closed 2020-11-09 by utzig

de-nordic, 2020-11-09:
> 
> Could be, I will try again later today to check. Do you have any particular option set for you mcumgr config, like mtu?

Nope. All I have added by, hand via `-t menuconfig`, is  `SHELL=y`, `UART_CONSOLE=n` and `MCUMGR_SMP_SHELL=y`.

utzig, 2020-11-09:
> > Could be, I will try again later today to check. Do you have any particular option set for you mcumgr config, like mtu?
> 
> Nope. All I have added by, hand via `-t menuconfig`, is `SHELL=y`, `UART_CONSOLE=n` and `MCUMGR_SMP_SHELL=y`.

I did not have to add any options apart from using the overlay files which are part of the smp_svr sample.