zephyrproject-rtos/zephyr #29896: new documentation build warning

pabigot, 2020-11-09:
From #28779 which included a closing `@}` in the header without any opening one.  Probably some of the macros in this are supposed to be in user documentation

```
 ---- New errors/warnings not tracked as .known-issues/, please fix them ----

/mnt/nordic/zp/zephyr/include/sys/util_macro.h:577: warning: unbalanced grouping commands
```

utzig, 2020-11-09:
@pabigot What is your Doxygen version?

pabigot, 2020-11-09:
> What is your Doxygen version?

Locally built, doxygen-Release_1_8_19-5-g8e87f6bb4 which is a little past 1.8.19.

pabigot, 2020-11-09:
Regardless, the file needs to be fixed to have balanced group commands.

utzig, 2020-11-09:
> Regardless, the file needs to be fixed to have balanced group commands.

I have not considered not fixing the file. I just found it weird that 1.8.13 ignores the problem. I will check if we can just move on to something a bit newer...

PR #29922 (2020-11-10): doc: fix reference/util after sys/util.h split
Add back Doxygen documentation that previously existed in sys/util.h and was moved to sys/util_macro.h.

Also fixes #29896
commit 448db808 (2020-11-10): doc: fix reference/util after sys/util.h split

Add back Doxygen documentation that previously existed in sys/util.h
and was moved to sys/util_macro.h.

Also fixes #29896
closed 2020-11-10 by carlescufi