zephyrproject-rtos/zephyr #29913: Question : Bluetooth mesh using long range

Tobi15, 2020-11-10:
For a specific project, i would like to use the concept of Bluetooth mesh with long range (coded phy). So far, i have tried to understand both parts. For the long rage part, i have created a beacon by using these specific methods.
```
bt_le_ext_adv_create();
bt_le_ext_adv_set_data();
bt_le_ext_adv_start();
```
[beacon_lr.zip](https://github.com/zephyrproject-rtos/zephyr/files/5515533/beacon_lr.zip)

By analyzing frames from my nrf52840dk, i found that those frames were really using long range feature (coded phy). 

For the other part, i flashed the specific example `zephyr/samples/board/nrf/mesh/onoff-app/` on my board and provisioning this board with the app for iphone `nRF Mesh`. The both parts were working well.
 
Then, i tried to mix the two of then to have my goal which is Bluetooth mesh on long range. I've seen in the zephyr SDK that the communication between the BLE layer and the Mesh layer is made inside the file `zephyr/subsys/bluetooth/mesh/adv.c`. There seems to be no implemention for using long range in this file. My first question is : Do i use the right way to implement the long range feature?

Does it already exist  an implementation of bluetooth mesh on long range in zephyr ? if not, is it planned in a nearest future an implementation for this feature ?

Thanks in advance for the reply.

Best regards

trond-snekvik, 2020-11-10:
Hi, 

You seem to be on the right track, but you're the first to pursue this in the Zephyr project, as far as I'm aware.
Long range is (at least for the moment) not available as a bearer for the mesh in the spec, and we're not expecting to support a proprietary version of this in the Zephyr project in the near future. 

I am planning on migrating adv.c to using the extended advertising API though, which should make it significantly easier to patch the mesh stack to use long range in a proprietary variant of the mesh. As far as I can tell, it should be possible to combine the long range feature with the mesh by converting to the extended advertising API and setting the `BT_LE_ADV_OPT_CODED` flag in the adv_param passed to `bt_le_ext_adv_create`.

trond-snekvik, 2020-11-13:
I have added support for ext adv in mesh in a PR: https://github.com/zephyrproject-rtos/zephyr/pull/30005

Once this goes in, it should be a lot simpler to add long range support.

Ripitchip, 2020-11-23:
Hello, (I'm a colleague of Tobi)

> Long range is (at least for the moment) not available as a bearer for the mesh in the spec, and we're not expecting to support a proprietary version of this in the Zephyr project in the near future.

With our actual understanding of the bluetooth mesh, there are two types of bearers :
- advertising bearer : used to transmit messages between devices
- GATT bearer : used to communicate from an external device to the mesh trough a proxy (cf. a mobile phone used for provisioning).

Does that mean that :
- advertising bearer : Does it mean that it's not possible to send messages in the mesh with long range by just changing adv.c ? Because at first, we thought it would be enough to change the phy medium and let adv.c scan and send messages in long range.
- GATT bearer : In this case, I suppose it would be possible to do self approvisionning (such as the mesh_demo sample) in 1M and then send / scan in 125k ?

As you seem to be from Nordic, could we ask you if the nRF connect SDK would have a proprietary version of the bearer ? In short, would we able to setup a long range mesh network with the nRF connect SDK and/or nRF5 SDK ?

Best regards

trond-snekvik, 2020-11-23:
It will not be possible to qualify a solution on the long range advertising bearer, as it'd be a direct violation of the first sentence in section 3.3.1 in the mesh profile specification. Even if we provide a solution for using long range in the mesh, it'll be proprietary, unless the Bluetooth SIG ends up adding support for it in a future version of the specification.

With that being said, we can provide a proprietary extension of the advertising bearer, similar to the [Nordic DevZone patch](https://devzone.nordicsemi.com/f/nordic-q-a/29813/change-phy-in-mesh/118379#118379) for the nRF5 SDK. I'm not sure whether we'll be able to put this in the stack, or whether it'll be a patch like the nRF5 SDK solution. This'll have to wait until the extended advertising support is merged

carlescufi, 2020-12-03:
Closing since the question is answered.

closed 2020-12-03 by carlescufi

rajeevkr945, 2024-05-01:
so @trond-snekvik , How can i Implement a Coded-PHY mesh using your SDK? . I am using a Vendor specific Model in which I send My sensor data via mesh. I am facing Range issues here, It will be a great help if I could make it coded

hassin23ayz, 2025-01-12:
Dear @carlescufi & @trond-snekvik , 
Using Zephyr SDK version 3.7.0 I successfully ran the Bluetooth mesh example on two rf52840 chips (XIAO_BLE module.) But the mesh range is terrible. It is only a few meters. For example, If one node is just put behind the room next door,  then the mesh does not work anymore. Is there any Project-related configuration available in the zephyr SDK to improve the range?  I would be very thankful if you kindly reply

trond-snekvik, 2025-01-12:
Hey, sorry, I no longer work at Nordic, or with Bluetooth Mesh. It sounds like your issues are antenna related, but you could look into increasing the TX power on the advertisements. I'd recommend asking on the discord or the Nordic devzone

hassin23ayz, 2025-01-13:
Dear @trond-snekvik thanks a lot for your kind attention and reply. I will then look for the answers in nordic devZone or the discord channel