zephyrproject-rtos/zephyr #29921: USB DFU with nrf52840dk (PCA10056)

nehalrohit, 2020-11-10:
Hello, community
I want to USB dfu with nrf52840dk I am following this link.
**https://docs.zephyrproject.org/1.12.0/samples/subsys/usb/dfu/README.html**

I  did a USB dfu with nrf52840dk. but I am facing a problem after resetting the second time, my code should run but It was not.
may I need to do any change in code or any configuration?

carlescufi, 2020-11-10:
You are pointing at a very old version of the doc, here's the latest:
https://docs.zephyrproject.org/latest/samples/subsys/usb/dfu/README.html

de-nordic, 2020-11-10:
Can you gather the mcuboot output during bootup and post it here?

nehalrohit, 2020-11-11:
> Can you gather the mcuboot output during bootup and post it here?

Hello de-nordic,

I want to USB DFU like if I press power on reset + button on DK board enter into dfu mode. and if I only press power-on reset device code run as usual.
This is the step I do.
1- Build and flash MCUboot.
  build mcuboot using **west build -b nrf52840dk_nrf52840** and flash using **west flash**.
2 build USB dfu 
 build mcuboot using **west build -b nrf52840dk_nrf52840** and flash using **west flash**.
[image]

3 Use the following command to update the application:
dfu-util --alt 1 --download build/zephyr/app_update.bin 
[image]

[image]

de-nordic, 2020-11-12:
@nehalrohit Everything happens as expected.

The documentation fails to inform that the binary will be uploaded, when `dfu-util --alt 1` is invoked, to the secondary application slot, and will be tested only on the first boot afterwards. When you induce hard reset, with the reset button, the application are swapped back, and from now on the primary slot will be always executed.

If you want the dfu to be permanent, you need to upload your binary into the first application slot; you do that by invoking:
`dfu-util --alt 0 --download signed-hello.bin`
instead of
`dfu-util --alt 1 --download signed-hello.bin`

nehalrohit, 2020-11-12:
Hi @de-nordic 
I am trying 
**dfu-util --alt 0 --download signed-hello.bin** but I get below error 
[image]

de-nordic, 2020-11-12:
> I am trying **dfu-util --alt 0 --download build/zephyr/app_update.bin** but I get below error

Yes. My bad. I have mistook the mcuboot DFU with Zephyr DFU.
I do not think that you will be able to do the `--alt 0` from running zephyr application as it would overwrite itself.
I need to check if there is a way to confirm the tested application via `dfu-util`.

You may try using mcuboot with USB dfu enabled instead: https://github.com/zephyrproject-rtos/zephyr/pull/28506

carlescufi, 2020-11-17:
>I need to check if there is a way to confirm the tested application via dfu-util.

I seem to recall that this was possible somehow, because we were able to do permanent DFU using dfu-util in the past, if I remember correctly. CC @jfischer-no

de-nordic, 2020-11-17:
> 
> I seem to recall that this was possible somehow, because we were able to do permanent DFU using dfu-util in the past, if I remember correctly. CC @jfischer-no

I have asked @nvlsianpu about this and it seems that the application is supposed to confirm itself.

jfischer-no, 2020-11-18:
> > I seem to recall that this was possible somehow, because we were able to do permanent DFU using dfu-util in the past, if I remember correctly. CC @jfischer-no
> 
> I have asked @nvlsianpu about this and it seems that the application is supposed to confirm itself.

yes, this is the responsibility of the application. If the application does not reach this point the update is rolled back.

de-nordic, 2020-11-26:
@nehalrohit Do you find the https://github.com/zephyrproject-rtos/zephyr/issues/29921#issuecomment-729542160 information satisfactory? Can the issue be closed now?

nehalrohit, 2020-11-27:
hi @de-nordic 
Thanks, I resolved my issue.

closed 2020-11-27 by de-nordic

cederom, 2020-12-11:
Hello world, I just want to add that `nrfutil` (insalled with Python's `pip`) works for me on FreeBSD to generate ZIP DFU and then program the firmware using NRF DFU :-)