zephyrproject-rtos/zephyr #29925: pinctrl error for disco_l475_iot1 board:

mtahirbutt, 2020-11-10:
**Describe the bug**
A clear and concise description of what the bug is.
What have you tried to diagnose or workaround this issue?

I tried to run sample/hello_world program on disco_l75_iot1 but on it does not compile as it expects 
<st/l4/stm32l475v(c-e-g)tx-pinctrl.dtsi> but it is not there in the specified path. I guess I have to 
provide implementation details?

I tried to use previous definition of disco_l475_iot1 implementation along with pinmux.c that worked but 
it is now problematic with fresh commits in the last 1-4 weeks.

**To Reproduce**
Steps to reproduce the behavior:
1. mkdir build; cd build
2. cmake -DBOARD=disco_l475_iot1\_xyz
3. make
4. See error

**Expected behavior**
A clear and concise description of what you expected to happen.

**Impact**
What impact does this issue have on your progress (e.g., annoyance, showstopper)

I want to create a PR but the problem is now with the recent commits in the last 1-3 (i guess) weeks.
**Logs and console output**
If applicable, add console logs or other types of debug information
e.g Wireshark capture or Logic analyzer capture (upload in zip archive).
copy-and-paste text and put a code fence (\`\`\`) before and after, to help
explain the issue. (if unable to obtain text log, add a screenshot)

/home/tahir/fork-new/https:/github.com/mtahirbutt/zephyr/zephyr/boards/arm/disco_l475_iot1/disco_l475_iot1.dts:9:10: fatal error: st/l4/stm32l475v(c-e-g)tx-pinctrl.dtsi: No such file or directory
    9 | #include <st/l4/stm32l475v(c-e-g)tx-pinctrl.dtsi>
      |          ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
compilation terminated.

**Environment (please complete the following information):**
 - OS: (e.g. Linux)
 - Toolchain (e.g Zephyr SDK, 0.11.4)
 - Commit SHA used: e51226b82779409fbbcc7fb9fa2d59bde7df0a71
   zephyr 2.4.99.

**Additional context**
Add any other context about the problem here.
If i have to provide implementation, it seems a bit of work.

gmarull, 2020-11-10:
Can you try `west update`?

mtahirbutt, 2020-11-10:
thanks a lot. it works. but was wondering I guess stm32l475v(c-e-g)tx-pinctrl.dtsi should be the include path but it is not.

gmarull, 2020-11-10:
@mtahirbutt These pinctrl files are part of the STM32 HAL, so you likely had an old version.

mtahirbutt, 2020-11-10:
@gmarull, thanks for the tip. anyways it works.

closed 2020-11-10 by mtahirbutt