zephyrproject-rtos/zephyr #29936: XCC Build isr_tables.c fail

hongshui3000, 2020-11-11:
**Describe the bug**

build FAIL:
file : sections.h
```
#define _SW_ISR_TABLE_SECTION_NAME	.gnu.linkonce.sw_isr_table*
```
build PASS
file : sections.h
```
#define _SW_ISR_TABLE_SECTION_NAME	.gnu.linkonce.sw_isr_table
```

**To Reproduce**
Steps to reproduce the behavior:
west build -b qemu_xtensa samples/hello_world

**Logs and console output**
FAILED: zephyr/arch/common/CMakeFiles/isr_tables.dir/isr_tables.c.obj 
C:\usr\xtensa\XtDevTools\install\tools\RI-2020.5-win32\XtensaTools\bin\xt-xcc.exe -DBUILD_VERSION=zephyr-v2.4.0-1113-g0bcd66ac36d8 -DKERNEL -D_FORTIFY_SOURCE=2 
-D__ZEPHYR_SUPERVISOR__ -D__ZEPHYR__=1 -I../kernel/include -I../arch/xtensa/include -I../include -Izephyr/include/generated -I../soc/xtensa/sample_controller -I../soc/xtensa/sample_controller/include -IE:/work/rtos_work/zephyrproject/modules/hal/xtensa/include -IE:/work/rtos_work/zephyrproject/modules/hal/xtensa/zephyr/soc/sample_controller -isystem ../lib/libc/minimal/include -Os -imacros E:/work/rtos_work/zephyrproject/zephyr/build/zephyr/include/generated/autoconf.h -ffreestanding -fno-common -g -imacrosE:/work/rtos_work/zephyrproject/zephyr/include/toolchain/xcc_missing_defs.h -fms-extensions -imacros E:/work/rtos_work/zephyrproject/zephyr/include/toolchain/zephyr_stdint.h -Wall -Wformat -Wformat-security -Wno-format-zero-length -Wno-main -Wno-pointer-sign -Wpointer-arith -Werror=implicit-int -fno-pic -fno-strict-overflow -ffunction-sections -fdata-sections -mlongcalls -std=c99 -MD -MT zephyr/arch/common/CMakeFiles/isr_tables.dir/isr_tables.c.obj -MF zephyr\arch\common\CMakeFiles\isr_tables.dir\isr_tables.c.obj.d -o zephyr/arch/common/CMakeFiles/isr_tables.dir/isr_tables.c.obj -c E:/work/rtos_work/zephyrproject/zephyr/arch/common/isr_tables.c
C:/Users/xxx/AppData/Local/Temp/cc0spin#1bce.a11124: Assembler messages:
C:/Users/xxx/AppData/Local/Temp/cc0spin#1bce.a11124:220: Warning: missing operand; zero assumed
C:/Users/xxx/AppData/Local/Temp/cc0spin#1bce.a11124:220: Error: invalid operands (*UND* and *ABS* sections) for `*'
xt-xcc.exe ERROR:  C:/usr/xtensa/XtDevTools/install/tools/RI-2020.5-win32/XtensaTools/bin/xt-as.exe returned non-zero status 1
[54/66] Building C object zephyr/lib/posix/CMakeFiles/lib__posix.dir/pthread_common.c.obj
ninja: build stopped: subcommand failed.
FATAL ERROR: command exited with status 1: 'C:\Program Files\CMake\bin\cmake.EXE' --build 'E:\work\rtos_work\zephyrproject\zephyr\build'

**Environment (please complete the following information):**
 - OS: Windows
 - Toolchain XCC

andrewboie, 2020-11-13:
Would you like to send a PR which addresses this?

nashif, 2021-01-24:
@dcpleung  can you please take a look?

PR #31600 (2021-01-26): linker: remove asterisk from IRQ/ISR section name macro
Both _IRQ_VECTOR_TABLE_SECTION_NAME and _SW_ISR_TABLE_SECTION_NAME
are defined with asterisk at the end in an attempt to include
all related symbols in the linker script. However, these two
macros are also being used in the source code to specify
the destination sections for variables. Asterisks in the name
results in older GCC (4.x) complaining about those asterisks.
So create new macros for use in linker script, and keep
the names asterisk free.

Fixes #29936
commit 0d099bdd (2021-01-26): linker: remove asterisk from IRQ/ISR section name macro

Both _IRQ_VECTOR_TABLE_SECTION_NAME and _SW_ISR_TABLE_SECTION_NAME
are defined with asterisk at the end in an attempt to include
all related symbols in the linker script. However, these two
macros are also being used in the source code to specify
the destination sections for variables. Asterisks in the name
results in older GCC (4.x) complaining about those asterisks.
So create new macros for use in linker script, and keep
the names asterisk free.

Fixes #29936
closed 2021-01-26 by nashif