zephyrproject-rtos/zephyr #29941: Unable to connect Leshan LwM2M server using x86 based LwM2M client

Hard1kPrajapat1, 2020-11-11:
Hi Team,

I am running one issue with the Leshan server. Please help.

I have downloaded the zephyr project from the below link to explore the LwM2M client-server scenario:
https://github.com/zephyrproject-rtos/zephyr

**Summary:**
I have compiled and run the lwm2m client sample for 'x86' and I was able to connect the Leshan demo server running locally in the same Linux host machine. 
But as soon as I add the URI of the Leshan LwM2M server in the client code which is running remotely, I am unable to connect it.

**Working setup locally:**
- Run the lwm2m client for x86 on a Linux machine.
- Run the Lwm2m server locally using "leshan-server-demo.jar and I can see the client connected locally at http://localhost:8080/

**A problem in connection of remote server:**
- In the client app, the URI of the lwm2m server "coap://leshan.eclipseprojects.io:5683" is added to connect.
- Compiled the app for x86 and tried to run.
- But I am unable to connect to the server, I am getting the below error:
`net_lwm2m_rd_client: Cannot init LWM2M engine (-22)`

Also, I have tried other links too like the following but no success:
- coap://leshan.eclipse.org:5783
- coap://leshan.iot-lab.info
- coap://5.39.83.206

Can you please help me with what I am missing here? I am attaching the whole error log for more information.
[Lwm2m_server_connection_error.txt](https://github.com/eclipse/leshan/files/5523417/Lwm2m_server_connection_error.txt)

Thanks,
Hardik

Hard1kPrajapat1, 2020-11-13:
Hi,

I have also contacted to Leshan team, and I don't think the issue is from the Leshan LwM2M server side. I am suspecting the configuration issue, so please help me to solve this.

I have confirmed from the Leshan team that the public server (https://leshan.eclipseprojects.io/) is proven. I have also tried to connect to it using `leshan-client-demo.jar` and I was able to connect it. But as soon as I add it to the code I am unable to connect it.

Please guide me where I left unchanged, also check my environment setup as following:
- Before I start the client in the host machine, I have to start `loop-socat.sh` and `loop-slip-tap.sh`.
- In the configuration file (prj.conf), I have updated the host IP and gateway IP.
- I have added the IPv4 address of the https://leshan.eclipseprojects.io/ in the lwm2m client code like following:
`lwm2m_engine_set_string("0/0/0", "coap://23.97.187.154:5683")`.
- And then it goes for the registration but unable to register with the Leshan server.

I have also captured the logs, please check this and let me know If I have missed anything. 
[LwM2M_zephyr_client_server_error.docx](https://github.com/zephyrproject-rtos/zephyr/files/5534818/LwM2M_zephyr_client_server_error.docx)

Thanks and Regards,
Hardik

tlr-id, 2020-11-13:
Hi !

I'm trying to do approximately the same at you, which is connecting a LWM2M device to a remote server.

My problem is somewhere else but I think I can solve your issue. We can advance together toward a solution if you want.

You have this error. I think I've already encountered her, and some people also.
```[00:00:01.530,000] <dbg> net_lwm2m_engine.lwm2m_parse_peerinfo: Parse url: <log_strdup alloc failed>```  
You can find the solution here. Worked for him, worked for me, should work for you.
https://lists.zephyrproject.org/g/devel/topic/76243642?p=,,,20,0,0,0::recentpostdate%2Fsticky,,,20,2,0,76243642
A Push Request should be done to solve this issue btw.

Now, you will most likely encounter an issue when you'll solve this.
You connect with your client to the Leshan Demo Server using coaps, which means that DTLS is on (overlay-dtls.conf to enable/desable). 
But the https://leshan.eclipseprojects.io/ is, as you said, for a coap purpose. 
Ain't managed to connect to it yet, looking for a way to do it.
It is the same for IOWA's server : https://iowa-server.ioterop.com/device

Also Hardik, I suggest that you select your IP address using the prj.conf file of your project. I don't know if that changes a lot of things, but I think the ```CONFIG_NET_CONFIG_PEER_IPV4_ADDR``` is here for a reason.

Good luck to you, hopefully I've been useful !

Regards,

Thomas

rlubos, 2020-11-17:
Hi Hardik, Thomas,

As for the networking with qemu, please make sure you configure the iptables, forwarding and DNS as described here:
https://docs.zephyrproject.org/1.13.0/subsystems/networking/qemu_setup.html#setting-up-zephyr-and-nat-masquerading-on-host-to-access-internet
(personally I just have a script that calls all of these commands before I run the other scripts from net-tools).

As for the lwm2m_client, I see that you've already tried two attempts, modifying `CONFIG_NET_CONFIG_PEER_IPV4_ADDR` or setting the URI in the application. Both approaches work, but they need proper configuration.

1. In case you modify `CONFIG_NET_CONFIG_PEER_IPV4_ADDR` you need to make sure that IPv6 is disabled in the sample, otherwise it will use the IPv6 address by default. You can disable IPv6 with the following:
```
CONFIG_NET_IPV6=n
CONFIG_NET_CONFIG_NEED_IPV6=n
```
Otherwise, it's enough to add the following to the prj.conf in order to reach the Leshan public server:
```
CONFIG_NET_CONFIG_PEER_IPV4_ADDR="23.97.187.154"
CONFIG_NET_CONFIG_MY_IPV4_GW="192.0.2.2"
```

2. If you set the URI manually in the application (for instance with `#define SERVER_ADDR "leshan.eclipseprojects.io"`), you have to enable DNS resolver in the sample in order to obtain the IP address. You can do this with the following configuration (note that heap is needed for `getaddrinfo()` to work):
```
CONFIG_DNS_RESOLVER=y
CONFIG_DNS_SERVER_IP_ADDRESSES=y
CONFIG_DNS_SERVER1="192.0.2.2"
CONFIG_MINIMAL_LIBC_MALLOC_ARENA_SIZE=2048
```

tlr-id, 2020-11-17:
Hi rlubos,

I've done those steps already, and managed to reach out a remote private network which is using coaps. 
Therefore I've already disabled IPv6 and made changes in my prj.conf file.

I'm now trying to log in a remote public network such as leshan eclipse or IOWA which doesn't have DTLS encryption.

I've not tried enough things yet, I'll edit this message whenever I've made it, or when the issue will be precised.

Thanks a lot for the informations tho.

Bests regards,

Thomas

rlubos, 2020-11-17:
@tlr-id The Leshan public server supports both, CoAP and CoAPs. If you wish to use CoAPs, you need to configure the security material for your device [here](https://leshan.eclipseprojects.io/#/security). The sample documentation explains how to configure PSK:
https://docs.zephyrproject.org/latest/samples/net/lwm2m_client/README.html#dtls-support

tlr-id, 2020-11-17:
@rlubos 

Thanks a lot ! I've succeeded in connecting my client to the public Leshan using CoAPs, and the mistake was as dumb as not configuring the Client's profile.

Do you have any idea of how to connect to the Leshan using CoAP ? 
I've tried several things such as compiling without the ```overlay-dtls.conf``` file, or changing these :
```
CONFIG_LWM2M_DTLS_SUPPORT=y    =>   CONFIG_LWM2M_DTLS_SUPPORT=n
CONFIG_LWM2M_PEER_PORT=5684    =>   CONFIG_LWM2M_PEER_PORT=5683
```

Also, do you have any idea on how to reach the IOWA's public server ?
( ```https://iowa-server.ioterop.com/device``` and ip is currently ```18.195.192.63```). 
I've tried to add a new client configuration using PSK, no to Oscore, Default CoAP settings and TLV for Payload format and response.

Thanks again for your help,

Bests Regards,

Thomas

rlubos, 2020-11-17:
@tlr-id 

The sample by default connects w/o security, so compiling with no overlay should do the trick. You may need to delete the build directory though if you compiled with an overlay enabled (I'm not sure how good Kconfig is with recompilation if overlays changed). Plus, if you've already configured your device with secuirty credentials in Leshan, you need to remove it first, otherwise the server will reject your device.

As for the IOWA's public server, I don't have experience with this server. I've tried to connect with no security and it worked with no issues. I'm not being able to connect with DTLS enabled though, the handshake keeps failing, I'm not sure why yet.

rlubos, 2020-11-17:
> I'm not being able to connect with DTLS enabled though, the handshake keeps failing, I'm not sure why yet.

Ok I get it now, this server accepts PSK in base64 format. I've used
```
static unsigned char client_psk[] = {
	'a', 'b', 'c', 'd'
};
```
in the sample and it's base64 representation on the server side
```
YWJjZA==
```

Hard1kPrajapat1, 2020-11-19:
Hi @rlubos, @tlr-id,

Sorry for the delayed response, I was on Diwali vacation. Now I have resumed.

@tlr-id,
I tried the solution you shared and now I don't see an error while parsing the URL (`net_lwm2m_engine.lwm2m_parse_peerinfo: Parse url: <log_strdup alloc failed>`), it was a strdup allocation issue. 
But I am still unable to connect to the Leshan server. Also, I am not able to ping DNS `net ping 8.8.8.8` or `net ping <my GW IP>`, it shows ping timeout. Can you please help with this?

@rlubos,
I have tried the steps you mentioned for using IPv4 and using URI.
1) Using peer address in prj.conf file:
- I have already disabled the IPv6 interface from prj.conf file as you mentioned.
- Removed the URI in the code.
- Added the IPv4 address of the Leshan server to prj.conf
> CONFIG_NET_CONFIG_PEER_IPV4_ADDR="23.97.187.154"
> CONFIG_NET_CONFIG_MY_IPV4_GW="10.100.104.1"
- Removed build directory and recompiled it.
- Still, I am unable to connect with the Leshan server.
- Please check the logs from [LwM2M_zephyr_client_server_error2.txt](https://github.com/zephyrproject-rtos/zephyr/files/5564913/LwM2M_zephyr_client_server_error2.txt)

2) Using URI in code:
- Removed the `CONFIG_NET_CONFIG_PEER_IPV4_ADDR` from the prj.conf file.
- Added URI in the code `#define SERVER_ADDR "leshan.eclipseprojects.io"`.
- Enabled the DNS resolver in prj.conf as well.
> CONFIG_DNS_RESOLVER=y
> CONFIG_DNS_SERVER_IP_ADDRESSES=y
> CONFIG_DNS_SERVER1="10.100.104.1"
> CONFIG_MINIMAL_LIBC_MALLOC_ARENA_SIZE=2048
- But it shows "Unable to resolve address", and I am not able to connect it.
- Please check the logs from [LwM2M_zephyr_client_server_URI_DNS_resolver.txt](https://github.com/zephyrproject-rtos/zephyr/files/5564951/LwM2M_zephyr_client_server_URI_DNS_resolver.txt)

By any of these two methods, I am unable to connect to the Leshan public server. Please help where I am missing the configuration.

Let me know If you need any file to check, I will send.

Thanks,
Hardik

Hard1kPrajapat1, 2020-11-19:
Hi,

Adding to my previous comment, I have noticed something that may be helpful for you to identify.
In my LwM2M client project, below IPv4 addresses are used for the different endpoints.
- Host machine's IPv4 where QEMU is running = 10.100.105.15
- Zephyr app's IPv4(CONFIG_NET_CONFIG_MY_IPV4_ADDR) = 10.100.105.18
- The tap0 IPv4 = 10.100.105.17
- The Leshan server IPv4 = 23.97.187.154

So when I start the app to connect to the Leshan server and from the Zephyr shell If I try to ping the server's IPv4 address then I am not able to ping it.
> uart:~$ net ping 23.97.187.154
> PING 23.97.187.154
> Ping timeout

From the Zephyr's shell, I am only able to ping 3 addresses(10.100.105.15, 10.100.105.17, and 10.100.105.18), even I am not able to ping google's IPv4 address as well. Is the internet connectivity from the app creating an issue?

Thanks,
Hardik

rlubos, 2020-11-19:
@Hard1kPrajapat1 It seems that you use non-default IP addresses, did you set up NAT masquerading for your qemu IP address as explained in the [guide](https://docs.zephyrproject.org/1.13.0/subsystems/networking/qemu_setup.html#setting-up-zephyr-and-nat-masquerading-on-host-to-access-internet)?
I. e. I mean this command:
```
iptables -t nat -A POSTROUTING -j MASQUERADE -s 10.100.105.18
```

Hard1kPrajapat1, 2020-11-19:
Hi Robert,

I just got it working. And Yes, I have already applied the NAT masquerading command before.

**Working:**
- I am able to connect with the Leshan server by configuring the gateway address is the same as tap0's IPv4 address in prj.conf.
- I changed from **CONFIG_NET_CONFIG_MY_IPV4_GW=10.100.104.1** to **CONFIG_NET_CONFIG_MY_IPV4_GW=10.100.105.17**. Where 10.100.104.1 was my organization's gateway address and 10.100.105.17 was tap0 interface's IPv4 address.
- Can you please explain why I had to apply tap0's address as a gateway address?

**Still not working:**
- If I remove the peer address from the prj.conf file and add the URI in the code like below, I am still unable to resolve the address.
> #define SERVER_ADDR "leshan.eclipseprojects.io"
- It shows me the error like following (already shared error file before):
> [00:00:03.040,000] <err> net_lwm2m_engine: Unable to resolve address                                                                                                        > [00:00:03.040,000] <err> net_lwm2m_rd_client: Cannot init LWM2M engine (-2)
- I have added the following lines in the prj.conf for the DNS.
> CONFIG_DNS_RESOLVER=y
> CONFIG_DNS_SERVER_IP_ADDRESSES=y
> CONFIG_DNS_SERVER1="10.100.104.1"
> CONFIG_MINIMAL_LIBC_MALLOC_ARENA_SIZE=2048

Please let me know what I am missing here.

Thanks,
Hardik

Hard1kPrajapat1, 2020-11-20:
Hi,

Please check my below observations and issues till now and guide me:
1) I am able to connect to the Leshan LwM2M server without security. Can you please explain why the gateway IP address should be the same as tap0's address to make it work?
2) I am unable to connect to the Leshan LwM2M server with DTLS security as mentioned in the [Link](https://docs.zephyrproject.org/latest/samples/net/lwm2m_client/README.html#dtls-support).
Issue: 
It fails and the handshake process so could not register, please check the logs [LwM2M_zephyr_client_server_DTLS_error.txt](https://github.com/zephyrproject-rtos/zephyr/files/5571589/LwM2M_zephyr_client_server_DTLS_error.txt)
3) I am able to connect with the Leshan bootstrap server but then it is not going to connect to the Leshan LwM2M server. 
Issue:
- I have added the configuration to connect to the server on the bootstrap server.
- I am able to register my machine with the bootstrap server but from the bootstrap server, it should connect me to the LwM2M server which is not going to do.

Can you please help me what am I missing here?

Thanks and Regards,
Hardik

rlubos, 2020-11-20:
1. I guess the gateway needs to be in the same subnet as your qemu instance? Otherwise, how would Zephyr know where to route packets if the gateway is in another network? Perhaps someone with more IPv4 knowledge could confirm.
2. The handshake failed with `MBEDTLS_ERR_SSL_TIMEOUT`. Are you sure you use the correct port number? Or perhaps firewall in your network blocks the traffic on ports used by LwM2M? For instance, that's the case at my workplace - the CoAP traffic is blocked after a few transactions.
3. I cannot provide any feedback w/o more details. Do you use a secure or non-secure connection for Bootstrap and LwM2M server? Has the bootstrap exchange finished successfully?

In general, it would be good to get wireshark pcaps in such cases to see where exactly the communication gets stuck.

Hard1kPrajapat1, 2020-11-20:
Hi Robert,

1) It was in the same subnet, my machine's IP is 10.100.105.15. Hence, I have provided 10.100.105.18 for Zephyr App, provided 10.100.105.17 for the tap0 interface and the gateway of my workplace was 10.100.104.1. But applying the 10.100.104.1 IP as GW, it was not working. As soon as I gave tap0's IP, it started working.
2) Yes, I am compiling the sample with `overlay-dtls.conf`, so it will take port 5684 as default. For the firewall, I will confirm and let you know.
3) I was using non-secure communication, and yes, the bootstrap registration was successful. Please check the logs from [LwM2M_zephyr_client_server_bootstrap.txt](https://github.com/zephyrproject-rtos/zephyr/files/5572958/LwM2M_zephyr_client_server_bootstrap.txt)

Please let me know If I can provide you more detail.

Thanks,
Hardik

rlubos, 2020-11-20:
1. Note that the default subnet mask in Zephyr is `/24` so `10.100.105.0` and `10.100.104.0` are different subnets from Zephyr perspective. The same goes for the tap interface, it's `/24` by default. Perhaps that's the issue? For Zephyr there's `NET_CONFIG_MY_IPV4_NETMASK` to change subnet mask, not sure about tap.
I'm also not sure if ARP packets will actually be forwarded from tap0 interface to others, perhaps @jukkar you know?
2. Nothing to add.
3. From your logs it doesn't seem that the bootstrap procedure was complete (registration is just a first step). Here's how the log from a complete bootstrap procedure looks like:
```
[00:00:00.110,000] <inf> net_lwm2m_rd_client: Start LWM2M Client: qemu_x86
[00:00:01.020,000] <dbg> net_lwm2m_engine.lwm2m_engine_get: path:0/0/1, buf:0x14fb6b, buflen:1
[00:00:01.020,000] <inf> net_lwm2m_rd_client: Bootstrap started with endpoint 'qemu_x86' with client lifetime 0
[00:00:01.020,000] <dbg> net_lwm2m_engine.lwm2m_parse_peerinfo: Parse url: coap://[2001:db8::2]
[00:00:01.020,000] <dbg> net_lwm2m_rd_client.sm_send_bootstrap_registration: Register ID with bootstrap server as 'ep=qemu_x86'
[00:00:01.530,000] <dbg> net_lwm2m_engine.lwm2m_udp_receive: checking for reply from [2001:db8::2]
[00:00:01.530,000] <dbg> net_lwm2m_rd_client.do_bootstrap_reply_cb: Bootstrap callback (code:2.4)
[00:00:01.530,000] <inf> net_lwm2m_rd_client: Bootstrap registration done!
[00:00:01.530,000] <dbg> net_lwm2m_client_app.rd_client_event: Bootstrap registration complete
[00:00:01.530,000] <dbg> net_lwm2m_engine.lwm2m_udp_receive: reply 0x134154 handled and removed
[00:00:01.530,000] <dbg> net_lwm2m_engine.lwm2m_udp_receive: checking for reply from [2001:db8::2]
[00:00:01.530,000] <dbg> net_lwm2m_engine.handle_request: No accept option given. Assume OMA TLV.
[00:00:01.530,000] <dbg> net_lwm2m_engine.lwm2m_udp_receive: checking for reply from [2001:db8::2]
[00:00:01.530,000] <dbg> net_lwm2m_engine.handle_request: No accept option given. Assume OMA TLV.
[00:00:01.550,000] <dbg> net_lwm2m_engine.lwm2m_udp_receive: checking for reply from [2001:db8::2]
[00:00:01.550,000] <dbg> net_lwm2m_engine.handle_request: No accept option given. Assume OMA TLV.
[00:00:01.550,000] <dbg> net_lwm2m_obj_security.security_create: Create LWM2M security instance: 0
[00:00:01.550,000] <dbg> net_lwm2m_engine.lwm2m_udp_receive: checking for reply from [2001:db8::2]
[00:00:01.550,000] <dbg> net_lwm2m_engine.handle_request: No accept option given. Assume OMA TLV.
[00:00:01.560,000] <dbg> net_lwm2m_engine.lwm2m_udp_receive: checking for reply from [2001:db8::2]
[00:00:01.560,000] <dbg> net_lwm2m_engine.handle_request: No accept option given. Assume OMA TLV.
[00:00:01.560,000] <dbg> net_lwm2m_obj_server.server_create: Create LWM2M server instance: 0
[00:00:01.570,000] <dbg> net_lwm2m_engine.lwm2m_udp_receive: checking for reply from [2001:db8::2]
[00:00:01.570,000] <inf> net_lwm2m_rd_client: Bootstrap data transfer done!
[00:00:02.040,000] <dbg> net_lwm2m_client_app.rd_client_event: Bootstrap transfer complete
[00:00:02.560,000] <dbg> net_lwm2m_engine.lwm2m_engine_get: path:0/0/1, buf:0x14fb6b, buflen:1
[00:00:02.560,000] <dbg> net_lwm2m_engine.lwm2m_engine_get: path:0/1/1, buf:0x14fb6b, buflen:1
[00:00:02.560,000] <dbg> net_lwm2m_engine.lwm2m_engine_get: path:0/1/10, buf:0x14fbca, buflen:2
[00:00:02.560,000] <inf> net_lwm2m_rd_client: RD Client started with endpoint 'qemu_x86' with client lifetime 0
[00:00:02.560,000] <dbg> net_lwm2m_engine.lwm2m_parse_peerinfo: Parse url: coap://[2001:db8::2]:5683
[00:00:02.560,000] <dbg> net_lwm2m_engine.lwm2m_engine_get: path:1/0/1, buf:0x14fb9c, buflen:4
[00:00:02.570,000] <dbg> net_lwm2m_rd_client.sm_send_registration: registration sent [2001:db8::2]
[00:00:03.070,000] <dbg> net_lwm2m_engine.lwm2m_udp_receive: checking for reply from [2001:db8::2]
[00:00:03.070,000] <dbg> net_lwm2m_rd_client.do_registration_reply_cb: Registration callback (code:2.1)
[00:00:03.070,000] <dbg> net_lwm2m_client_app.rd_client_event: Registration complete
[00:00:03.070,000] <inf> net_lwm2m_rd_client: Registration Done (EP='kdFTleuHGM')
```
It seems that for whatever reason the some of the packets from the LwM2M server did not reach your qemu instance (could be the firewall mentioned). If you register w/o bootstrap and DTLS, can you exchange packets with the server after you register? Like reading multiple objects/resoures.

jukkar, 2020-11-20:
>     1. I'm also not sure if ARP packets will actually be forwarded from tap0 interface to others, perhaps @jukkar you know?

I don't think ARP packets have any routing capabilities.

It looks like that either the netmask needs to be adjusted, or gateway to be set if the networks are in different subnets, which seems to be the case here.

rlubos, 2022-03-02:
This issue has gotten pretty old w/o any activity, please reopen if needed.

closed 2022-03-02 by rlubos