zephyrproject-rtos/zephyr #29946: SD card initialization is wrong

trembel, 2020-11-11:
Trying to initialize a Samsung Evo Plus SD card it failed. Other SD cards worked just fine.

Looking at a LA the problem seems to be the selection of the SPI mode:
After insertion, 74 clock pulses needs to be applied to SCLK, with holding CS and DI high.
According to what I see on the LA, this is not the case since CS is low:
[image]

trembel, 2020-11-11:
IMO the error happens, because the initialization is done in `disk_access_spi_sdhc.c` in the function `static int sdhc_spi_go_idle(struct sdhc_spi_data *data)`:
```
sdhc_spi_tx(data, sdhc_ones, 10);
spi_release(data->spi, &data->cfg);
return sdhc_spi_cmd_r1_idle(data, SDHC_GO_IDLE_STATE, 0);
```
It writes 10bytes of ones using SPI onto the line, which is equivalent to MOSI being High and the clock toggling 80 times. However this also implies that CS is low.

carlescufi, 2021-02-09:
@trembel thanks for the analysis. Have you tested your proposed solution? If so, and if it works, could you propose a PR, even if it's a Draft or RFC?

trembel, 2021-02-16:
I've for now switched to the nRF5 SDK, since I had not enough time to fix this. One possibility would be to use an "invalid" pin as CS for the first toggles, however I could not find out if something like this exists in Zephyr.

jfischer-no, 2021-05-14:
> Looking at a LA the problem seems to be the selection of the SPI mode:
> After insertion, 74 clock pulses needs to be applied to SCLK, with holding CS and DI high (http://elm-chan.org/docs/mmc/mmc_e.html#spiinit).
> According to what I see on the LA, this is not the case since CS is low:

From Physical Layer Simplified Specification, **6.4.1 Power Up Sequence for SD Bus Interface**:
"_A device shall be ready to accept the first command within 1ms from detecting VDD min.
Device may use up to 74 clocks for preparation before receiving the first command._"

Does not look like it is a requirement.

Switching to SPI Mode is done by CMD0 reception over SPI, **7.2.1 Mode Selection and Initialization**:
"_The SD Card is powered up in the SD mode. It will enter SPI mode if the CS signal is asserted (negative)
during the reception of the reset command (CMD0)._"

carlescufi, 2021-05-19:
@trembel @richardbarlow as per @jfischer-no's analysis, for now we will close this issue since work done since this issue was posted can potentially fix the problem. If this is not the case we can reopen it.

closed 2021-05-19 by carlescufi