zephyrproject-rtos/zephyr #29954: intel_adsp_cavs18 fails with heap errors on current Zephyr

andyross, 2020-11-11:
@lyakh reports that the SOF application fails on intel_adsp_cavs18 hardware with current code.  If I understand correctly:

1. The https://github.com/andyross/zephyr/tree/sof-rebased-19Oct20 test branch works on 1.8 hardware.

2. https://github.com/andyross/zephyr/tree/sof-rebased-22Oct20 does not, failing with an error that appears to be heap related.

The main delta between these trees (other than a Zephyr rebase) is
that the first contains a full reversion of the heap optimization
series from @npitre (commit 63ee8517f7f1bc20a785a69032548abf9ab8acb8).

Manually reverting lib/os/heap to that state in the failing tree fixes
the issues (is that right @lyakh ?)

(This is a continuation of discussion started in https://github.com/thesofproject/sof/pull/3544)

andyross, 2020-11-11:
@lyakh @npitre I opened this bug in Zephyr to have a better place to discuss

andyross, 2020-11-11:
OK, so to start, we should clean up the trees and stop using those intermediate ones I was pushing for integration/demo work.  Virtually everything should be in zephyr/sof masters now.  All you need is to add the SOF app to Zephyr and then apply one trivial build fixup to SOF:

    Zephyr: https://github.com/thesofproject/sof/pull/3601
    SOF: https://github.com/thesofproject/sof/pull/3601

This works for me to get through SOF initialization on Up Squared (CAVS 1.5) hardware.

andyross, 2020-11-11:
Next, since I remain relatively certain this is going to turn out to be heap corruption, can you try something like this on your failing SOF tree and see if it acts to suppress the error:

```diff
diff --git a/zephyr/wrapper.c b/zephyr/wrapper.c
index 298160312..39ef575cd 100644
--- a/zephyr/wrapper.c
+++ b/zephyr/wrapper.c
@@ -61,6 +61,9 @@ static void *heap_alloc_aligned(struct k_heap *h, size_t align, size_t bytes)
 {
        void *ret = NULL;
 
+       // FIXME: pad buffer to hunt for overflows
+       bytes += 16;
+
        k_spinlock_key_t key = k_spin_lock(&h->lock);
 
        ret = sys_heap_aligned_alloc(&h->heap, align, bytes);
```

It's easy and cheap, and if "works" then it tells us the bug is probably on the SOF side.  We can then start digging through all the specific allocations at runtime and see if we can find the specific one that needs to be padded.  It's tedious, but there actually aren't that many; this is the sort of thing I was doing hunting for the original alignment bug.

andyross, 2020-11-11:
And finally: @lyakh can you push your current 1.8 changes somewhere I can look at them?  The last versions I have are the ones you pushed to my 17Sep branch, and that's very stale now.  At this stage, you should be able to submit directly to Zephyr master via pull request.  We're ready finally, I promise. :)

lyakh, 2020-11-11:
@andyross sorry, I was hacking away to test the Zephyr "master" and I've done that: took the current Zephyr master, applied your 2 SOF integration patches and my cAVS 1.8 work. To that I've taken your SOF tree https://github.com/thesofproject/sof/pull/3601 and also added my stuff on top. That still didn't work. I think I better stop for today, but I'll do what you asked about tomorrow.

lyakh, 2020-11-12:
> ```diff
> diff --git a/zephyr/wrapper.c b/zephyr/wrapper.c
> index 298160312..39ef575cd 100644
> --- a/zephyr/wrapper.c
> +++ b/zephyr/wrapper.c
> @@ -61,6 +61,9 @@ static void *heap_alloc_aligned(struct [...]

@andyross this didn't help

lyakh, 2020-11-12:
> And finally: @lyakh can you push your current 1.8 changes somewhere I can look at them? The last versions I have are the ones you pushed to my 17Sep branch, and that's very stale now. At this stage, [...]

@andyross Zephyr patches are in #29965. For reference I've pushed 2 branches with my current outstanding Zephyr+cAVS 1.8 SOF patches:
https://github.com/thesofproject/sof/compare/master...lyakh:topic/zephyr/master-cavs18 - based on master, contains your 3 patches + 3 more patches from me
https://github.com/thesofproject/sof/compare/stable-v1.6...lyakh:topic/zephyr/v1.6-cavs18 - based on stable-v1.6, contains 7 patches, backported from master, and the same 3 yours and 3 my patches as above.

lyakh, 2020-11-12:
...and here's a log:
```
131843295: validate(), ll drift detected, delta = 23824
131846660: dai_copy(): Copy_bytes 272 + avail bytes 0 < period bytes 384, possible glitch
131855198: dai_get type 3 index 4 new sref 2
131855654: dai_put type 3 index 4 new sref 1
131856149: dai_config() dai type = 3 index = 4
131856569: dai_config(), channel = 2
131862891: dai_playback_params() fifo 0x0
--- 3 messages dropped ---
131863333: pipe prepare
131866359: validate(), ll drift detected, delta = 23064
131868113: dai_copy(): Copy_bytes 368 + avail bytes 0 < period bytes 384, possible glitch
131890526: validate(), ll drift detected, delta = 24167
131913942: validate(), ll drift detected, delta = 23416
131917294: dai_copy(): Copy_bytes 352 + avail bytes 0 < period bytes 384, possible glitch
131937790: validate(), ll drift detected, delta = 23848
131961561: validate(), ll drift detected, delta = 23771
131963315: dai_copy(): Copy_bytes 352 + avail bytes 0 < period bytes 384, possible glitch
131969517: pipe trigger cmd 1
131970098: task add 0xbe055228 0x1fffa150U
131970621: task params pri 0 flags 0 start 0 period 1000
131971121: num_tasks 4 total_num_tasks 4
131985748: validate(), ll drift detected, delta = 24187
131989159: dai_copy(): Copy_bytes 352 + avail bytes 0 < period bytes 384, possible glitch
131992388: dai_get type 3 index 0 new sref 2
131992843: dai_put type 3 index 0 new sref 1
131993267: dai_config() dai type = 3 index = 0
131993674: dai_config(), channel = 3
131997770: pipe params stream_tag 4 channels 2 sample_valid_bytes 2 sample_container_bytes 2
--- 1 messages dropped ---
131999227: dai_playback_params() dest_dev = 0 stream_id = 0 src_width = 4 dest_width = 4
131999846: dai_playback_params() fifo 0x0
132000282: pipe prepare
[00006759] <err> os:  ** FATAL EXCEPTION
[00006759] <err> os:  ** CPU 0 EXCCAUSE 9 (load/store alignment)
[00006759] <err> os:  **  PC 0xbe01e907 VADDR 0x0000173b
[00006759] <err> os:  **  PS 0x00060a20
[00006759] <err> os:  **    (INTLEVEL:0 EXCM: 0 UM:1 RING:0 WOE:1 OWB:10 CALLINC:2)
[00006759] <err> os:  **  A0 0xbe01f808  SP 0xbe043b80  A2 0xbe049360  A3 0x0000173b
[00006759] <err> os:  **  A4 0xbe043ba0  A5 0xbe01e904  A6 0x00000000  A7 0x00004ec0
[00006759] <err> os:  **  A8 0xbe01e915  A9 0xbe043b60 A10 0x07de4e2e A11 0x00000000
[00006759] <err> os:  ** A12 0x00000000 A13 0xbe0492e0 A14 0xbe0492e0 A15 0x00000001
[00006759] <err> os:  ** LBEG 0xbe011fe9 LEND 0xbe011fef LCOUNT 0x00000000
[00006759] <err> os:  ** SAR 0x00000020
[00006759] <err> os: >>> ZEPHYR FATAL ERROR 0: CPU exception on CPU 0
[00006759] <err> os: Current thread: 0xbe0493a8 (unknown)
[00006759] <err> os: Halting system
```

lyakh, 2020-11-12:
The PC seems to be pointing here:
```
static bool timer_domain_is_pending(struct ll_schedule_domain *domain,
                                    struct task *task, struct comp_dev **comp)
{
be01e904:       004136                  entry   a1, 32
        return task->start <= platform_timer_get(timer_get());
be01e907:       0348                    l32i.n  a4, a3, 0
//^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
be01e909:       1338                    l32i.n  a3, a3, 4
        return sof_get()->platform_timer;
be01e90b:       fff765                  call8   be01e880 <sof_get>
be01e90e:       7aa8                    l32i.n  a10, a10, 28
be01e910:       120c                    movi.n  a2, 1
be01e912:       015be5                  call8   be01fed0 <platform_timer_get>
```

lyakh, 2020-11-12:
@andyross do I understand it correctly, that it's trying to load a 64 bit (split into two 32 bit) value(s) from 0x0000173b?

lyakh, 2020-11-12:
on a different occasion I've only got repeating
```
31474382: dmic_irq_handler(): full fifo A or PDM overrun
--- 3 messages dropped ---
31476987: dmic_irq_handler(): full fifo A or PDM overrun
--- 5 messages dropped ---
31481303: dmic_irq_handler(), OUTSTAT0 = 0x28000000, OUTSTAT1 = 0x80000000
--- 4 messages dropped ---
31484305: validate(), ll drift detected, delta = 23109
--- 4 messages dropped ---
31494886: dmic_irq_handler(): full fifo A or PDM overrun
--- 4 messages dropped ---
31496803: dmic_irq_handler(), OUTSTAT0 = 0x28000000, OUTSTAT1 = 0x80000000
```
in the log... weird

lyakh, 2020-11-12:
looks like a run-away DMA to me

andyross, 2020-11-12:
DMA scribbling would definitely be consistent with heap corruption, yeah.  And that assembly looks relatively straightforward.  Indeed, the task pointer is corrupt for sure.  But the corruption is probably of the environment upstream, not the task struct itself.

andyross, 2020-11-12:
I'm not seeing a lot of calls to the "domain_is_pending" callback interestingly.  The most likely seems to be in ll_schedule.c:schedule_ll_is_pending(), which is enumerating a list of tasks.  So for example that list might contain a bad item.

lyakh, 2020-11-13:
For a test I changed `HDA_DMA_BUFFER_ADDRESS_ALIGNMENT` from 128 to 256 and these problems disappeared. The HDA standard mandates 128 byte alignment, but I checked with a native SOF build and it never returns buffers with less than 256 byte alignment... @lgirdwood can it be that we indeed need 256-byte alignment on CNL?

lyakh, 2020-11-13:
SOF never hits this problem, because it "naturally" always allocates DMA data with 256-byte alignment, and that happens, because our allocator allocates DMA buffers from a heap of 256-byte chunks, which are all 256-byte aligned logically.

lyakh, 2020-11-27:
https://github.com/thesofproject/sof/pull/3645 fixes the problem. It can however also be a good idea to add generic cache-line alignment to the Zephyr allocator.

andyross, 2020-12-09:
Closing this one.  For clarity, the root cause was that SOF was making requests for N bytes of DMA memory and requesting that it be aligned to a cache line.  But it was not **padding** N to an integer number of cache lines, so the unused suffix memory could be used by the heap for other purposes, and clobbered by DMA.

closed 2020-12-09 by andyross