zephyrproject-rtos/zephyr #29958: mcuboot hangs when CONFIG_BOOT_SERIAL_DETECT_PORT value not found

captcha1, 2020-11-11:
**Describe the bug**

When building mcuboot w/ CONFIG_MCUBOOT_SERIAL=y , mcuboot hangs if CONFIG_BOOT_SERIAL_DETECT_PORT value is not found.  Eg : if the value is "gpioc" it hangs.  If the value is "GPIOC" it works.

**To Reproduce**
Steps to reproduce the behavior:

```
MCUBOOT_DIR=$ZEPHYR_BASE/../bootloader/mcuboot ; SRC1=$MCUBOOT_DIR/boot/zephyr

mkdir build; cd build
export BOARD=disco_l475_iot1 ; PORT0=gpioc ; PIN0=13 # label = "User"
west build $SRC1 -p always -- -DOVERLAY_CONFIG=usb_cdc_acm_log_recovery.conf  -DCONFIG_BOOT_SERIAL_DETECT_PORT=\"$PORT0\" -DCONFIG_BOOT_SERIAL_DETECT_PIN=$PIN0 
```

If you then flash that and run it, it hangs after printing "mcuboot: Starting bootloader" on the UART.  If you crawl around in gdb, you'll see that device_get_binding returned null and when gpio_pin_configure tries to dereference the pointer, "fatal" is called. Oddly, there is an __ASSERT in the code (right after "device_get_binding"), but it does not catch this (dunno why).

**Expected behavior**

If you change PORT0=GPIOC and re-run the "west", then it works (lots of messages, if "user" button is pressed during reset, then "mcuboot: Enter the serial recovery mode")

**Impact**

Annoyed, but felt better when I found the problem...

**Environment (please complete the following information):**
 - OS: Fedora 33 !
 - Toolchain : gcc-arm-none-eabi-9-2020-q2-update
 - Version : Zephyr 2.4.0

**Additional context**

Barometric pressure : 1019.62 hPa

erwango, 2020-11-17:
@captcha1 From what I can see BOOT_SERIAL_DETECT_PORT expects a GPIO device name so for STM32 SoCs it would be "GPIO[A-K]". So I think you figured it right.

Could you explain what you'd like to be fixed ?

captcha1, 2020-11-19:
FYI, for this discussion : MCUBOOT_DIR=$ZEPHYR_BASE/../bootloader/mcuboot ; SRC1=$MCUBOOT_DIR/boot/zephyr

Suggestions : 

In $SRC1/main.c , the line : `

    __ASSERT(detect_port, "Error: Bad port for boot serial detection.\n");

is a no-op because CONFIG_ASSERT=n (and if CONFIG_ASSERT=y, then the build fails with : section `rodata' will not fit in region `FLASH').

I would suggest an error message that is not a no-op.  Better would be to call cmd_device_list from $ZEPHYR_BASE/subsys/shell/modules/device_service.c

Also, when gpio_pin_configure tries to dereference the pointer, it calls z_arm_usage_fault (in $ZEPHYR_BASE/arch/arm/core/aarch32/cortex_m/fault_s.S), which calls "log" routines, but none of the logs get send to the console because it ends in arch_system_halt, which hangs in a busy wait.

So ... it seems like if an error routine calls a log routine, it should get sent to the console.  Dunno why it doesn't. Perhaps because mcuboot has its own log processing thread.

Perhaps code more special cases in $SRC1/Kconfig or $SRC1/boards/$BOARD.conf

Values I've found so far : 

BOARD=particle_xenon ; PORT=GPIO_0 ; PIN=11 # label = "Mode Button"
BOARD=frdm_k64f ; PORT=GPIO_2; PIN=6 # PTC6, label = "User SW2"
BOARD=disco_l475_iot1 ; PORT=GPIOC ; PIN=13 # label = "User"

FYI, it seems the names for the GPIO devices are set in $ZEPHYR_BASE/drivers/gpio/gpio_*.c .  Some are not obvious.  Eg, in gpio_mcux.c, "portc" gets the name "GPIO_2". The only way I've found to find the names of devices is to build $ZEPHYR_BASE/samples/subsys/shell/shell_module and run "devices list".

erwango, 2020-11-20:
Sounds like a pure pure issue.
@utzig do you know if this "issue" should be addressed here or in https://github.com/mcu-tools/mcuboot directly ?

utzig, 2020-11-20:
> @utzig do you know if this "issue" should be addressed here or in https://github.com/mcu-tools/mcuboot directly ?

Yes, it should. I will assign to @nvlsianpu because he maintains the Zephyr port.

captcha1, 2020-11-21:
More of an RFE, but maybe instead of specifying the "pin" and "port", just specify the device tree node somehow (using the "label"?) and then look up the "pin" using DT_GPIO_PIN and the "port" using DT_GPIO_LABEL.

closed 2021-02-05 by github-actions[bot]