zephyrproject-rtos/zephyr #29960: Checkpatch compliance errors do not fail CI

cvinayak, 2020-11-12:
**Describe the bug**
Checkpatch.pl compliance errors in commit description style are not flagged in PR CI builds.

**To Reproduce**
Steps to reproduce the behavior:
1. Submit a PR with commit description style error, example in 013cacf91ed0afc8ada2b4d8e56b4e0684b1674c

**Expected behavior**
CI compliance check shall fail the PR, and restrict merging of the commits upstream.

**Impact**
showstopper.

**Logs and console output**
Local failure:
[image]

**Environment (please complete the following information):**
 - OS: Linux
 - Toolchain: Zephyr SDK
 - Commit SHA or Version used: 89f6e24590

**Additional context**
NA

carlescufi, 2020-11-12:
Please consider that checkpatch is invoked on a diff, not on a commit in CI:
https://github.com/zephyrproject-rtos/zephyr/blob/master/scripts/ci/check_compliance.py#L205

nashif, 2020-11-12:
do not see the issue here, can you show that this error occurs in CI? What happens locally in your tree is not the same thing in CI, hence the local error.

cvinayak, 2020-11-13:
@nashif Take a look at the commit body of https://github.com/zephyrproject-rtos/zephyr/commit/013cacf91ed0afc8ada2b4d8e56b4e0684b1674c, its missing the title for the referenced commit hash, is it acceptable to be merged to master?

cvinayak, 2020-11-13:
> 
> 
> Please consider that checkpatch is invoked on a diff, not on a commit in CI:
> https://github.com/zephyrproject-rtos/zephyr/blob/master/scripts/ci/check_compliance.py#L205

`./scripts/checkpatch.pl -g HEAD...upstream/master` this how I run when I have multiple commits, so that commit message too is checked for compliance. 

I think lately typo's and commit description style are not flagged by CI anymore.

nashif, 2020-11-13:
> I think lately typo's and commit description style are not flagged by CI anymore.

commit messages in CI are not checked by checkpatch, we have gitlint for this.

cvinayak, 2020-11-15:
ok, so gitlint does not check typo and commit description style? Is this deliberate choice to be not included in PR CI?

Example of typo thats not flagged in PR CI: https://github.com/zephyrproject-rtos/zephyr/pull/29987/files#diff-8e1f696f9ab3c0c376eb7260774364f18de97f3205739af8bffc52cac22455c3R18

cvinayak, 2020-11-18:
Another example, mispelled "instread"  here: https://github.com/zephyrproject-rtos/zephyr/pull/29973/commits/d8e088db48c7a594f4478ba20bff9d4c8accf6be

nashif, 2020-11-19:
gitlint does not check for typos, those are part of the code, not the commit message.

nashif, 2020-11-19:
checkpatch has spelling check, but that is only a warning and is not flagged by CI.

jfischer-no, 2020-12-01:
I reopen this bug report.
The initial description is correct. Checkpatch in CI does not work properly and does not report style issues. I could reproduce it reviewing https://github.com/zephyrproject-rtos/zephyr/pull/30299. PR checked out, `git diff HEAD~1 | checkpatch.pl --mailback --no-tree -` just as `checkpatch.pl -g HEAD` or `checkpatch.pl -g 1cb4ac697d8fe4160f42e82a202f5cbd5b1d01f7` report style issues, but CI does not.

nashif, 2020-12-01:
> report style issues, but CI does not.

because those are warnings and not errors, checkpatch never caused CI to fail on warnings.

nashif, 2020-12-01:
for reference, this is what I get with the PR above:

```
i9:zephyr((HEAD detached from upstream/pull/30299)): git diff upstream/master | ./scripts/checkpatch.pl --mailback --no-tree -
-:340: WARNING:LONG_LINE: line length of 81 exceeds 80 columns
#340: FILE: drivers/display/ls0xx.c:63:
+#if defined(CONFIG_LS0XX_VCOM_DRIVER) && DT_INST_NODE_HAS_PROP(0, extcomin_gpios)

-:349: WARNING:LONG_LINE: line length of 91 exceeds 80 columns
#349: FILE: drivers/display/ls0xx.c:72:
+		gpio_pin_toggle(driver->extcomin_dev, DT_INST_GPIO_PIN(0, extcomin_gpios));

-:351: WARNING:LONG_LINE: line length of 91 exceeds 80 columns
#351: FILE: drivers/display/ls0xx.c:74:
+		gpio_pin_toggle(driver->extcomin_dev, DT_INST_GPIO_PIN(0, extcomin_gpios));

-:359: WARNING:LONG_LINE: line length of 84 exceeds 80 columns
#359: FILE: drivers/display/ls0xx.c:82:
+#elif defined(CONFIG_LS0XX_VCOM_DRIVER) && !DT_INST_NODE_HAS_PROP(0, extcomin_gpios)

-:377: WARNING:LONG_LINE: line length of 82 exceeds 80 columns
#377: FILE: drivers/display/ls0xx.c:100:
+	return gpio_pin_set(driver->disp_dev, DT_INST_GPIO_PIN(0, disp_gpios), 1);

-:397: WARNING:LONG_LINE: line length of 82 exceeds 80 columns
#397: FILE: drivers/display/ls0xx.c:120:
+	return gpio_pin_set(driver->disp_dev, DT_INST_GPIO_PIN(0, disp_gpios), 0);

-:452: WARNING:LONG_LINE: line length of 82 exceeds 80 columns
#452: FILE: drivers/display/ls0xx.c:175:
+#if defined(CONFIG_LS0XX_VCOM_DRIVER) && !DT_INST_NODE_HAS_PROP(0, extcomin_gpios)

-:468: WARNING:LONG_LINE: line length of 93 exceeds 80 columns
#468: FILE: drivers/display/ls0xx.c:191:
+			memcpy(&line_buf[1], buf, LS0XX_PANEL_WIDTH / LS0XX_PIXELS_PER_BYTE);

-:470: WARNING:LONG_LINE: line length of 89 exceeds 80 columns
#470: FILE: drivers/display/ls0xx.c:193:
+			err |= spi_write(driver->spi_dev, &driver->spi_config, &buf_set);

-:479: WARNING:LONG_LINE: line length of 81 exceeds 80 columns
#479: FILE: drivers/display/ls0xx.c:202:
+		err |= spi_write(driver->spi_dev, &driver->spi_config, &buf_set);

-:511: WARNING:LONG_LINE: line length of 81 exceeds 80 columns
#511: FILE: drivers/display/ls0xx.c:234:
+#if defined(CONFIG_LS0XX_VCOM_DRIVER) && DT_INST_NODE_HAS_PROP(0, extcomin_gpios)

-:654: WARNING:LONG_LINE: line length of 90 exceeds 80 columns
#654: FILE: drivers/display/ls0xx.c:377:
+	driver->spi_config.operation |= SPI_CS_ACTIVE_HIGH | SPI_HOLD_ON_CS | SPI_LOCK_ON;

-:668: WARNING:LONG_LINE: line length of 81 exceeds 80 columns
#668: FILE: drivers/display/ls0xx.c:391:
+	driver->disp_dev = device_get_binding(DT_INST_GPIO_LABEL(0, disp_gpios));

-:674: WARNING:LONG_LINE: line length of 96 exceeds 80 columns
#674: FILE: drivers/display/ls0xx.c:397:
+	gpio_pin_configure(driver->disp_dev, DT_INST_GPIO_PIN(0, disp_gpios), GPIO_OUTPUT_HIGH);

-:677: WARNING:LONG_LINE: line length of 81 exceeds 80 columns
#677: FILE: drivers/display/ls0xx.c:400:
+#if defined(CONFIG_LS0XX_VCOM_DRIVER) && DT_INST_NODE_HAS_PROP(0, extcomin_gpios)

-:678: WARNING:LONG_LINE: line length of 89 exceeds 80 columns
#678: FILE: drivers/display/ls0xx.c:401:
+	driver->extcomin_dev = device_get_binding(DT_INST_GPIO_LABEL(0, extcomin_gpios));

-:684: WARNING:LONG_LINE: line length of 85 exceeds 80 columns
#684: FILE: drivers/display/ls0xx.c:407:
+	gpio_pin_configure(driver->extcomin_dev, DT_INST_GPIO_PIN(0, extcomin_gpios),

-:690: WARNING:LONG_LINE: line length of 91 exceeds 80 columns
#690: FILE: drivers/display/ls0xx.c:413:
+						  K_THREAD_STACK_SIZEOF(vcom_toggle_stack),

-:695: WARNING:LONG_LINE_COMMENT: line length of 82 exceeds 80 columns
#695: FILE: drivers/display/ls0xx.c:418:
+#endif  /* CONFIG_LS0XX_VCOM_DRIVER && DT_INST_NODE_HAS_PROP(0, extcomin_gpios) */

- total: 0 errors, 19 warnings, 777 lines checked

NOTE: For some of the reported defects, checkpatch may be able to
      mechanically convert to the typical style using --fix or --fix-inplace.

Your patch has style problems, please review.

NOTE: Ignored message types: AVOID_EXTERNS BRACES C99_COMMENT_TOLERANCE CONFIG_EXPERIMENTAL CONST_STRUCT DATE_TIME FILE_PATH_CHANGES MINMAX NETWORKING_BLOCK_COMMENT_STYLE PREFER_KERNEL_TYPES PREFER_SECTION PRINTK_WITHOUT_KERN_LEVEL REPEATED_WORD SPDX_LICENSE_TAG SPLIT_STRING VOLATILE

NOTE: If any of the errors are false positives, please report
      them to the maintainers.
```

jfischer-no, 2020-12-01:
@nashif But in the past CI published a report about it even if it was a warning. Can we treat this warnings as errors, or at least report them?

nashif, 2020-12-01:
> @nashif But in the past CI published a report about it even if it was a warning. Can we treat this warnings as errors, or at least report them?

IMO reporting and not enforcing is confusing. I am all for enforcing warnings and probably disabling some non-applicable warnings completely so we can have crystal clear CI results. Just reporting warnings is going to yield inconsistent results and response because they are not being enforced.

To get there, the biggest issue is going to be the line length issue we see above, we might need to consider moving to 100 characters per line.  Will you agree to that ;-)

PR #30427 (2021-01-14): ci: handle checpatch warnings as errors and increase line limit to 100
Fail CI if we have both errors and warnings.

Fixes #29960
Closes #30426
closed 2021-01-14 by nashif