zephyrproject-rtos/zephyr #29961: Add i2c driver tests for microchip evaluation board

ashaannajoy, 2020-11-12:
The current tests in zephyr/tests/drivers/i2c are not  enabled for microchip board
either create a new test or perform a generalized test  like 
Perhaps they can perform a generalization from our tests like the one for PM
[](https://github.com/zephyrproject-rtos/zephyr/pull/2773)
The gpio ioexpander is one i2c device on the board which can be tested for detect, read, write and stress tested

ashaannajoy, 2020-12-10:
We request for a generic test which can be common to all processors and maybe configurable further  like configure the basic speed for bus0, try to read register 0x00 for all  slave addresses and identify which  are the slaves which exist in the bus. Further stress testing of data communication can be done.

closed 2021-02-04 by jocelyn-li

chen-png, 2021-02-04:
the patch is merged, and the new testcase " tests/boards/mec15xxevb_assy6853/i2c_api" worked well, so closed this