zephyrproject-rtos/zephyr #29963: dfu/boot/mcuboot: consider usage of boootloader/mcuboot code

nvlsianpu, 2020-11-12:
## Introduction

mcuboot.c file implements functionality required for interact with MCUBoot bootloader.
This had been introduced before the west was introduced.

Some parts of this code are reimpregnations of what is implemented in mcuboot repository.
Mcuboot's repository provide implementation of function required for application for interact with the MCUboot.

### Problem description

As some code are defined here and there it becomes problem when module is compiled within MCUboot.
It is possible to mitigate problem using #fdefry - but this is temporary hack.

carlescufi, 2020-11-12:
CC @utzig and @d3zd3z

de-nordic, 2020-11-30:
Do you consider to make common the FLASH_* definitions so that the macros that are enabling "image" access don't get redefined all over the code?

d3zd3z, 2020-11-30:
I think it makes sense to have much of the definition here in a single place, probably making sense to have it in the mcuboot repo. It mostly makes sense to build this dfu/boot/mcuboot module when mcuboot itself is also being included, so it shouldn't be an issue of the module being present.

PR #30370 (2021-01-14): dfu/boot: rework using newly introduced MCUBOOT_BOOTUTIL library
depends on https://github.com/mcu-tools/mcuboot/pull/892

As some code are defined here and there it becomes problem when module is compiled within MCUboot.
It is possible to mitigate problem using #fdefry - but this is temporary hack.

aim to supersede #30060

fixes #29963

- [x] takes needs of https://github.com/zephyrproject-rtos/zephyr/pull/29669 into account

**step needed for merge this PR**

get the Zephyr's release-team member who can assist with merges sequence smoothly

1. [x]  1. PR approved, 
2. [x] 2.  merge https:// [... cut]
closed 2021-01-14 by carlescufi