zephyrproject-rtos/zephyr #29967: sanitycheck fails to build samples/bluetooth/peripheral_hr/sample.bluetooth.peripheral_hr_rv32m1_vega_ri5cy

galak, 2020-11-12:
`./scripts/sanitycheck -s samples/bluetooth/peripheral_hr/sample.bluetooth.peripheral_hr_rv32m1_vega_ri5cy -l`

Get:

```
/home/galak/git/zephyr/subsys/bluetooth/controller/ll_sw/ull.c:248:7: error: 'BT_CTLR_ADV_SET' undeclared here (not in a function)
  248 |       BT_CTLR_ADV_SET + BT_CTLR_SCAN_SYNC_SET))
      |       ^~~~~~~~~~~~~~~
/home/galak/git/zephyr/subsys/bluetooth/controller/ll_sw/ull.c:252:15: note: in expansion of macro 'PDU_RX_POOL_SIZE'
  252 |  uint8_t pool[PDU_RX_POOL_SIZE];
      |               ^~~~~~~~~~~~~~~~
gmake[2]: *** [zephyr/subsys/bluetooth/controller/CMakeFiles/subsys__bluetooth__controller.dir/build.make:213: zephyr/subsys/bluet
```

galak, 2020-11-12:
This looks like it was introduced by this commit:

commit c2fc629dd21bbaa1fd36678ea4adb28e778b8a0d
Author: Vinayak Kariappa Chettimada <vich@nordicsemi.no>
Date:   Thu Oct 22 18:40:38 2020 +0530

    Bluetooth: controller: 255 byte AD payload support

cvinayak, 2020-11-13:
@MaureenHelm @dleach02 Not sure who is maintaining the openisa port now. I will try patch up build related failure, but I will not be able to test any regressions in functionality. Please bring to notice of resources who can help maintain the port. (I do have a vega board but I dont have a debugger that I can use with it to flash and test).

PR #29987 (2020-11-17): Bluetooth: controller: openisa: Fix sanitycheck failures 
Fix sanitycheck failures when building
samples/bluetooth/peripheral_hr for rv32m1_vega_ri5cy
platform.

Updates from nRF platform related to 255 byte AD data
support have been directly copied into openisa port to
fix build failures.

Relates to commit c2fc629 ("Bluetooth: controller:
255 byte AD payload support").

Fixes #29967.
commit ef2ece82 (2020-11-17): Bluetooth: controller: openisa: Fix sanitycheck failures

Fix sanitycheck failures when building
samples/bluetooth/peripheral_hr for rv32m1_vega_ri5cy
platform.

Updates from nRF platform related to 255 byte AD data
support have been directly copied into openisa port to
fix build failures.

Relates to commit c2fc629dd21b ("Bluetooth: controller:
255 byte AD payload support").

Fixes #29967.
closed 2020-11-17 by carlescufi