zephyrproject-rtos/zephyr #29977: nrf9160: use 32Mhz HFCLK

parthitce, 2020-11-12:
**Describe the bug**
- As per datasheet, user manual and hardware specification, [doc](https://docs.zephyrproject.org/latest/boards/arm/nrf9160dk_nrf9160/doc/index.html#hardware), it's mentioned that nrf9160dk can run in 2 frequencies.
1. 32KHz (32768Hz) from RTC
2. 32MHz (32000000) from High Frequency Clock
- Default value for CONFIG_SYS_CLOCK_HW_CYCLES_PER_SEC is 32Khz [here](https://github.com/zephyrproject-rtos/zephyr/blob/master/soc/arm/nordic_nrf/Kconfig.defconfig#L21).
- Default value for SYS_CLOCK_TICKS_PER_SEC. is 32KHz [here](https://github.com/zephyrproject-rtos/zephyr/blob/master/soc/arm/nordic_nrf/Kconfig.defconfig#L24).

When compiling & running any sample applications for nrf9160 with the default hardware and software frequency of operation is 32KHz. With the default settings, nrf9160dk runs fine without any problem.

We want to run nrf9160 with High Frequency clock i.e 32MHz (32000000Hz). So to achieve the same, I added the below configurations to prj.conf.
```
CONFIG_NRF_RTC_TIMER=n
CONFIG_CORTEX_M_SYSTICK=y
CONFIG_SYS_CLOCK_HW_CYCLES_PER_SEC=32000000
CONFIG_SYS_CLOCK_TICKS_PER_SEC=32000000
```
Adding this configuration to any sample application, fails to boot the target.

**To Reproduce**
Steps to reproduce the behavior:
1. Compile any sample application with below configuration added to prj.conf
```
CONFIG_NRF_RTC_TIMER=n
CONFIG_CORTEX_M_SYSTICK=y
CONFIG_SYS_CLOCK_HW_CYCLES_PER_SEC=32000000
CONFIG_SYS_CLOCK_TICKS_PER_SEC=32000000
```

**Expected behavior**
nrf9160dk should boot fine.

**Impact**
Application business logic and further discussion is detailed [here](https://devzone.nordicsemi.com/f/nordic-q-a/68017/nrf9160dk-using-hfclk-with-zephyr-sdk).

**Logs and console output**
Console log for thread sample tried with 32MHz
```
*** Booting Zephyr OS build zephyr-v2.4.0-1445-g9d1c47521f13  ***
Toggled led0; counter=0
Toggled led1; counter=0
```

**Environment (please complete the following information):**
- git SHA: 9d1c47521f13c8f1cf78483492b1b188eb0c4c1d
- Environment: Linux
- Toolchain: 0.11.4

parthitce, 2020-11-12:
Tagging @anangl as the correct configuration is pointed for nrf52840 here: https://github.com/zephyrproject-rtos/zephyr/issues/28469#issuecomment-704164283

Thanks in advance,

anangl, 2020-11-19:
@parthitce The problem is that the clock signal used by the SysTick counter is stopped when the application core of the nRF9160 is put into sleep mode. Consequently, when the system becomes idle and the core is put into sleep, the counter stops counting and the core cannot be properly waken up after the idle period ends. That's what you are observing.
Because of the above, the SysTick counter is not really applicable as a system clock for nRF9160, as to use it, you would need to somehow prevent the system from suspending the core and putting it into sleep mode. If you really want to try to run your application with the SysTick counter as the system clock, you can try to force the clock signal that the SysTick counter needs to be present even if the core sleeps. This can be done by running the application in the debugger session or by activating all trace and debug clocks via the TAD module (please note that the below register is accessible only from secure code):
```
NRF_TAD_S->CLOCKSTART = 1;
```
But it obviously will be a hack, not a solution for normal applications.
The frequency of the clock signal for the SysTick counter is 64 MHz (32 MHz is the frequency of the external oscillator that the 64 MHz main clock HFXO is derived from), so you'll need the following configuration (assuming that you want the tick frequency of 1 MHz):
```
CONFIG_NRF_RTC_TIMER=n
CONFIG_CORTEX_M_SYSTICK=y
CONFIG_SYS_CLOCK_HW_CYCLES_PER_SEC=64000000
CONFIG_SYS_CLOCK_TICKS_PER_SEC=1000000
```

parthitce, 2020-11-19:
@anangl Thanks for your response. Regarding the below statement,

>Consequently, when the system becomes idle and the core is put into sleep

When enabling the systick timer, nrf9160 is put into sleep mode even before it starts the main thread or application? Because I don't see the main function called or even early.

>This can be done by running the application in the debugger session

Yes, I quickly tried with `west debug` and break very early in `z_cstart` and single stepped until the application main is called and it did work.

>NRF_TAD_S->CLOCKSTART = 1;

In nrf SDK I see API interfaces to get random number, rebooting and so on from non secure application. But I see the nrf9160 is put into sleep very early (even before application starts), adding API for controlling this register isn't an option if I am not wrong. Could you please share how to access this register in this scenario?
Or any other option other than enabling the debug/trace register?

With my current understanding, `NRF_RTC_TIMER` is used as source of timer interrupts for the Zephyr scheduler by default. Changing the source to systick based on system clock isn't possible straight forward with nrf9160? But nrf9160's CPU clock is still running at 64MHz HFXO on demand? Please correct me if I am wrong here.

anangl, 2020-11-19:
> When enabling the systick timer, nrf9160 is put into sleep mode even before it starts the main thread or application? Because I don't see the main function called or even early.

No, it will be put into sleep mode when all your threads become idle, when they will be waiting on some object or just sleeping. If you try to use the hello_world sample with SysTick as the system clock, you should see the "Hello World!" message printed normally. Even if you add a loop there that will print periodically the value returned by `k_cycle_get_32()`, you should see the counter counting, provided that you'll generate intervals between those printings with `k_busy_wait()`. After you switch to `k_sleep()`, you'll see that the application will stop at this sleeping. That's because SysTick will stop counting there, so it won't generate its interrupt that could wake up the core. 

> In nrf SDK I see API interfaces to get random number, rebooting and so on from non secure application. But I see the nrf9160 is put into sleep very early (even before application starts), adding API [...]

I'm not aware of any, As I mentioned, this would be a hack. You'd need to add this code somewhere in the secure firmware.

> With my current understanding, `NRF_RTC_TIMER` is used as source of timer interrupts for the Zephyr scheduler by default. Changing the source to systick based on system clock isn't possible straight [...]

Yes, the CPU always runs at 64 MHz, there is no other option in nRF9160.

closed 2021-02-03 by github-actions[bot]