zephyrproject-rtos/zephyr #29997: format specifies type 'unsigned short' but the argument has type 'int' error in network stack

nashif, 2020-11-13:
**Describe the bug**
Build failure when building with llvm:

```
[ 82%] Building C object zephyr/subsys/net/ip/CMakeFiles/subsys__net__ip.dir/connection.c.obj
[ 83%] Building C object zephyr/subsys/net/ip/CMakeFiles/subsys__net__ip.dir/tcp2.c.obj
/home/nashif/Work/zephyrproject/zephyr/subsys/net/ip/tcp2.c:885:2: error: format specifies type 'unsigned short' but the argument has type 'int' [-Werror,-Wformat]
        conn_send_data_dump(conn);
        ^~~~~~~~~~~~~~~~~~~~~~~~~
/home/nashif/Work/zephyrproject/zephyr/subsys/net/ip/./tcp2_priv.h:102:3: note: expanded from macro 'conn_send_data_dump'
                conn_mss((_conn)));                                     \
                ^~~~~~~~~~~~~~~~~~
/home/nashif/Work/zephyrproject/zephyr/include/net/net_core.h:49:6: note: expanded from macro 'NET_DBG'
                        ##__VA_ARGS__)
```

**To Reproduce**

export ZEPHYR_TOOLCHAIN_VARIANT=llvm
west build -b qemu_x86 tests/net/socket/poll

**Expected behavior**
clean build

nashif, 2020-11-18:
the issue still there, was able to fix it with:

```diff
diff --git a/subsys/net/ip/tcp2_priv.h b/subsys/net/ip/tcp2_priv.h
index f68eee05af..3c0b71742b 100644
--- a/subsys/net/ip/tcp2_priv.h
+++ b/subsys/net/ip/tcp2_priv.h
@@ -99,7 +99,7 @@
                "send_win=%hu, mss=%hu",                                \
                (_conn), net_pkt_get_len((_conn)->send_data),           \
                conn->unacked_len, conn->send_win,                      \
-               conn_mss((_conn)));                                     \
+               (uint32_t)conn_mss((_conn)));                                   \
        NET_DBG("conn: %p send_data_timer=%hu, send_data_retries=%hu",  \
                (_conn),                                                \
                (bool)k_delayed_work_remaining_get(&(_conn)->send_data_timer),\

```

jukkar, 2020-11-18:
Is there a bug in llvm, as the `conn_mss()` is returning `uint16_t` always so there should not be any warning.
The `conn_mss()` is specified as
```
#define conn_mss(_conn)					\
	((_conn)->recv_options.mss_found ?		\
	 (_conn)->recv_options.mss : (uint16_t)NET_IPV6_MTU)
```
and the type of `recv_options.mss` is
```
struct tcp_options {
	uint16_t mss;
	uint16_t window;
	bool mss_found : 1;
	bool wnd_found : 1;
};
```

nashif, 2020-11-19:
> Is there a bug in llvm, as the `conn_mss()` is returning `uint16_t` always so there should not be any warning.

possibly, I was puzzled with this myself.

PR #30038 (2020-11-16): net: tcp2: Use proper int type for connection MSS
When printing MSS (Maximum Segment Size) value, use uint16_t
always.

Fixes #29997
commit 82dbda3b (2020-11-16): net: tcp2: Use proper int type for connection MSS

When printing MSS (Maximum Segment Size) value, use uint16_t
always.

Fixes #29997
PR #30113 (2020-11-19): net: tcp2: Fix llvm issue when printing mss
llvm uses wrong int type for some reason, force cast to uint16_t

Fixes #29997
commit 753412bd (2020-11-19): net: tcp2: Fix llvm issue when printing mss

llvm uses wrong int type for some reason, force cast to uint16_t

Fixes #29997
closed 2020-11-19 by nashif