zephyrproject-rtos/zephyr #29999: nrf52840 Slave mode is not supported on SPI_0

mfinmuch, 2020-11-13:
I am trying to do SPI slave on nrf52840
I refer to the link to make changes: https://github.com/foldedtoad/spi_slave
But his version is too old (2.1.99), and mine is already 2.4.99
After the modification, the result I presented showed it did not support SPI_0
Here is my spi slave setting code
```

#include <string.h>
#include <errno.h>
#include <zephyr.h>
#include <sys/printk.h>
#include <device.h>
#include "spi.h"

#include "spi_slave.h"
int spi_slave_write(struct device * spi, 
                    struct spi_config * spi_cfg,
                    uint16_t * data)
{
    struct spi_buf bufs = {
            .buf = data,
            .len = 2
    };
    struct spi_buf_set tx = {
        .buffers = &bufs
    };
    
    tx.count = 1;

    return spi_write(spi, spi_cfg, &tx);
}
int spi_slave_read(struct device * spi, 
                   struct spi_config * spi_cfg,
                   uint16_t * data)
{
    struct spi_buf bufs = {
            .buf = data,
            .len = 2
    };
    struct spi_buf_set rx = {
        .buffers = &bufs
    };
    
    rx.count = 1;

    return spi_read(spi, spi_cfg, &rx);
}

void spi_slave_init(void)
{
    struct device * spi;
    struct spi_config spi_cfg;

    uint16_t tx_data = 0x5678;
    uint16_t rx_data = 0;

    printk("SPI Slave example application\n");

    printk("rx_data buffer at %p\n", &rx_data);

    //spi = device_get_binding(DT_LABLE(DT_ALIAS(spi0)));
    //spi = device_get_binding(DT_LABEL(DT_INST(0, nordic_nrf52_spis)));
    //spi = device_get_binding(DT_INST_0_NORDIC_NRF_SPIS_LABEL);
    //spi = device_get_binding(DT_ALIAS_SPI_0_LABEL);
    spi = device_get_binding("SPI_0");
    if (!spi) {
        printk("Could not find SPI driver\n");
        return;
    }

    memset(&spi_cfg, 0, sizeof(spi_cfg));

    spi_cfg.operation = SPI_WORD_SET(8) | SPI_OP_MODE_SLAVE;
    spi_cfg.frequency = 1000000;

    printk("%s: slave config @ %p:"
            " wordsize(%u), mode(%u/%u/%u)\n", __func__, &spi_cfg,
            SPI_WORD_SIZE_GET(spi_cfg.operation),
            (SPI_MODE_GET(spi_cfg.operation) & SPI_MODE_CPOL) ? 1 : 0,
            (SPI_MODE_GET(spi_cfg.operation) & SPI_MODE_CPHA) ? 1 : 0,
            (SPI_MODE_GET(spi_cfg.operation) & SPI_MODE_LOOP) ? 1 : 0);

    while (1) {
        spi_slave_read(spi, &spi_cfg, &rx_data);

        if (rx_data == 0x1234) {
            spi_slave_write(spi, &spi_cfg, &tx_data);
        }
        printk("Received: 0x%04X -- %s\n", rx_data,
               (rx_data == 0x1234) ? "ok" : "wrong"); 
    }
}

```
I am not sure if there is a problem with my prj.conf settings
```
CONFIG_DEBUG=y
CONFIG_GPIO=y
CONFIG_SPI=y
CONFIG_SPI_NRFX=y
CONFIG_SPI_ASYNC=y
CONFIG_SPI_SLAVE=y
CONFIG_SPI_0=y
#CONFIG_SPI_0_NRF_SPIS=y
#CONFIG_SPI=y
#CONFIG_SPI_0=y
#CONFIG_SPI_SLAVE=y
CONFIG_SPI_0_OP_MODES=2
#CONFIG_SPI_0_NRF_SPIS=y
CONFIG_SPI_0_NRF_ORC=0xff
CONFIG_NRFX_SPIS0=y

CONFIG_PRINTK=y
CONFIG_LOG=y
CONFIG_LOG_DEFAULT_LEVEL=3
CONFIG_LOG_OVERRIDE_LEVEL=0
CONFIG_LOG_MAX_LEVEL=4
CONFIG_LOG_FUNC_NAME_PREFIX_DBG=y

CONFIG_LOG_PRINTK=y
CONFIG_LOG_PRINTK_MAX_STRING_LENGTH=128
CONFIG_LOG_MODE_OVERFLOW=y
CONFIG_LOG_PROCESS_TRIGGER_THRESHOLD=10
CONFIG_LOG_PROCESS_THREAD=y
CONFIG_LOG_PROCESS_THREAD_SLEEP_MS=1000
CONFIG_LOG_PROCESS_THREAD_STACK_SIZE=768
CONFIG_LOG_BUFFER_SIZE=6144
CONFIG_LOG_DETECT_MISSED_STRDUP=y
CONFIG_LOG_STRDUP_MAX_STRING=32
CONFIG_LOG_STRDUP_BUF_COUNT=4

CONFIG_LOG_BACKEND_SHOW_COLOR=n
```
In addition, in board/nrf52_pca10056.overlay, the setting of spi is as follows
```
&spi0 {
	compatible = "nordic,nrf52-spis";
	label = "SPI_0";
	status = "okay";
	mosi-pin = <28>;
	miso-pin = <29>;
	sck-pin  = <30>;
	csn-pin  = <31>;
	def-char = <0xFF>;
};
```

The result is as follows
 spi_nrfx_spi: Slave mode is not supported on SPI_0
[image]

Is there something wrong with my setting in spi = device_get_binding("SPI_0");
How do I set the device_get_binding of spi
Or, is there an example of spi master/slave of nrf52840 of zephyr 2.4.99v

anangl, 2020-11-13:
The overlay file must match the board name, so in your case it needs to be named `nrf52840dk_nrf52840.overlay` (`nrf52_pca10056` is the old, deprecated name of that board), and you need to use `compatible = "nordic,nrf-spis";`.

mfinmuch, 2020-11-13:
> The overlay file must exactly match the board name, so in your case it needs to be named `nrf52840dk_nrf52840.overlay` (`nrf52840_pca10056` is the old, deprecated name of that board), and you need to use `compatible = "nordic,nrf-spis";`.

How should I use compatible = "nordic,nrf-spis";`
I tried : spi = device_get_binding(DT_LABEL(DT_INST(0, nordic_nrf52_spis)));
But an error occurred
In addition, I also changed the name to nrf52840dk_nrf52840.overlay
```
../src/spi_slave.c: In function 'spi_slave_init':
/home/neat/zephyrtest/zephyr/include/devicetree.h:297:40: error: 'DT_N_INST_0_nordic_nrf52_spis_P_label' undeclared (first use in this function); did you mean 'DT_N_INST_0_nordic_nrf52_spi'?

```
Does he mean change nordic_nrf52_spis to nordic_nrf52_spi?
But I see people say that "s" is for the slave
How can I modify it?

anangl, 2020-11-13:
> How should I use compatible = "nordic,nrf-spis";`
 
Use it in the overlay. Currently you've got there:
```
	compatible = "nordic,nrf52-spis";
```
Change it to:
```
	compatible = "nordic,nrf-spis";
```

mfinmuch, 2020-11-13:
> Change it to:
> 
> ```
> 	compatible = "nordic,nrf-spis";
> ```

 I have made changes in nrf52840dk_nrf52840.overlay
```
&spi0 {
	compatible = "nordic,nrf-spis";
	label = "SPI_0";
	status = "okay";
	mosi-pin = <28>;
	miso-pin = <29>;
	sck-pin  = <30>;
	csn-pin  = <31>;
	def-char = <0xFF>;
};
```
But my compiler still says `spi = device_get_binding(DT_LABEL(DT_INST(0, nordic_nrf_spis))); `
the compiler show as follow

```
../src/spi_slave.c: In function 'spi_slave_init':
/home/neat/zephyrtest/zephyr/include/devicetree.h:297:40: error: 'DT_N_INST_0_nordic_nrf_spis_P_label' undeclared (first use in this function); did you mean 'DT_N_INST_0_nordic_nrf_spim'?
  297 | #define DT_INST(inst, compat) UTIL_CAT(DT_N_INST, DT_DASH(inst, compat))
      |                                        ^~~~~~~~~
/home/neat/zephyrtest/zephyr/include/devicetree.h:1911:38: note: in definition of macro 'DT_CAT'
 1911 | #define DT_CAT(node_id, prop_suffix) node_id##prop_suffix
      |                                      ^~~~~~~
/home/neat/zephyrtest/zephyr/include/devicetree.h:496:27: note: in expansion of macro 'DT_PROP'
  496 | #define DT_LABEL(node_id) DT_PROP(node_id, label)
      |                           ^~~~~~~
../src/spi_slave.c:71:30: note: in expansion of macro 'DT_LABEL'
   71 |     spi = device_get_binding(DT_LABEL(DT_INST(0, nordic_nrf_spis)));
      |                              ^~~~~~~~
/home/neat/zephyrtest/zephyr/include/sys/util_internal.h:105:26: note: in expansion of macro 'UTIL_PRIMITIVE_CAT'
  105 | #define UTIL_CAT(a, ...) UTIL_PRIMITIVE_CAT(a, __VA_ARGS__)
      |                          ^~~~~~~~~~~~~~~~~~
/home/neat/zephyrtest/zephyr/include/devicetree.h:297:31: note: in expansion of macro 'UTIL_CAT'
  297 | #define DT_INST(inst, compat) UTIL_CAT(DT_N_INST, DT_DASH(inst, compat))
      |                               ^~~~~~~~
../src/spi_slave.c:71:39: note: in expansion of macro 'DT_INST'
   71 |     spi = device_get_binding(DT_LABEL(DT_INST(0, nordic_nrf_spis)));
      |                                       ^~~~~~~
/home/neat/zephyrtest/zephyr/include/devicetree.h:297:40: note: each undeclared identifier is reported only once for each function it appears in
  297 | #define DT_INST(inst, compat) UTIL_CAT(DT_N_INST, DT_DASH(inst, compat))
      |                                        ^~~~~~~~~
/home/neat/zephyrtest/zephyr/include/devicetree.h:1911:38: note: in definition of macro 'DT_CAT'
 1911 | #define DT_CAT(node_id, prop_suffix) node_id##prop_suffix
      |                                      ^~~~~~~
/home/neat/zephyrtest/zephyr/include/devicetree.h:496:27: note: in expansion of macro 'DT_PROP'
  496 | #define DT_LABEL(node_id) DT_PROP(node_id, label)
      |                           ^~~~~~~
../src/spi_slave.c:71:30: note: in expansion of macro 'DT_LABEL'
   71 |     spi = device_get_binding(DT_LABEL(DT_INST(0, nordic_nrf_spis)));
      |                              ^~~~~~~~
/home/neat/zephyrtest/zephyr/include/sys/util_internal.h:105:26: note: in expansion of macro 'UTIL_PRIMITIVE_CAT'
  105 | #define UTIL_CAT(a, ...) UTIL_PRIMITIVE_CAT(a, __VA_ARGS__)
      |                          ^~~~~~~~~~~~~~~~~~
/home/neat/zephyrtest/zephyr/include/devicetree.h:297:31: note: in expansion of macro 'UTIL_CAT'
  297 | #define DT_INST(inst, compat) UTIL_CAT(DT_N_INST, DT_DASH(inst, compat))
      |                               ^~~~~~~~
../src/spi_slave.c:71:39: note: in expansion of macro 'DT_INST'
   71 |     spi = device_get_binding(DT_LABEL(DT_INST(0, nordic_nrf_spis)));
```

mfinmuch, 2020-11-17:
I called spi = device_get_binding("SPI_0"); and the problem is solved
thanks~

carlescufi, 2020-11-17:
Closing this as this seems to be solved now.

closed 2020-11-17 by carlescufi