zephyrproject-rtos/zephyr #30012: riscv: many memory protection tests are not building for qemu_riscv

maksimmasalski, 2020-11-13:
**Describe the bug**
Most of the tests from memory protection directory can't be build for Qemu RISCV.

<b>Build failed on qemu_riscv32 and qemu_riscv64:</b>
`tests/kernel/mem_protect/futex/`
`tests/kernel/mem_protect/mem_map/`
`tests/kernel/mem_protect/mem_protect/`
`tests/kernel/mem_protect/obj_validation/`
`tests/kernel/mem_protect/syscalls/`
`tests/kernel/mem_protect/userspace/`

<b>Can build and execute on qemu_riscv32 and qemu_riscv64:</b>
`tests/kernel/mem_protect/protection/` -Test Fails
`tests/kernel/mem_protect/stack_random/` -Test Success
`tests/kernel/mem_protect/stackprot/` -Test Success
`tests/kernel/mem_protect/sys_sem/` -Test Success

**To Reproduce**
Steps to reproduce the behavior:
1. Build that tests one by one
west build -t run -p -b qemu_riscv64 tests/kernel/mem_protect/futex/
2. See error

**Expected behavior**
Tests can be build and executed successfully.

**Impact**
Can't test RISCV memory protection.

**Environment (please complete the following information):**
 - OS: Linux
 - Toolchain Zephyr SDK 0.11.4
 - Commit SHA or Version used cb1cf54406abafb4a781660161a93d5c363c7a1f

**Additional context**
Add any other context about the problem here.

amergnat, 2020-11-13:
Hello,
It's totally normal. QEMU doesn't support PMP properly as describe in the RISCV userspace support pull request.
https://github.com/zephyrproject-rtos/zephyr/pull/27063

A new RISC-V QEMU will be build to support it in the new Zephyr SDK 0.12

If you want test RISCV memory protection on QEMU, please follow the indications in the pull request or use this:
https://github.com/zephyrproject-rtos/sdk-ng/pull/279

With a good QEMU, all test succeed.

In my opinion, this issue can be closed.

andrewboie, 2020-11-13:
I'd like to leave this open until we have a working emulator as this is high priority.
Without CI testing of RiscV userspace, it can bit-rot very quickly. Close this when all the QEMU stuff lands.

amergnat, 2020-11-13:
Ok, you can still test RISCV userspace with hifive_revb board.

andrewboie, 2020-11-13:
> Ok, you can still test RISCV userspace with hifive_revb board (sanity run with it)

We know this.
The point is to get this exercised in continuous integration for pull requests.
We have not had this on ARC, for example, and it has been an ongoing source of problems we are *still* dealing with.

maksimmasalski, 2020-11-14:
The main problem CI runs memory protection tests and it doesn't pass on qemu riscv https://github.com/zephyrproject-rtos/zephyr/pull/29380

amergnat, 2020-11-14:
CI can't run memory protection tests on riscv QEMU because CONFIG_CORE_E31 is false (on Zephyr mainline) for this board.
https://github.com/zephyrproject-rtos/zephyr/blob/master/arch/riscv/Kconfig
https://github.com/zephyrproject-rtos/zephyr/blob/master/boards/riscv/qemu_riscv32/qemu_riscv32_defconfig

Another point: The RISC-V userspace PR couldn't have been merge if CI ran memory protection test.

amergnat, 2020-11-14:
Have you something in your branch which force USERSPACE flag ?

maksimmasalski, 2020-11-14:
> Have you something in your branch which force CORE_E31 flag ?

My PR #29380 doesn't have any RISCV code, it was developed for ARM, x86, ARC. For other platform, it will skip test. Problem, that same test in mainline has build error for RISCV, and when my PR has CI check, it fails the same way. I think maybe necessary to submit a PR (I can do that), to exclude RISCV boards from the tests `tests/kernel/mem_protect/mem_protect/` and `tests/kernel/mem_protect/obj_validation.`

maksimmasalski, 2020-11-14:
> CI can't run memory protection tests on riscv QEMU because CONFIG_CORE_E31 is false (on Zephyr mainline) for this board.
> [https://github.com/zephyrproject-rtos/zephyr/blob/master/arch/riscv/Kconfig](https://github.com/zephyrproject-rtos/zephyr/blob/master/arch/riscv/Kconfig?rgh-link-date=2020-11-14T10%3A10%3A53Z)
> [https://github.com/zephyrproject-rtos/zephyr/blob/master/boards/riscv/qemu_riscv32/qemu_riscv32_defconfig](https://github.com/zephyrproject-rtos/zephyr/blob/master/boards/riscv/qemu_riscv32/qemu_riscv32_defconfig?rgh-link-date=2020-11-14T10%3A10%3A53Z)
> Without this flag, ARCH_HAS_USERSPACE isn't selected then USERSPACE must be false, and RISCV_PMP isn't selected too, that mean all PMP code can't be build, then even if USERSPACE is forced, compilation should fail due to missing pmp code.
> 
> Obviously, the RISC-V userspace PR couldn't have been merge if CI ran memory protection test on QEMU.

I can submit PR to exclude RISCV platform by adding a flag in testcase.yaml file.

amergnat, 2020-11-16:
As I understand from CI logs (https://buildkite.com/zephyr/zephyr/builds/13386#99a367fc-62d7-4b67-992a-066bf4e67f46), this is the piece of code in your patch which trigger the build error:
```
#if defined(CONFIG_X86) || (CONFIG_ARM)
	zassert_true(
		(sem_addr > start_addr) && (sem_addr <= (start_addr + size_heap)),
		"semaphore not in bound of thread's memory pool");
#elif defined(CONFIG_ARC)
	zassert_true(
		(sem_addr < start_addr) && (sem_addr > (start_addr - size_heap)),
		"semaphore not in bound of thread's memory pool");
#else
#error "Not implemented for this architecture"
	zassert_unreachable("Not implemented for this architecture");
#endif
```
RISC-V is missing.

The filter suggestion is a workaround, the tests tests/kernel/mem_protect/mem_protect/ and tests/kernel/mem_protect/obj_validation have been tested and work with QEMU from 0.12 SDK as I explained before, then put an exclude RISCV isn't a good option in my opinion. Additionally, hifive1_revb board (from RISCV arch) handle these tests as well.

IMO, the additional test (semaphore) should support RISC-V too.

About QEMU from 0.11 SDK which isn't support userspace, that should not trigger an issue on your PR because these tests have **filter: CONFIG_ARCH_HAS_USERSPACE**, which is not supported (yet) by QEMU. Please refer to the links in my previous comment for details / evidences. Then, if these tests are built/executed for RISC-V QEMU, that mean something force qemu_riscv32 to be built with userspace capability. I see nothing in your patch that can do that, and it seems impossible someone pushed this without triggering CI issue (like yours), that why I'm suggesting you to rebase your master branch. I have no other idea for now.

maksimmasalski, 2020-11-16:
> As I understand from CI logs ([buildkite.com/zephyr/zephyr/builds/13386#99a367fc-62d7-4b67-992a-066bf4e67f46](https://buildkite.com/zephyr/zephyr/builds/13386#99a367fc-62d7-4b67-992a-066bf4e67f46)), this is the piece of code in your patch which trigger the build error:
> 
> ```
> #if defined(CONFIG_X86) || (CONFIG_ARM)
> 	zassert_true(
> 		(sem_addr > start_addr) && (sem_addr <= (start_addr + size_heap)),
> 		"semaphore not in bound of thread's memory pool");
> #elif defined(CONFIG_ARC)
> 	zassert_true(
> 		(sem_addr < start_addr) && (sem_addr > (start_addr - size_heap)),
> 		"semaphore not in bound of thread's memory pool");
> #else
> #error "Not implemented for this architecture"
> 	zassert_unreachable("Not implemented for this architecture");
> #endif
> ```
> 
> RISC-V is missing. (hifive1_revb board support userspace and is used in CI).
> 
> The filter suggestion is a workaround, the tests tests/kernel/mem_protect/mem_protect/ and tests/kernel/mem_protect/obj_validation have been tested and work with QEMU from 0.12 SDK as I explained before, then put an exclude RISCV isn't a good option in my opinion. Additionally, hifive1_revb board (from RISCV arch) handle these tests as well.
> 
> IMO, the additional test added in your PR should support RISC-V too.
> 
> About 2nd issue:
> QEMU from 0.11 SDK which isn't support userspace, that should not trigger an issue on your PR because these tests have **filter: CONFIG_ARCH_HAS_USERSPACE**, which is not supported (yet) by QEMU. Please refer to the links in my previous comment for details / evidences. Then, if these tests are built/executed for RISC-V QEMU, that mean something force qemu_riscv32 to be built with userspace capability. I see nothing in your patch that can do that, and it seems impossible someone pushed this without triggering CI issue, that why I'm suggesting you to rebase your master branch. I have no other idea for now.

Thank you for the detailed comment, I will follow your advice.