zephyrproject-rtos/zephyr #30014: STM32F411RE PWM support

flnhst, 2020-11-13:
**Describe the bug**
I am using a ST Nucleo F411RE board with an STM32F411RE microcontroller. According to the documentation, it should support PWM. However, according to the DTS file for the board it does not enable any timer and PWM device.

I have tried to enable it manually with an overlay:

    / {
        soc {
            timers1: timers@40010000 {
                status = "okay";
            
                pwm {
                    status = "okay";
                    pinctrl-0 = <&tim1_ch1_pwm_pa8 &tim1_ch2_pwm_pa9>;
                };
            };

            timers2: timers@40000000 {
                status = "okay";
            
                pwm {
                    status = "okay";
                    pinctrl-0 = <&tim2_ch1_pwm_pa0 &tim2_ch1_pwm_pa5 &tim2_ch2_pwm_pa1>;
                };
            };

            timers3: timers@40000400 {
                status = "okay";
            
                pwm {
                    status = "okay";
                    pinctrl-0 = <&tim3_ch1_pwm_pb4>;
                };
            };

            timers4: timers@40000800 {
                status = "okay";
            
                pwm {
                    status = "okay";
                    pinctrl-0 = <&tim4_ch3_pwm_pb8 &tim4_ch4_pwm_pb9>;
                };
            };
        };
    };

And specifying the following in `prj.conf`:

    CONFIG_PWM=y
    CONFIG_PWM_STM32=y

This compiles and flashes successfully. No error is logged at any point when using the PWM API, however the actual produced pulse is 'quirky'.

I used this code to test the PWM: https://gitlab.com/flnhst/st_nucleo_pwm/-/blob/master/src/main.cpp 

On PWM 2, channel 1, it produces a pulse, however, when the first loop through the periods is done, the output stays low, and no pulse is ever produced again.

On PWM 4, channel 3, it produces a pulse, with an inaccurate period (few dozen microseconds off), and the wrong pulse length.

On some PWM devices and channels there is no pulse at all.

**To Reproduce**

You can see the source at https://gitlab.com/flnhst/st_nucleo_pwm/-/tree/master .

I build and flash using:

1. west build -b nucleo_f411re
2. west flash -r jlink

**Expected behavior**
Functional PWM capabilities as described in the documentation.

**Logs and console output**

    *** Booting Zephyr OS build zephyr-v2.4.0-984-g660ad9791bd0  ***
    [00:00:00.006,000] <inf> main: Booting...
    [00:00:00.006,000] <inf> main: Have PWM device.
    [00:00:00.006,000] <inf> main: Entering main loop.
    [00:00:00.006,000] <inf> main: PWM channel 3 set to period 100, pulse 50.
    [00:00:01.006,000] <inf> main: PWM channel 3 set to period 200, pulse 100.
    [00:00:02.006,000] <inf> main: PWM channel 3 set to period 300, pulse 150.
    [00:00:03.006,000] <inf> main: PWM channel 3 set to period 400, pulse 200.
    [00:00:04.006,000] <inf> main: PWM channel 3 set to period 500, pulse 250.
    [00:00:05.006,000] <inf> main: PWM channel 3 set to period 600, pulse 300.
    [00:00:06.007,000] <inf> main: PWM channel 3 set to period 700, pulse 350.
    [00:00:07.007,000] <inf> main: PWM channel 3 set to period 800, pulse 400.

**Environment (please complete the following information):**
- Windows 10
- Zephyr zephyr-v2.4.0-984-g660ad9791bd0

gmarull, 2020-11-18:
Have you checked the if `st,prescaler` value is suitable for the frequency you are working? Note that PWM fixes have been recently merged to the `master` branch, so check if you are using latest changes.
Others:
- `tim4_ch3_pwm_pb8` has been removed, use `tim4_ch3_pb8`
- `CONFIG_PWM_STM32=y` is not required (it is autoselected on STM32 when enabling PWM)

flnhst, 2020-11-21:
First i tried a different prescaler value. Setting it to zero for the timers involved seems to fix all issues, and the output signal is as expected.

Afterwards i pulled the latest changes for Zephyr, and i tried it without the overlay. This still results in PWM and timer devices still being disabled.

I am okay with using an overlay to enable and configure the timers and PWMs, but it would be nice if this was mentioned in documentation, that this extra step is required (for this board at least) for PWM to work.

gmarull, 2020-11-22:
@flnhst The extra-step is only needed if you intend to use PWM at frequencies that require other prescaler values than the default one. DTS bindings for PWM are documented here: https://docs.zephyrproject.org/latest/reference/devicetree/bindings/st%2Cstm32-pwm.html#st-stm32-pwm

Would you mind closing the issue if your issue is fixed?

flnhst, 2020-11-22:
My issue is fixed, thanks for your help 👍

closed 2020-11-22 by flnhst