zephyrproject-rtos/zephyr #30023: Device model: add debug helpers for when device_get_binding() fails

mbolivar-nordic, 2020-11-14:
**Is your enhancement proposal related to a problem? Please describe.**

A common question posed by Zephyr users seeking help, especially new ones, is "why did `device_get_binding()` return `NULL`?"

The answer of course is "the name is wrong or the device initialization function failed". However, the device.h API doesn't have any convenient public routines for checking what happened. It would be nice to define some so users can more easily narrow down the cause of the error without having to grep through driver source code for the init function, turn on logging or set a breakpoint, etc.

**Describe the solution you'd like**

Public functions that can be used to ask the kernel for the answers to these questions:

1. Is `name` the name of a device (i.e. it's the name of something between `__device_start` and `__device_end`)?
2. Assuming the answer to 1. is "yes", is the device usable? If not, why not?

**Describe alternatives you've considered**

Additional logging in `z_sys_init_run_level()` that prints each device's name and initialization function return code

I think having finer-grained functions that don't depend on a log transport being available is a bit more generic and thus hopefully more helpful, e.g. for users that want to debug with printk() or so.

tbursztyka, 2020-11-18:
@mbolivar-nordic something like that #23589, it was an RFC, quite outdated now but I believe the base idea is still valid. (storing initalization status, call status, letting the user continue using the device if the status is ok or the last error was not unrecoverable, etc..)

mbolivar-nordic, 2020-11-18:
Seems like the right direction, yes.

Although thinking about this more, if this is going to continue to work when devices can be put to sleep at any time, we might need some sort of `device_get(struct device*)` / `device_put(struct device*)` instead.

tbursztyka, 2020-11-20:
@mbolivar I don't follow you. Being put to sleep would not wipe the device context. I don't see why you'd need get/put functions.

mbolivar-nordic, 2020-11-23:
I'm thinking of something like this once @pabigot gets ref-counted device power management going:

1. user calls `sensor_sample_fetch(S, ...)` on a sensor device `S`, which has a GPIO device `G` configured in the DT
2. `G` asynchronously gets put to sleep
3. the driver implementation for `sample_fetch` runs and tries to do something with `G`

I guess the sensor driver can't just check the status of `G` at boot time and make sure it's OK; somewhere we need a mechanism to increase the `G` reference to prevent it from being powered down, and drop the reference after the `sample_fetch` call, no?

Maybe I'm misunderstanding where this is all going. If so, sorry for noise.

pabigot, 2020-11-23:
Yes, something is going to keep track of which devices are in demand, and ensure that they're ready to go before they're used: the `device_if_ready()` API description in #30196 is prepared for the possibility that you have to verify more than once.  It's unclear how the runtime dependencies are going to be tracked.

mbolivar-nordic, 2022-12-01:
I'm going to close this as no longer needed now that we have DEVICE_DT_GET and friends.

closed 2022-12-01 by mbolivar-nordic