zephyrproject-rtos/zephyr #30026: [ReOpen]Can not make multiple BLE IPSP connection to the same host

hguomin8, 2020-11-15:
@carlescufi I do more tests with "tcpdump -i bt0" for this issue at the weekend, it looks like the issue come from zephyr but not Bluez, please check below.

IPv6 address for each devices:
> RPI 3B+: global addr: 2001:db8::2, link-local addr: fe80::dca6:32ff:feb3:5b7e
> ipsp node1: global addr: 2001:db8::3, link-local addr: fe80::f2f0:1ff:fec6:c12e
> ipsp node2: global addr: 2001:db8::1, link-local addr: fe80::c387:ecff:fee8:e79a

Connect node1 with 'echo "connect F2:F0:01:C6:C1:2E 2" > /sys/kernel/debug/bluetooth/6lowpan_control', ping from node1 to rpi works fine:
[image]
[image]

and ping from rpi to node1 also works fine:
[image]

Keep the ping from rpi to node1 running, then connect node2 with 'echo "connect C3:87:EC:E8:E7:9A 2" > /sys/kernel/debug/bluetooth/6lowpan_control', at this time rpi is keep sending ping but didn't get any reply from node1:
[image]
try to ping from node1 to rpi, we saw that node1 send NS to query who has 2001:db8::2(rpi) and rpi replied NA to node1, but when rpi send NS to query who is 2001:db8::3(node1), rpi didn't get any reply from node1...
[image]

at this point ping from rpi to node1 or node2,  rpi is keep sending ping message, but didn't get any reply from node1/node2
[image]

Although can't ping each other with global ipv6 address, but it's ok to ping with the link local address:
[image]
[image]
[image]

So it looks like the some issues from zephyr, could you please help to check it again? thanks.

_Originally posted by @hguomin8 in https://github.com/zephyrproject-rtos/zephyr/issues/29684#issuecomment-727569324_

carlescufi, 2021-01-26:
@Vudentz any chance you can comment on this?

carlescufi, 2021-01-28:
@hguomin8 could you please post exact instructions on how to reproduce this problem, including BlueZ and Linux kernel versions?

carlescufi, 2021-01-28:
>Keep the ping from rpi to node1 running, then connect node2 with 'echo "connect C3:87:EC:E8:E7:9A 2" > /sys/kernel/debug/bluetooth/6lowpan_control', at this time rpi is keep sending ping but didn't get any reply from node1:

If that is the case, I don't see how this can be a Zephyr problem. node1, running Zephyr, should not have noticed at all that Linux connected to a second host. It seems to me that the Linux kernel is not correctly configured with multiple BLE nodes on one `bt0` network interface.

carlescufi, 2021-01-28:
Lowering priority since it is not immediately clear that this is an issue on the Zephyr side.

hguomin8, 2021-01-28:
> @hguomin8 could you please post exact instructions on how to reproduce this problem, including BlueZ and Linux kernel versions?

Thanks Carles, details please refer to the issue I opened before: https://github.com/zephyrproject-rtos/zephyr/issues/29684

hguomin8, 2021-01-28:
> If that is the case, I don't see how this can be a Zephyr problem. node1, running Zephyr, should not have noticed at all that Linux connected to a second host. It seems to me that the Linux kernel [...]

If this is an issue from Linux kernel, do you or your team know any solution or work around can be applied to make this kind of multiple 6lowpan nodes connection to a linux host? if no, the 6lowpan over ble might be useless...

carlescufi, 2021-02-04:
> If this is an issue from Linux kernel, do you or your team know any solution or work around can be applied to make this kind of multiple 6lowpan nodes connection to a linux host? if no, the 6lowpan over ble might be useless...

You can contact BlueZ here:
http://www.bluez.org/contact/
or create an issue here, copying @Vudentz:
https://github.com/bluez/bluez/issues

carlescufi, 2021-02-09:
Could you provide full `btmon` logs for the complete connection cycle? We can take a look at those and decide if this is indeed a Zephyr issue.

hguomin8, 2021-02-10:
> Could you provide full `btmon` logs for the complete connection cycle? We can take a look at those and decide if this is indeed a Zephyr issue.

Sure, but let me make it after this Chinese New Year holidays...

carlescufi, 2021-03-11:
> Sure, but let me make it after this Chinese New Year holidays...

Any news @hguomin8?

hguomin8, 2021-03-13:
> > Sure, but let me make it after this Chinese New Year holidays...
> 
> Any news @hguomin8?
Please check the btmon.zip attached, I use the "btmon -w btmon.log" command to generate.

Test step as follows:
**Setup RPI host:**
sudo -i
modprobe bluetooth_6lowpan
echo 1 > /sys/kernel/debug/bluetooth/6lowpan_enable
echo 1 > /proc/sys/net/ipv6/conf/all/forwarding
service radvd restart
hcitool lescan
Ctrl+C

**Connect node1**
echo "connect EF:E4:8B:50:0F:71 2" > /sys/kernel/debug/bluetooth/6lowpan_control
ip address add 2001:db8::1/64 dev bt0

ping6 fe80::efe4:8bff:fe50:0f71%bt0
=>link local ip ping works fine
ping6 2001:db8::efe4:8bff:fe50:0f71
=>global ip ping works fine works fine

**Then connect node2**
echo "connect F2:F0:01:C6:C1:2E 2" > /sys/kernel/debug/bluetooth/6lowpan_control
=>node1's global ip ping be stop...
ping6 fe80::f2f0:01ff:fec6:c12e%bt0
=>link local ip ping works fine
ping6 2001:db8::f2f0:01ff:fec6:c12e
=>global ip ping not working

ping6 fe80::efe4:8bff:fe50:0f71%bt0
=>node1's link local ping also works fine

[btmon.zip](https://github.com/zephyrproject-rtos/zephyr/files/6134110/btmon.zip)

carlescufi, 2021-04-08:
As per the log, the BLE communication between node1 and Linux and node2 and Linux seems stable all through the trace, even after connecting node2. This looks instead like a TCP-related problem, as shown in the trace if you load it in Wireshark:

[image]

@Vudentz or @jukkar could you please take a look at the trace in Wireshark and analyze what is wrong with TCP?

jukkar, 2021-04-08:
What Zephyr version was used here?

I looked the latest upstream TCP code and the RST is send to peer after SYN only if the packet is malformed or we received unknown TCP option. The dump file shows that the only option we receive is the Maximum Segment Size and current TCP code should be able to handle that. There was some fixes for the option handling in this commit 9e8c1b0bbfe02f84185f81002a42967017bcb61a but it is certainly possible that there are still some issues present.

jukkar, 2021-04-08:
I tested the TCP and option handling by sending TCP SYN with the same options as in the btmon.log file. Zephyr did not send RST to peer so our TCP MSS option handling works ok.
Was the application in Zephyr listening the 1883 port?

hguomin8, 2021-04-08:
> What Zephyr version was used here?
> 
> I looked the latest upstream TCP code and the RST is send to peer after SYN only if the packet is malformed or we received unknown TCP option. The dump file shows that the only option we receive is the Maximum Segment Size and current TCP code should be able to handle that. There was some fixes for the option handling in this commit [9e8c1b0](https://github.com/zephyrproject-rtos/zephyr/commit/9e8c1b0bbfe02f84185f81002a42967017bcb61a) but it is certainly possible that there are still some issues present.

Zephyr version is v2.2.0

hguomin8, 2021-04-08:
> I tested the TCP and option handling by sending TCP SYN with the same options as in the btmon.log file. Zephyr did not send RST to peer so our TCP MSS option handling works ok.
> Was the application in Zephyr listening the 1883 port?

Yes, I am running an MQTT app that listening the 1883 port, but this issue can be simply reproduced with MQTT running.

jukkar, 2021-04-08:
> Zephyr version is v2.2.0

Ok, that is an ancient version. Would you be able to try your setup in latest upstream so we could get information if the TCP issue is fixed there.

hguomin8, 2021-04-09:
> > Zephyr version is v2.2.0
> 
> Ok, that is an ancient version. Would you be able to try your setup in latest upstream so we could get information if the TCP issue is fixed there.

Let me try it and get back to you later next week.

hguomin8, 2021-04-17:
@jukkar Just tested this issue with same environment and configuration on Zephyr latest v2.5.0, this issue still exist and the same. the two node's local ip(start with fe80::) ping both works fine, but the global ip(start with 2001:db8::) ping not working.

carlescufi, 2021-04-22:
> @jukkar Just tested this issue with same environment and configuration on Zephyr latest v2.5.0, this issue still exist and the same. the two node's local ip(start with fe80::) ping both works fine, but the global ip(start with 2001:db8::) ping not working.

@jukkar could you take a look to see if this indeed can be TCP-related?

carlescufi, 2021-05-20:
@jukkar ping

carlescufi, 2021-06-17:
@rlubos is there any chance you could take a look at the Wireshark log and see if this could be an issue on the TCP stack?

jukkar, 2021-06-17:
Is this the same issue https://devzone.nordicsemi.com/f/nordic-q-a/70593/nrfconnect-ipsp-example-global-address-unreachable

This might also be related to #33185 where the root cause was the incorrect handling of `lowpan_cb->gw` in linux ipsp driver.

rlubos, 2021-06-17:
@carlescufi The log could indeed indicate some issue with the TCP stack on the Zephyr side, as @jukkar pointed out, however from what I understood from this conversation the log was generated with "Zephyr version v2.2.0" - we still used the "old" TCP stack at that point. So it'd be good to see a more up do date logs.

However, I think the TCP topic slightly deviates from the original report, where TCP wasn't involved, only pings. So I don't think the core issue with handling multiple devices is related to TCP at all.

carlescufi, 2021-07-15:
Closing due to inactivity, @hguomin8 please reopen if you reproduce this with the latest Zephyr and Linux kernels.

closed 2021-07-15 by carlescufi