zephyrproject-rtos/zephyr #30027: sanitycheck failures on `tests/bluetooth/init/bluetooth.init.test_ctlr_peripheral_ext`

galak, 2020-11-15:
`./scripts/sanitycheck -s tests/bluetooth/init/bluetooth.init.test_ctlr_peripheral_ext -l`

Gets:

```
ERROR   - nrf52840dk_nrf52840       tests/bluetooth/init/bluetooth.init.test_ctlr_peripheral_ext FAILED: Build failure
ERROR   - see: /home/galak/git/zephyr/sanity-out/nrf52840dk_nrf52840/tests/bluetooth/init/bluetooth.init.test_ctlr_peripheral_ext/build.log
INFO    - Total complete:    1/   3  33%  skipped:    0, failed:    1
ERROR   - nrf52dk_nrf52832          tests/bluetooth/init/bluetooth.init.test_ctlr_peripheral_ext FAILED: Build failure
ERROR   - see: /home/galak/git/zephyr/sanity-out/nrf52dk_nrf52832/tests/bluetooth/init/bluetooth.init.test_ctlr_peripheral_ext/build.log
INFO    - Total complete:    2/   3  66%  skipped:    0, failed:    2
ERROR   - nrf51dk_nrf51422          tests/bluetooth/init/bluetooth.init.test_ctlr_peripheral_ext FAILED: Build failure
ERROR   - see: /home/galak/git/zephyr/sanity-out/nrf51dk_nrf51422/tests/bluetooth/init/bluetooth.init.test_ctlr_peripheral_ext/build.log
```

From log:

```
/home/galak/git/zephyr/subsys/bluetooth/controller/ll_sw/ull_adv.c: In function 'ull_adv_init':
/home/galak/git/zephyr/subsys/bluetooth/controller/ll_sw/ull_adv.c:1302:9: error: implicit declaration of function 'ull_adv_sync_init'; did you mean 'ull_adv_aux_init'? [-Werror=implicit-function-declaration]
 1302 |   err = ull_adv_sync_init();
      |         ^~~~~~~~~~~~~~~~~
      |         ull_adv_aux_init
/home/galak/git/zephyr/subsys/bluetooth/controller/ll_sw/ull_adv.c: In function 'ull_adv_reset':
/home/galak/git/zephyr/subsys/bluetooth/controller/ll_sw/ull_adv.c:1340:9: error: implicit declaration of function 'ull_adv_sync_reset'; did you mean 'ull_adv_aux_reset'? [-Werror=implicit-function-declaration]
 1340 |   err = ull_adv_sync_reset();
      |         ^~~~~~~~~~~~~~~~~~
      |         ull_adv_aux_reset
cc1: all warnings being treated as errors
gmake[2]: *** [zephyr/subsys/bluetooth/controller/CMakeFiles/subsys__bluetooth__controller.dir/build.make:225: zephyr/subsys/bluetooth/controller
```

cvinayak, 2020-11-16:
Fixed in #29987 and #30007

carlescufi, 2020-11-19:
> Fixed in #29987 and #30007

Closing as this is fixed already.

closed 2020-11-19 by carlescufi