zephyrproject-rtos/zephyr #30028: MPU fault on SAM E70 XPLD with CONFIG_NO_OPTIMIZATIONS=y

Jacharcus, 2020-11-15:
**Describe the bug**
When opening network sockets or running DHCP on a SAM E70 XPLD board, if CONFIG_NO_OPTIMIZATIONS is set to y, I get an MPU fault with the tcp sockets test, as well as the dhcpv4 demo. This does not happen if CONFIG_NO_OPTIMIZATIONS is set to n. 

**To Reproduce**
Edit prj.conf in tests/net/socket/tcp and add CONFIG_NO_OPTIMIZATIONS=n. Build the test and flash it to a SAM E70 XPLD.

**Expected behavior**
The test should pass

**Impact**
This issue stops me from being able to debug code on the SAM E70

**Logs and console output**
START - test_v4_send_recv
E: ***** MPU FAULT *****
E:   Stacking error (context area might be not valid)
E:   Data Access Violation
E:   MMFAR Address: 0x2040acb8
E: r0/a1:  0x20401838  r1/a2:  0x20401030  r2/a3:  0x00000000
E: r3/a4:  0x00000000 r12/ip:  0x00000000 r14/lr:  0x00000000
E:  xpsr:  0x00000000
E: Faulting instruction address (r15/pc): 0x00000000
E: >>> ZEPHYR FATAL ERROR 2: Stack overflow on CPU 0
E: Current thread: 0x20400870 (unknown)
E: Halting system

**Environment (please complete the following information):**
 - OS: Pop!OS 20.10 (Ubuntu 20.10)
 - Toolchain Zephyr SDK
 - Commit SHA: f004410aa1a792201e51adcd9358608f74f6f5c3

**Additional context**
Due to my version of the SAM E70 XPLD board, I also have CONFIG_ETH_SAM_GMAC_MAC_I2C_EEPROM=y, otherwise Zephyr is unnable to pick my MAC address.

ioannisg, 2020-12-16:
> Edit prj.conf in tests/net/socket/tcp and add CONFIG_NO_OPTIMIZATIONS=n. Build the test and flash it to a SAM E70 XPLD.

I suppose you mean =y, @Jacharcus , correct?

ioannisg, 2020-12-16:
@nashif i am not sure about this ticket.

This is a regular stack overflow due to removing all compiler optimizations.
@Jacharcus if you want to test this:
do something along the following in prj.conf:

```
CONFIG_MAIN_STACK_SIZE=4096
CONFIG_ISR_STACK_SIZE=4096
CONFIG_ZTEST_STACKSIZE=8192
CONFIG_IDLE_STACK_SIZE=4096
CONFIG_SYSTEM_WORKQUEUE_STACK_SIZE=4096
CONFIG_TEST_EXTRA_STACKSIZE=4096
CONFIG_PRIVILEGED_STACK_SIZE=4096
CONFIG_NET_RX_STACK_SIZE=4096
CONFIG_NET_TX_STACK_SIZE=4096
```
and the test should pass.

Frankly, i am not sure this should be a bug. Clearly removing optimization increases stack usage, so thread stacks should be resized appropriately.
Not sure whether to close this or not. I leave this to @nashif . Setting to low, for now.

nashif, 2021-02-07:
not a bug, when you change default options, you have to adjust stack sizes.

closed 2021-02-07 by nashif