zephyrproject-rtos/zephyr #30048: Define new label `wifi_serial` for boards with external wifi support (esp8266-01)

ExtremeGTX, 2020-11-16:
## Introduction

Some boards have a connector for an external wifi module/daughterboard over UART.
We need a unified label to access this UART port to avoid creating an overlay file per board.

### Problem description

Some boards have a specific UART connection for an external wifi module like esp8266-01, thus we need to define this new label `wifi_serial` to always point to the UARTx module connected to the external wifi module on a particular board, similar to `arduino_serial` usage.

For example, the following boards have this connector:
- STM32F769i-Disco
- STM32F730I8-Disco
- STM32F723IE-Disco

Note: It is not limited to STM32 boards only.

To get an idea about how this connector may look like, here is an example from the STM32F769I-Disco manual:
[image]

### Proposed change

Define the new label `wifi_serial` for the boards with an external connection to 
a wifi module over UART.

Example: `wifi_serial: &uart5 {};`

## Detailed RFC

### Proposed change (Detailed)

Adding the support of `wifi_serial` to `shields/esp_8266` in  `zephyr/boards/shields/esp_8266/esp_8266_wifi_serial.overlay`
```cs
&wifi_serial {
	status = "okay";
	current-speed = <115200>;

	esp8266 {
		compatible = "espressif,esp";
		label = "esp8266";
		status = "okay";
	};
};

```

then in the board specific dts file:
```cs
wifi_serial: &uart5 {};
```

### Dependencies

Boards which have a serial port for an external wifi module will need to assign it to `wifi_serial`

### Concerns and Unresolved Questions

The extra GPIOs needed for the reset and power control, where should it be defined ?
```dts
		reset-gpios = <&gpioj 14 GPIO_ACTIVE_LOW>;
		power-gpios = <&gpioh 7 GPIO_ACTIVE_HIGH>;
```

## Alternatives
N/A

ExtremeGTX, 2020-11-16:
CC @erwango

nandojve, 2020-11-16:
Hi @ExtremeGTX , I think having the **dts/bindings/gpio/esp-xxx-header.yaml** may resolve the ***-gpios** concerns.