zephyrproject-rtos/zephyr #30066: CI test build with RAM overflow

amergnat, 2020-11-16:
**Describe the bug**
The mem_protect test isn't run during sanity because build detect a RAM overflow
This test shouldn't be skipped

**To Reproduce**
Any sanity on mainline

**Expected behavior**
mem_protect test executed

**Impact**
Can't detect regression (memory domain, permission inheritance, kobject access,...) on future merge.

**Logs and console output**
`INFO    - 12/18 hifive1_revb              tests/kernel/mem_protect/mem_protect/kernel.memory_protection SKIPPED (RAM overflow)`

**Additional context**
For now, this test is manually ran and succeed on a similar board with more RAM.

nashif, 2020-11-17:
if a test binary does not fit in flash/ram, then we there is not much we can do beside optimizing the test code to be smaller, otherwise it will be skipped and is not reported as a failure.

andrewboie, 2020-11-18:
> if a test binary does not fit in flash/ram, then we there is not much we can do beside optimizing the test code to be smaller, otherwise it will be skipped and is not reported as a failure.

The ask is really unclear but what he wants is for the mem_protect test to work on RISC-V with 16K of RAM. This is important to maintain code coverage of memory protection on this arch.

It works on ARM with 16K of ram, so some analysis is necessary to find out why this overflows on RISCV.

nashif, 2020-12-10:
> For now, this test is manually ran and succeed on a similar board with more RAM.

which board is that?

/home/nashif/bin/zephyr-sdk-0.12.0-beta-2/riscv64-zephyr-elf/bin/../lib/gcc/riscv64-zephyr-elf/10.2.0/../../../../riscv64-zephyr-elf/bin/ld: region `RAM' overflowed by 10864 bytes

This is 10K overflow, not sure how we are going to get this into 16k of RAM. This board just needs to be excluded from this test IMO.

@andrewboie  which ARM board with 16K works with this test?

andrewboie, 2020-12-10:
> @andrewboie which ARM board with 16K works with this test?

frdm_k64f:

```
[149/160] Linking C executable zephyr/zephyr_prebuilt.elf
Memory region         Used Size  Region Size  %age Used
           FLASH:       72644 B         1 MB      6.93%
            SRAM:       14276 B       192 KB      7.26%
        IDT_LIST:          88 B         2 KB      4.30%
[160/160] Linking C executable zephyr/zephyr.elf
```

andrewboie, 2020-12-10:
I think I see where this got muddled, ram-report has different numbers:

```
$ ninja ram_report
[0/1] cd /home/apboie/projects/zephyr2/zephyr/tests/ke...ests/kernel/mem_protect/mem_protect/build/ram.json ra
Path                                                                                             Size    %
==============================================================================================================
root                                                                                             4620 17.60%
├── _kernel                                                                                        48 0.18%
├── _memory_protection_test_suite.9302                                                            760 2.90%
├── _sw_isr_table                                                                                 688 2.62%
├── arch                                                                                            1 0.00%
│   └── arm                                                                                         1 0.00%
│       └── core                                                                                    1 0.00%
│           └── aarch32                                                                             1 0.00%
│               └── cortex_m                                                                        1 0.00%
│                   └── mpu                                                                         1 0.00%
│                       └── nxp_mpu.c                                                               1 0.00%
│                           └── static_regions_num                                                  1 0.00%
├── drivers                                                                                       100 0.38%
│   ├── console                                                                                     4 0.02%
│   │   └── uart_console.c                                                                          4 0.02%
│   │       └── uart_console_dev                                                                    4 0.02%
│   ├── gpio                                                                                       60 0.23%
│   │   └── gpio_mcux.c                                                                            60 0.23%
│   │       ├── gpio_mcux_port0_data                                                               12 0.05%
│   │       ├── gpio_mcux_port1_data                                                               12 0.05%
│   │       ├── gpio_mcux_port2_data                                                               12 0.05%
│   │       ├── gpio_mcux_port3_data                                                               12 0.05%
│   │       └── gpio_mcux_port4_data                                                               12 0.05%
│   ├── serial                                                                                     16 0.06%
│   │   └── uart_mcux.c                                                                            16 0.06%
│   │       ├── uart_mcux_0_data                                                                    8 0.03%
│   │       └── uart_mcux_1_data                                                                    8 0.03%
│   └── timer                                                                                      20 0.08%
│       └── cortex_m_systick.c                                                                     20 0.08%
│           ├── announced_cycles                                                                    4 0.02%
│           ├── cycle_count                                                                         4 0.02%
│           ├── last_load                                                                           4 0.02%
│           ├── lock                                                                                4 0.02%
│           └── overflow_cyc                                                                        4 0.02%
├── dynamic_regions.9535                                                                          216 0.82%
├── g_xtal0Freq                                                                                     4 0.02%
├── g_xtal32Freq                                                                                    4 0.02%
├── k_mem_domain_default                                                                          204 0.78%
├── kernel                                                                                        922 3.51%
│   ├── init.c                                                                                    865 3.30%
│   │   ├── z_idle_stacks                                                                         352 1.34%
│   │   ├── z_main_stack                                                                          512 1.95%
│   │   └── z_sys_post_kernel                                                                       1 0.00%
│   ├── mem_domain.c                                                                                1 0.00%
│   │   └── max_partitions                                                                          1 0.00%
│   ├── mutex.c                                                                                     4 0.02%
[... cut ...]
├── z_main_thread                                                                                 200 0.76%
├── z_mem_domain_lock                                                                               4 0.02%
└── ztest_mem_partition                                                                            12 0.05%
==============================================================================================================
                                                                                                26243
```

andrewboie, 2020-12-10:
I opened https://github.com/zephyrproject-rtos/zephyr/issues/30591

The extra memory comes from generating privilege mode stacks.

> This is 10K overflow, not sure how we are going to get this into 16k of RAM. This board just needs to be excluded from this test IMO.

No, we need Risc V coverage, this is not an appropriate resolution.

There is more simply some more work to do in this test case not to generate so many separate stacks, there are currently 9 in total. CONFIG_PRIVILEGED_STACK_SIZE=1024 so if we don't use so many stacks I think we can shave off 10K.

nashif, 2020-12-10:
> No, we need Risc V coverage, this is not an appropriate resolution.

sure, but this is not going to be with this board with only 16k of RAM. We have RISCv coverage in emulation on both renode and qemu. For userspace we probably need to pick another board with more RAM and resources, maybe the VEGA board.

> > @andrewboie which ARM board with 16K works with this test?
> 
> frdm_k64f:

frdm_k64f has way more RAM than 16K.

yerabolu, 2020-12-10:
@nashif : I don't have a VEGA board .. can i reproduce it any other way ?

nashif, 2020-12-10:
> @nashif : I don't have a VEGA board .. can i reproduce it any other way ?

this board might not be right for testing this feature, I was just thinking loud.

TBH, not sure what the bug is. This architecture needs boards both hardware and emulation that would allow testing features that require resources. Right now we do not have many boards support in riscv and those we have either do not support MMU or are low on RAM/Flash.

so I am inclined to just close it and ask for someone to submit a new board configuration that can be used for regression testing or change the configuration of the board or tests to make them fit on the boards we have.

amergnat, 2020-12-14:
> > For now, this test is manually ran and succeed on a similar board with more RAM.
> 
> which board is that?

It's a FPGA (arty 100t) simulating RISC-V core E34 (looks like hifive1_revb) with 32k (or 64k) RAM and Floating Point Unit support.

I think add a new board to test RISC-V's userspace feature isn't the best solution, it look more like a workaround.
I admit rework the test or RISC-V specific code to fit in 16k RAM could be huge, then do what you want, I just giving my opinion.

yerabolu, 2020-12-16:
Based on the discussion above closing this issue.

closed 2020-12-16 by yerabolu