zephyrproject-rtos/zephyr #30075: dfu: mcuboot: fail to build with CONFIG_BOOTLOADER_MCUBOOT=n and CONFIG_IMG_MANAGER=y

mniestroj, 2020-11-16:
**Describe the bug**
Using CONFIG_IMG_MANAGER without CONFIG_BOOTLOADER_MCUBOOT is not longer possible. This is most likely introduced with 5cfafb041b2ab3d63e73c9f800b1174ba8ad57aa.

**To Reproduce**
Steps to reproduce the behavior:
1. west build -b nrf52840dk_nrf52840 samples/hello_world/ -- -DCONFIG_IMG_MANAGER=y -DCONFIG_FLASH=y
2. See error

**Expected behavior**
It is possible to use functionality provided by IMG_MANAGER without building firmware for chain-loading with mcuboot.

**Impact**
Annoyance.

**Logs and console output**
```
zephyr/subsys/dfu/boot/mcuboot.c: In function 'boot_magic_write':
zephyr/subsys/dfu/boot/mcuboot.c:390:34: error: 'boot_img_magic' undeclared (first use in this function)
  390 |  rc = flash_area_write(fa, offs, boot_img_magic, BOOT_MAGIC_SZ);
      |                                  ^~~~~~~~~~~~~~
zephyr/subsys/dfu/boot/mcuboot.c:390:34: note: each undeclared identifier is reported only once for each function it appears in
```

**Environment (please complete the following information):**
 - OS: Linux
 - Toolchain: Zephyr SDK v0.12.0-beta-1
 - Commit SHA: 4d930057d6d667634752e3d0d0dbd677761714e2 (tip of master)

**Additional context**
Add any other context about the problem here.

de-nordic, 2020-11-17:
> It is possible to use functionality provided by IMG_MANAGER without building firmware for chain-loading with mcuboot.

Shouldn't then the mcuboot.c be taken out of compilation for such case?

utzig, 2020-11-17:
I am curious about what would be purpose of using `CONFIG_IMG_MANAGER` without `CONFIG_BOOTLOADER_MCUBOOT`. I agree with @de-nordic that removing mcuboot.c from the build is the correct solution, but I think the image manager module in mcumgr is very tied to mcuboot, for commands like `image test`, etc and won't build without mcuboot.c out of the box.

mniestroj, 2020-11-17:
> I am curious about what would be purpose of using CONFIG_IMG_MANAGER without CONFIG_BOOTLOADER_MCUBOOT.

I use that case for development all the time. It allows me to debug firmware that boots from 0x0 and without bootloader. I use it for both real hw (such as nrf52840dk_nrf52840) and emulated targets (such as qemu_x86) to develop and test features related to firmware upgrade. This is (was) possible, because all the APIs provided by `CONFIG_IMG_MANAGER` (so `flash_img.c` and `mcuboot.c`) are accessible. After finding out that everything there was working (e.g. firmware download and flash) I was building bootloader and application with `CONFIG_BOOTLOADER_MCUBOOT` and do final tests of the feature being developed (which was just seeing if bootloader does the switch correctly).

nvlsianpu, 2020-12-18:
assigned to me, #30370 fixes this issue.

nvlsianpu, 2021-01-14:
#30370 merge fixes the issue

closed 2021-01-14 by nvlsianpu