zephyrproject-rtos/zephyr #30094: tests: kernel: fpu_sharing: Tests in tests/kernel/fpu_sharing fail on nrf platforms

PerMac, 2020-11-17:
**Describe the bug**
Tests in tests/kernel/fpu_sharing fail on nrf platforms (nrf52dk_nrf52840 and nrf52840dk_nrf52840)

**To Reproduce**
Steps to reproduce the behavior:
1. go to zephyr dir
2. have either of the platforms connected
3. run `./scripts/sanitycheck --device-testing -p nrf52dk_nrf52832 -T tests/kernel/fpu_sharing/ --device-serial /dev/ttyACM0 -v -v`
4. See error

**Expected behavior**
Tests passes

**Impact**
Not clear

**Logs and console output**
```
DEBUG   - DEVICE: *** Booting Zephyr OS build zephyr-v2.4.0-1584-g92ecc7866cbe  ***
DEBUG   - DEVICE: Running test suite E: ***** MPU FAULT *****
DEBUG   - DEVICE: E:   Stacking error (context area might be not valid)
DEBUG   - DEVICE: E:   Data Access Violation
DEBUG   - DEVICE: E:   MMFAR Address: 0x20002c18
DEBUG   - DEVICE: E: r0/a1:  0x00000000  r1/a2:  0x00000000  r2/a3:  0x00000000
DEBUG   - DEVICE: E: r3/a4:  0x00000000 r12/ip:  0x00000000 r14/lr:  0x00000000
DEBUG   - DEVICE: E:  xpsr:  0x61000000
DEBUG   - DEVICE: E: s[ 0]:  0x00000000  s[ 1]:  0xffffffff  s[ 2]:  0x00000000  s[ 3]:  0x00000000
DEBUG   - DEVICE: E: s[ 4]:  0x00000000  s[ 5]:  0x00000000  s[ 6]:  0x00000000  s[ 7]:  0x00000000
DEBUG   - DEVICE: E: s[ 8]:  0x00000000  s[ 9]:  0x00000000  s[10]:  0x00000000  s[11]:  0x00000000
DEBUG   - DEVICE: E: s[12]:  0x00000000  s[13]:  0x00000000  s[14]:  0x00000000  s[15]:  0x00000000
DEBUG   - DEVICE: E: fpscr:  0x00000013
DEBUG   - DEVICE: E: Faulting instruction address (r15/pc): 0x000049c8
DEBUG   - DEVICE: E: >>> ZEPHYR FATAL ERROR 2: Stack overflow on CPU 0
DEBUG   - DEVICE: E: Current thread: 0x200004a0 (main)
DEBUG   - DEVICE: E: Halting system
DEBUG   - Timed out while monitoring serial output on nrf52840dk_nrf52840
DEBUG   - run status: timeout
INFO    - 1/1 nrf52840dk_nrf52840       tests/kernel/fpu_sharing/float_disable/kernel.fpu_sharing.float_disable.arm FAILED Timeout (device 65.977s)
```

**Environment (please complete the following information):**
 - OS: Linux 18.4
 - Toolchain Zephyr SDK
 - Commit SHA or Version used: 92ecc7866c

**Additional context**
It seems that another bug (or need for enhancement) was caught here. The test ended with timeout instead of raising an error and terminating the test earlier

PerMac, 2020-11-17:
@nashif (copy from the issue): It seems that another bug (or need for enhancement) was caught here. The test ended with a timeout instead of raising an error and terminating the test earlier.
Should I raise a new issue for this?

nashif, 2020-11-17:
@PerMac  not sure I understand what the issue is?

PerMac, 2020-11-17:
The issue I asking you about is the fact that the test failed causing system halt. However, sanitycheck had to wait for 65s and ended as 
```
DEBUG   - Timed out while monitoring serial output on nrf52840dk_nrf52840
DEBUG   - run status: timeout
```
I wonder if this is expected behavior? I would assume that if there is some fatal error on the board the test should be terminated earlier, instead of waiting for the timeout. Hence the question: Is it a feature (and might be improved if needed) or is it a bug?

ioannisg, 2020-11-17:
This seems like a stack overflow - @PerMac i wonder if you can find the offending commit :)

PerMac, 2020-11-17:
bb99422c8a5f97d250e3e7c956c57ec78fae0e77 is the first bad commit

ioannisg, 2020-11-17:
> @ioannisg [bb99422](https://github.com/zephyrproject-rtos/zephyr/commit/bb99422c8a5f97d250e3e7c956c57ec78fae0e77) is the first bad commit

Great, @PerMac thanks.

That's something for @pabigot to comment on :) Why do we need more stack now?

PerMac, 2020-11-17:
FYI, Tests passes on nrf9160dk_nrf9160 and nrf5340pdk_nrf5340_cpuapp

ioannisg, 2020-11-17:
> FYI, Tests passes on nrf9160dk_nrf9160 and nrf5340pdk_nrf5340_cpuapp

This makes sense - those boards have a different architecture.

pabigot, 2020-11-17:
> That's something for @pabigot to comment on :) Why do we need more stack now?

`CONFIG_CBPRINTF_FP_SUPPORT` defaults to true when `CONFIG_FPU=y` is selected, which causes `*print*` to allocate a larger buffer on the stack for formatted numeric values.  If you set it off (the test doesn't format floating point numbers) the test passes.  Alternatively the test could select `CBPRINTF_NANO=y`, which would have a similar effect.

I'll file a PR that fixes this and another problem identified while diagnosing it.

ioannisg, 2020-11-17:
Thanks @pabigot

PR #30096 (2020-11-17): cbprintf float-related fixes
The `l` length modifier was incorrectly interpreted as invalid when applied to float conversions; it should instead be ignored.

A test that does not format floating point numbers inherited a default configuration that increased stack space to support formatting floating point numbers, causing a failure.  Override the configuration to one that doesn't overflow.

Fixes #30094
closed 2020-11-17 by nashif