zephyrproject-rtos/zephyr #30099: sanitycheck --build-only gets stuck

andrewboie, 2020-11-18:
The run completes but it just sits there without returning to the shell:

```
$ sanitycheck -a riscv32 --build-only
INFO    - Zephyr version: zephyr-v2.4.0-1654-g52dd8e60a1
INFO    - JOBS: 60
INFO    - Selecting default platforms per test case
INFO    - Building initial testcase list...
INFO    - 14202 test configurations selected, 298472 configurations discarded due to filters.
INFO    - Adding tasks to the queue...
INFO    - Total complete:  211/ 211  100%  skipped:   49, failed:    0
INFO    - 162 of 162 tests passed (100.00%), 0 failed, 14040 skipped with 0 warnings in 139.41 seconds
INFO    - In total 1634 test cases were executed on 307 out of total 307 platforms (100.00%)
INFO    - 0 tests executed on platforms, 162 tests were only built.
```

yerabolu, 2020-11-19:
Ran the tests couple of times and it takes a little time to return to the shell. But it does return to the shell.

nashif, 2020-11-19:
> Ran the tests couple of times and it takes a little time to return to the shell. But it does return to the shell.

it is still a bug, should return immediately, based on initial observation it is stuck writing the reports, so something is fishy at this level.

nashif, 2020-11-19:
It is basically spending most of the time writing reports about stuff that did not happen for all platforms, although we asked to only run on a few platforms within riscv32

nashif, 2020-11-28:
I think I fixed this while cleaning up in another PR, see #28372

PR #28372 (2020-12-01): Various sanitycheck fixes and improvements..
use python multiprocessing instead of a thread pool, resolves
various issues.

- [x] Device testing support with multiprocessing

Fixes #28040
Fixes #26949
Fixes #24652
Fixes #30099
Fixes #27583
commit 657b1ac3 (2020-12-01): sanitycheck: fix arch filtering

If we provide arch on command line, do not check for all platforms of
architectures not specified.

Fixes #30099
closed 2020-12-01 by nashif