zephyrproject-rtos/zephyr #30101: tests should not be silently skipped due to insufficient RAM

andrewboie, 2020-11-18:
**Describe the bug**
We have a problem in that if a testcase doesn't fit in target RAM, which would normally produce a build error, instead it just gets skipped:

```
INFO - 12/18 hifive1_revb tests/kernel/mem_protect/mem_protect/kernel.memory_protection SKIPPED (RAM overflow)
```

This is actually a serious problem. It can result in a particular test case being skipped on ALL platforms for a particular arch, and nobody will have any idea or visibility on this without careful inspection of the logs.

**To Reproduce**
See https://github.com/zephyrproject-rtos/zephyr/issues/30066 for an example.
This mem_protect test is uncovered completely on RiscV.

**Expected behavior**
At an absolute bare minimum, some way of ensuring that we don't completely lose code coverage for tests on at least a sub-architecture basis (examples of sub-architectures: x86 64 and 32-bit, ARM cortex-A, -R, -M)

**Impact**
Certain kernel features could end up with no code coverage and nobody aware that this is happening.

PerMac, 2020-12-08:
I was asked about the same issue today by or developers. What @andrewboie is describing seems like a real threat.
I would say that ram overflow during building should be treated as an error. However, tests should also be better in providing how much memory they need. Tests can define min_ram and flash required to run in their yamls.  
The test Andrew is pointing to does not have these min values defined so I think it would also require a fix (and if the policy would be that ram overflow = error, than probably all tests would have to provide min memory values).

nashif, 2020-12-08:
if you want them to be errors, there is not an options ```--overflow-as-errors``` that enables that.

The issue with tests specifying ram limits is that each platform and architecture behaves differently, some boards enable all kind of irrelevant features by default which make them fail.

In the past this was always a very not-fun game, CI would start failing left and right and would keep us modifying RAM limits on tests, doing this without too much thought and excluding more platforms along the way.

PerMac, 2020-12-08:
It seems you actually added this option with recent PR :) https://github.com/zephyrproject-rtos/zephyr/commit/f68146f74ec994b7632921c8b41d62cae188fb0a
But I guess the issue is about a use in a CI, and I guess this --overflow-as-errors won't be default for the CI.

What about a workflow, where if a test has defined min ram/flash sanitycheck will throw an error when it spills over during the building? If so, developers could add min values to the tests they wish to monitor in the CI?

nashif, 2021-02-02:
after enabling overflows as errors, the only 2 platforms getting errors are with very low RAM indeed, so we are not missing much coverage, so lowering priority.

closed 2021-04-06 by PerMac