zephyrproject-rtos/zephyr #30103: Should the kernel be using K_LOWEST_APPLICATION_THREAD_PRIO?

jmgao, 2020-11-18:
My project has some extremely low-priority work that needs to get done at some point eventually, but I want everything else to be able to preempt it. Unfortunately, the kernel uses the `K_LOWEST_APPLICATION_THREAD_PRIO` for shell/logging, so any work done at that priority will block those until it's done.

I found one instance in the tests that adjusts the logging thread's priority, and it seems [rather distasteful](https://github.com/zephyrproject-rtos/zephyr/blob/e696c65de1dd8574cc7e44ce11d0566601238b62/tests/subsys/logging/log_core_additional/src/log_test.c#L375).

Should there be a global K_LOWEST_KERNEL_THREAD_PRIO, and/or individual options for each thread's priority?

jukkar, 2020-11-18:
Could you use `K_LOWEST_THREAD_PRIO` instead?

jmgao, 2020-11-18:
The scheduler [asserts that](https://github.com/zephyrproject-rtos/zephyr/blob/35264cc214fd37e6da8cc682b4f47e86342f517a/kernel/sched.c#L1179) `Z_ASSERT_VALID_PRIO`, and that requires that the caller [must be the idle thread](https://github.com/zephyrproject-rtos/zephyr/blob/d2c71796afc17ae10763b16ad59de9e18831821c/kernel/include/ksched.h#L21) if it uses [`K_LOWEST_THREAD_PRIO`](https://github.com/zephyrproject-rtos/zephyr/blob/4b2de233567f62dd0680289ef9c4e953847f8bab/include/kernel.h#L72).

(I'm not sure if there's a similar assertion for threads that begin existence with `K_LOWEST_THREAD_PRIO`, but I'm running this on the main thread after initial setup is complete, and configuring it to start with `K_LOWEST_THREAD_PRIO` seems like a bad idea? Either way, if there isn't an assertion, that itself seems like a bug: you shouldn't be able to start in priority you can't return to.)

closed 2020-11-19 by nashif