zephyrproject-rtos/zephyr #30105: RFC: C11 (C17) requirement

carlescufi, 2020-11-18:
With the C language evolving, there are a [few features](https://en.wikipedia.org/wiki/C11_(C_standard_revision)#Changes_from_C99) in  ISO/IEC 9899:2011 (aka C11) that would be useful to Zephyr developers. Given that compiler support for it seems to be already widespread, I'd like to start the conversation to discuss whether switching to it as a requirement to build Zephyr is something we would like to consider.

Please use this issue to describe both features from C11 that you'd like to use and objections or showstoppers that you can foresee if we moved forward with the transition.

nordic-krch, 2020-11-18:
It seems that logger would benefit greatly from `_Generic` feature added in `C11`. It allows to evaluate at compile time type of the parameter allowing to detect at compile time: 
- potential strings (char *) which would allow to do internal string duplication without need of explicit `log_strdup` call which was always a problem
- floats for which double space needs to be allocated.
- logging 64 bit values

pabigot, 2020-11-18:
Detecting multi-word arguments should already be handled in cbprintf, but `log_strdup()` is a problem.  Though I'm not clear how `_Generic` could help with a varargs API like logging.

Browsing the [C11](https://en.wikipedia.org/wiki/C11_(C_standard_revision)#Changes_from_C99) additions:
* Using standard thread APIs could be useful.  People clearly want that in C++ for Zephyr, and the C semantics should be compatible.

A consideration is that MISRA is not yet past C99.

carlescufi, 2020-12-17:
@tejlmand could you comment on the full list of toolchains to check support for?
My assumption here is that the list is currently restricted to:

- GCC (version?)
- clang (version?)
- Metaware
- armclang
- esp32 IDF

tejlmand, 2020-12-21:
> My assumption here is that the list is currently restricted to:

Looks precise enough to me.
Unsure how this would impact custom toolchains, @daor-oti comments ?

For the versions, I think:
gcc
- 9.x.x, Zephyr SDK 0.12 is using 10, but gnuarmemb is GCC 9 based.

clang
- 10.0.0 as that is what has recently been introduced.

pabigot, 2020-12-22:
The current ESP32 (xtensa) toolchain is:
```
xtensa-esp32-elf-cc (crosstool-NG esp-2020r3) 8.4.0
Copyright (C) 2018 Free Software Foundation, Inc.
```

manoj153, 2021-01-05:
> > My assumption here is that the list is currently restricted to:
> 
> Looks precise enough to me.
> Unsure how this would impact custom toolchains, @daor-oti comments ?
> 
> For the versions, I think:
> gcc
> 
> * 9.x.x, Zephyr SDK 0.12 is using 10, but gnuarmemb is GCC 9 based.
> 
> clang
> 

> * 10.0.0 as that is what has recently been introduced.

issuing command :

```
./arm-zephyr-eabi-gcc -v

```

returns : 

```
gcc version 10.2.0 (crosstool-NG 1.24.0.212_d7da3a9) 
```

**Development environment:**

1. zephyr-sdk-0.12.0
2. STM32H7xx

from reading from devdocs seems GCC v10 supports C17, I can write my app code successfully with C17?

daor-oti, 2021-01-05:
@tejlmand Thanks for asking. Our toolchain uses clang 9.0.0 as frontend and as such it supports both C11 and C17. We actually set -std=c11 for clang today and have done for a long time. For our custom backend, is a little more difficult to determine whether there will be some corner C11/C17 unsupported areas, but I suspect there will be very few, if any.
So, for our toolchain point of view, I see no immediate problem with Zephyr requiring either C11 or C17.

pabigot, 2021-01-05:
*A propósito*, somebody should verify or update the recommended toolchain for [Vega](https://docs.zephyrproject.org/latest/boards/riscv/rv32m1_vega/doc/index.html#get-the-toolchain-and-openocd), which is currently GCC 7.1.1.  @MaureenHelm 

I'm unable to make it work with west with the following commands:

    export ZEPHYR_TOOLCHAIN_VARIANT=cross-compile
    export CROSS_COMPILE=/usr/local/openisa/bin/riscv32-unknown-elf-

which does work with cmake/ninja directly.

evgeniy-paltsev, 2021-01-12:
Latest ARC GNU toolchain is GCC 10.2.0 based (from Zephyr SDK 0.12 and standalone prebuilt one)
Latest ARC MWDT toolchain is LLVM 10.0.1 based.

pabigot, 2021-01-28:
Here are version numbers that can be used to determine C standard versions at compile-time with code like:
```
#if ((__STDC__VERSION__ - 0) >= 201112L)
/* at least C11 */
#endif
```

Version   | `__STDC__VERSION__`
--------- | ----------------------
C90       | not defined
C95 AMD 1 | 199409L (unofficial)
C99       | 199901L (official)
C11       | 201112L
C17       | 201710L
C20       | 202000L

galak, 2021-02-11:
@nashif can you add details on intel compiler here.

nashif, 2021-03-08:
cadence compiler does not support C11

dleach02, 2021-03-10:
> cadence compiler does not support C11

What is the version it does support?

dleach02, 2021-03-10:
Just a note, MISRA C:2012 - Amendment 2: Updates for ISO/IEC 9899:2011 core functionality, ISBN 978-906400-25-5 (PDF), February 2020.

This document amends MISRA C:2012 as required to introduce support for ISO/IEC 9899:2011. Subsequent amendments will be used to introduce specific guidance for the features introduced by ISO/IEC 9899:2011

galak, 2021-03-10:
We should look to add something in the docs about the fact that we support C99 and why we don't support something newer.

yashi, 2021-05-06:
Just to add a yet another compiler, TI's CGT for ARM (#10182) seems to support C11 but not C17 with v20.2.0.LTS.
https://www.ti.com/lit/pdf/SPNU151V

cfriedt, 2021-11-09:
Wow - I just used C11 generics to implement a certain ISO C library function.

Same code size, but 166x performance improvement.

teburd, 2021-11-17:
C11 could potentially allow for a static call path for drivers using _Generic over a device class API

cfriedt, 2022-08-03:
> > cadence compiler does not support C11
> 
> What is the version it does support?

That's a good question - is anyone able to grok the output of `xt-xcc --help` and `xt-clang --help` and paste the relevant output here?

If xcc is the only toolchain that's holding up switching to C11, it would be good to know.

SebastianBoe, 2022-08-31:
C11 gives use memset_s.

Useful for securely clearing RAM.

stephanosio, 2022-08-31:
> C11 gives us memset_s.

Unfortunately, `memset_s` is part of the Annex K, which is an optional feature (hated by many OSS developers, especially the GNU/Linux people ...).

In fact, none of the standard C libraries we support currently implement `memset_s` (or any other Annex K functions).

aborisovich, 2022-09-16:
> cadence compiler does not support C11

I'm quite sure this is not true.
Cadence is obsoleting xt-xcc and moving to xt-clang. As far as I know there is C++17 support with xt-clang RI-2021.7.
I double there would be no C17 support. I'll check in a moment.

aborisovich, 2022-09-16:
Can't really copy content of the Xtensa guides here as they are copyright protected...
From what they say for 2021.7 toolchain docs is basically:

- Use XT-CLANG for new C and C++ language features, XCC is legacy - stable but not developed any more
- By default xt-clang uses libstdc++11 GNU based
- For best support of the STL use libc++ library and it is required to use C++14 and C++17

Toolchain: RI-2021.7-linux
Short test - C++17:
I've made following main.cpp:
[image]
and executed command: `./xt-clang -std=c++17 main.cpp -o test`
Got linking errors:
[image]
so it looks like C++17 is supported.

Short test - C17:
Created following main.c:
[image]
and executed command: `./xt-clang -std=c17 main.c -o test` . Successfully created executable.

So I think we can confirm C17 and C++17 support for this toolchain. Maybe somebody should check state for 2020.5 which we currently use I believe. I don't really want to push my luck with copyright laws ;-)

carlescufi, 2025-04-28:
@jhol you recently pushed for changes to improve GCC 4.x support in #84119. Could you please state the impact of this proposal to you and your employer so we can take it into account?

nandojve, 2025-04-28:
Could C23, formally ISO/IEC 9899:2024 be considered?

carlescufi, 2025-04-28:
> Could C23, formally ISO/IEC 9899:2024 be considered?

I discussed this briefly with @nashif. I think that might be a bit too recent to require it at this point.

jhol, 2025-04-28:
We are stuck with GCC 4.1.2. I have a collection of patches to get the features that we need working with it - many of which I have upstreamed. If the C standard is increased it obviously has a bigger impact if this is applied to core constructions such as the device tree, sysinit or device macros versus more peripheral features.

carlescufi, 2025-04-28:
> We are stuck with GCC 4.1.2.

@jhol For the foreseeable future? We have been extremely cautious when it comes to updating our minimum required C standard version, but this issue is now close to 5 years old, and C11 itself is close to 15 years old.

carlescufi, 2025-04-29:
> I think zephyr just needs to go ahead with whatever is best for the project. I don't see the need to aggressively break support for C99, but I also don't think the project should expend effort trying to support ancient language standards.

Thanks @jhol. If you are interested this will be discussed tomorrow in the [TSC meeting](https://github.com/zephyrproject-rtos/zephyr/wiki/Technical-Steering-Committee-%28TSC%29).

carlescufi, 2025-04-30:
FYI @RobinKastberg @LoveKarlsson @bjorniuppsala I assume this won't be an issue for IAR?

LoveKarlsson, 2025-04-30:
> FYI [@RobinKastberg](https://github.com/RobinKastberg) [@LoveKarlsson](https://github.com/LoveKarlsson) [@bjorniuppsala](https://github.com/bjorniuppsala) I assume this won't be an issue for IAR?

I don't think so

cfriedt, 2025-04-30:
Generics FTW \o/

JordanYates, 2025-05-01:
> > Could C23, formally ISO/IEC 9899:2024 be considered?
> 
> I discussed this briefly with [@nashif](https://github.com/nashif). I think that might be a bit too recent to require it at this point.

It's worth not forgetting about though. There seems to be quite a number of features in C23 that would be useful.
* Specify the storage type of `enums` (Big fan of this one)
* Builtin overflow safe integer arithmatic, `ckd_add`, `ckd_mul`, etc.
* [#embed](https://en.cppreference.com/w/c/preprocessor/embed) (In combination with west binary blobs)
* [Bit manipulation](https://en.cppreference.com/w/c/numeric/bit_manip) (Replace a bunch of toolchain specific macros, popcount etc)
* Binary number formatting for `printf/scanf`
* Specifying numbers in a clearer fashion, e.g. `0b10101010`, `299'792'458`
* Surely `__VA_OPT__` could be abused to introduce fun new macrobatics :)

Thalley, 2025-05-01:
> Just a note, MISRA C:2012 - Amendment 2: Updates for ISO/IEC 9899:2011 core functionality, ISBN 978-906400-25-5 (PDF), February 2020.
>
> This document amends MISRA C:2012 as required to introduce [...]

Today MISRA C:2023 and C:2025 also exists with support for C11 and C17/18: https://www.perforce.com/resources/qac/misra-c-cpp

donatieng, 2025-05-01:
We internally already require C11 to be used in all of our Zephyr work, mostly for static assertions and anonymous structures, so we'd definitely welcome this change! C23 has a lot of interesting features too but we've not tried it out in a Zephyr project yet.

marc-hb, 2025-05-01:
> We internally already require C11 to be used in all of our Zephyr work, mostly for static assertions and anonymous structures, so we'd definitely welcome this change!

Can you elaborate? AFAIK it's already possible to use C11 today without making changes to Zephyr, is it not? This issue is about bumping the _minimum_ version required.

kartben, 2025-05-01:
> Can you elaborate? AFAIK it's already possible to use C11 today without making changes to Zephyr, is it not? This issue is about bumping the _minimum_ version required.

I think they're just saying that they're having their _downstream_ Zephyr work setup with C11 as a minimum so as to make sure they can leverage the new features, and are happy with it :)

donatieng, 2025-05-01:
Apologies, my first comment was not the clearest.
We have a SDK (https://github.com/blecon/blecon-device-sdk) that our users can integrate in their Zephyr application to use our technology.
Using our SDK requires them to use C11 (or higher) with their application as well, so for us it's better if this becomes the default in Zephyr so that we can drop this additional requirement/difference.

carlescufi, 2025-05-01:
> It's worth not forgetting about though. There seems to be quite a number of features in C23 that would be useful.

Definitely. One thing I will do is open a similar RFC for C23 once we've established C11 as the base C standard for Zephyr, so that we can track it closely. Hopefully it won't take nearly as long to switch to it.

teburd, 2025-05-01:
Skip 11, go straight to the fixed up C17 with its corrections

cfriedt, 2025-05-02:
100% on atomics

cfriedt, 2025-05-02:
I think we will definitely need a system to track all of the changes related to the uplift in minimum version. Maybe a new label `area: c17`?

There are quite a few weird hacks all over the place to account for different language revisions. Do any toolchains or static analysis tools exist to scan an entire codebase and come up with recommended changes?

@fabiobaltieri - can AI help here ?? 😅

RobinKastberg, 2025-05-02:
From IAR:

* As toolchain vendors, we don't mind as long as it is <= C17.
* As Zephyr contributors: If lifting, we prefer C17 to C11.
* As Zephyr contributors: We worry there are other code hygiene issues that time could be better spent on (e.g. VLAs or removing GNUisms).

teburd, 2025-05-12:
Does IAR support gnu statement expressions? This is something that isn't in the standard but is very useful. Hard to see how we avoid using them entirely tbh.

LoveKarlsson, 2025-05-13:
> Does IAR support gnu statement expressions? This is something that isn't in the standard but is very useful. Hard to see how we avoid using them entirely tbh.

IAR does support them nowadays

cfriedt, 2025-05-27:
stdatomic would be nice to use when we upgrade.
https://github.com/zephyrproject-rtos/zephyr/issues/39641