zephyrproject-rtos/zephyr #30106: Refactor zcan_frame.

alexanderwachter, 2020-11-18:
The actual zcan_frame struct and zcan_filter struct is not compatible with the Linux struct.
It would make sense to align them since the actual struct is not efficient in terms of word boundaries.
Additionally we need to add support for CAN-FD.

My proposal is to drop the disticntion of std_id and ext_id.
I tried to somehow keep it, but it lead to undefined behavior.

What I have tried is
```C
struct zcan_frame {
	union {
		struct {
			/** 29-bit Frame identifier.*/
			u32_t ext_id  : 29;
			/** Frame is in the CAN-FD frame format */
			u32_t fd      : 1;
			/** Set the message to a transmission request instead of
			 * data frame. Use can_rtr enum for assignment.
			 */
			u32_t rtr     : 1;
			/** Indicates the identifier type (standard or extended).
			 * Use can_ide enum for assignment.
			 */
			u32_t id_type : 1;
		};
		struct {
			/** 11-bit Frame identifier.*/
			u32_t std_id  : 11;
		};
	};

	/** The length of the message (max. 8) in bytes.*/
	u8_t dlc;
	/** Bit Rate Switch. Increade rate during data phase. */
	u8_t brs : 1;
	/** Reserved for future flags */
	u8_t res : 7;

#if defined(CONFIG_CAN_RX_TIMESTAMP)
	/** Timer value of the CAN free running timer.
	 * The timer is incremented every bit time and captured at the start
	 * of frame bit (SOF).
	 */
	u16_t timestamp;
#else
	/** @cond INTERNAL_HIDDEN */
	u8_t res0;  /* reserved / padding */
	u8_t res1;  /* reserved / padding */
	/** @endcond */
#endif
	/** The frame payload data.*/
	union {
		u8_t data[CAN_MAX_DLEN];
		u32_t data_32[ROUND_UP(CAN_MAX_DLEN / sizeof(u32_t),
				       sizeof(u32_t))];
	};
};
```

My new proposal:

```C
struct zcan_frame {
	u32_t id  : 29;
	/** Frame is in the CAN-FD frame format */
	u32_t fd      : 1;
	/** Set the message to a transmission request instead of
	 * data frame. Use can_rtr enum for assignment.
	 */
	u32_t rtr     : 1;
	/** Indicates the identifier type (standard or extended).
	 * Use can_ide enum for assignment.
	 */
	u32_t id_type : 1;

	/** The length of the message (max. 8) in bytes.*/
	u8_t dlc;
	/** Bit Rate Switch. Increade rate during data phase. */
	u8_t brs : 1;
	/** Reserved for future flags */
	u8_t res : 7;

#if defined(CONFIG_CAN_RX_TIMESTAMP)
	/** Timer value of the CAN free running timer.
	 * The timer is incremented every bit time and captured at the start
	 * of frame bit (SOF).
	 */
	u16_t timestamp;
#else
	/** @cond INTERNAL_HIDDEN */
	u8_t res0;  /* reserved / padding */
	u8_t res1;  /* reserved / padding */
	/** @endcond */
#endif
	/** The frame payload data.*/
	union {
		u8_t data[CAN_MAX_DLEN];
		u32_t data_32[ROUND_UP(CAN_MAX_DLEN / sizeof(u32_t),
				       sizeof(u32_t))];
	};
};
```
The drawback is that we don't get a compiler warning if we accidently exceed 0x7ff for a standard ID

alexanderwachter, 2020-11-18:
@henrikbrixandersen @karstenkoenig @martinjaeger @nixward @lowlander
Can you guys give some input on this topic?

martinjaeger, 2020-11-18:
I prefer the new format. I would even argue that `data_32` should be removed aswell, as it leads to different results depending on the endianness. (ping @martin-aberg)

alexanderwachter, 2020-11-18:
@martinjaeger the data_32 is just for convenience. I'm using it in the drivers to copy the data (for 8 bytes data it is faster to do copy twice instead iof calling memcpy)

martinjaeger, 2020-11-18:
> @martinjaeger the data_32 is just for convenience. I'm using it in the drivers to copy the data (for 8 bytes data it is faster to do copy twice instead of calling memcpy)

Ok, but what about CAN-FD in the future? I guess we don't want to copy 8 times a 32-bit variable instead of just calling memcpy on the buffer.

alexanderwachter, 2020-11-18:
> Ok, but what about CAN-FD in the future? I guess we don't want to copy 8 times a 32-bit variable instead of just calling memcpy on the buffer.

Ture, but then it does not hurt. It makes sure that data is always aligned to a 32-bit boundary.

martin-aberg, 2020-11-18:
Nice improvement!

> Ture, but then it does not hurt. It makes sure that data is always aligned to a 32-bit boundary.

The portable way would be to use the `z_max_align_t` type (https://github.com/zephyrproject-rtos/zephyr/blob/master/include/zephyr/types.h#L34). That will align the data properly on systems which can do large memory accesses but may have stronger alignment requirements.

```c
struct zcan_frame {
        ...
        union {
                uint8_t       bytes[8];
                z_max_align_t aligned_to_anything;
        } data;
};
```
That allows for casting `&data.bytes[0]` into `uint64_t *` for example.

Is it a requirement that the CAN-FD and non-CAN-FD frames are represented by the same data type?

If not, then a flag could indicate if the `zcan_frame` can be interpreted as (cast to) to a `zcanfd_frame` with same content as `zcan_frame`, but having `bytes[64]` instead of `bytes[8]`. Static asserts can be used to make sure the two structs stay in sync.

alexanderwachter, 2020-11-18:
It is not planned to have different frames for FD and classic.

alexanderwachter, 2020-11-19:
Would it be beneficial if we allow external data buffers?
```C
struct zcan_frame {
	uint32_t id  : 29;
	/** Frame is in the CAN-FD frame format */
	uint32_t fd      : 1;
	/** Set the message to a transmission request instead of
	 * data frame. Use can_rtr enum for assignment.
	 */
	uint32_t rtr     : 1;
	/** Indicates the identifier type (standard or extended).
	 * Use can_ide enum for assignment.
	 */
	uint32_t id_type : 1;

	/** The length of the message (max. 8) in bytes.*/
	uint8_t dlc;
	/** Bit Rate Switch. Increade rate during data phase. */
	uint8_t brs : 1;
	/** if set, buf points to the data, else data hold the data */
	uint8_t ext_buf : 1;
	/** Reserved for future flags */
	uint8_t res : 6;

#if defined(CONFIG_CAN_RX_TIMESTAMP)
	/** Timer value of the CAN free-running timer.
	 * The timer is incremented every bit time and captured at the start
	 * of frame bit (SOF).
	 */
	uint16_t timestamp;
#else
	/** @cond INTERNAL_HIDDEN */
	uint8_t res0;  /* reserved / padding */
	uint8_t res1;  /* reserved / padding */
	/** @endcond */
#endif
	/** The frame payload data.*/
	union {
		uint8_t data[CAN_MAX_DLEN];
		uint32_t data_32[ROUND_UP(CAN_MAX_DLEN / sizeof(u32_t),
				       sizeof(u32_t))];
		uint8_t *buf;
	};
};
```

alexanderwachter, 2020-11-19:
I pushed the proposal to #28345

alexanderwachter, 2020-12-18:
Solved by #28345

closed 2020-12-18 by alexanderwachter